tork-a/rtmros_nextage #143: [rtmlaunch] Could not Activate ( > localhost:15005/HrpsysSeqStateROSBridge0.rtc ) : Invalid CORBA naming

130s, 2015-03-04:
A user is experiencing the following error repeatedly after he changes grippers, while he can use the robot as he expects.

> [ERROR] [WallTime: 1425461429.572409] [37.250000] [collision_state.py]
> catch exception, restart rtc_init.
> Make sure collision_pair is set in .conf file. See
> https://github.com/start-jsk/rtmros_common/issues/247
> Original exception:
> CORBA.OBJECT_NOT_EXIST(omniORB.OBJECT_NOT_EXIST_NoMatch, CORBA.COMPLETED_NO)
> ^C[rtmlaunch] [ERROR] Could not Connect ( /localhost:15005/sh.rtc:qOut ,
> /localhost:15005/HrpsysSeqStateROSBridge0.rtc:mcangle ):
> CORBA.COMM_FAILURE(omniORB.COMM_FAILURE_WaitingForReply,
> CORBA.COMPLETED_MAYBE)
> [rtmlaunch] Could not Activate (
> localhost:15005/HrpsysSeqStateROSBridge0.rtc ) : Invalid CORBA naming
> service: localhost:15005

Note that he should be using `nextage_ros_bridge` [0.6.0 from source](9dd8422bdb85630236ee67d8bc2e388b251e37e4), and hrpsys 315.2.8 on QNX.

I have no idea why changing hardware (grippers) on the real robot causes error on `hrpsys`.
@emijah

emijah, 2015-03-06:
Seem to have missed this one(probably more...)
It seems that the RTC is not connecting the ports. Would it be possible to either use the eclipse rtc system builder or the rtshell command line tool to see what sh.rtc:qOut expects to connect to. Also with HrpsysSeqStateROSBridge0.rtc.mcangle it would be nice to know what this expects to connect to too.

k-okada, 2016-09-05:
please reopen if this still happens

closed 2016-09-05 by k-okada