tork-a/rtmros_nextage #145: travis not triggered for all pull requests

130s, 2015-03-09:
Last time [PR was merged on Feb 3, travis must have been working since we can see a check sign](https://github.com/tork-a/rtmros_nextage/pull/140).
Today, with 2 remaining PRs travis doesn't even show its icon, nor it gets triggered.

k-okada, 2015-03-09:
it sometimes happens, please  restart by yourself

130s, 2015-03-09:
Title changed; what I meant is there's even no travis option on pull request window.

k-okada, 2015-03-09:
[image]

130s, 2015-03-09:
I meant this. No trait of travis.
[image]

k-okada, 2015-03-09:
I see, but solution would be same, restart travis or add empty commit, for example just to change commit long and force update the commit

130s, 2015-03-09:
Hm, an empty commit added to https://github.com/tork-a/rtmros_nextage/pull/144 but still no sign of travis.

k-okada, 2015-03-09:
I see, I create new PR #146 and change webhook ssettings but not works,
better to ask more upstream level

k-okada, 2015-03-09:
or start over all settings

130s, 2015-03-09:
Ok I'll re-set the setting first.

130s, 2015-03-09:
Removed and re-added webhook then travis started working again. I suspect I might have messed up "token" on webhook page. The one I took from https://travis-ci.org/profile/info is the right one.

closed 2015-03-09 by 130s