tork-a/rtmros_nextage #148: ImportError: No module named rtmstart

longjie, 2015-03-09:
但馬です。

nextage_ros_bridgeのurdfモデルを変更したくてソースビルドを試みているのですが、

https://github.com/tork-a/rtmros_nextage
https://github.com/start-jsk/rtmros_hironx
https://github.com/start-jsk/rtmros_common

をcloneしてきて、ビルドはエラーなくできましたが、

tajima@nutmeg:~/Codes/ros/hydro/src$ rtmlaunch nextage_ros_bridge nextage_ros_bridge_simulation.launch
Traceback (most recent call last):
  File "/home/tajima/Codes/ros/hydro/devel/bin/rtmlaunch", line 28, in <module>
    import rtmstart
ImportError: No module named rtmstart

と言われてしまいます。なにが足りないのでしょうか。

130s, 2015-03-09:
Since you're using `rtmlaunch` I assume you're on simulation. Please try to follow this [tutorial for simulation](http://wiki.ros.org/rtmros_nextage/Tutorials/Run%20simulation), particularly:
```
$ source `rospack find openrtm_tools`/scripts/rtshell-setup.sh
```

longjie, 2015-03-09:
It worked, thank you!

closed 2015-03-09 by longjie