tork-a/rtmros_nextage #153: ACTION REQUIRED: Important update with version 0.6.0 or higher (March, 2015)

130s, 2015-03-12:
This is an important message for the NEXTAGE OPEN users.

With the upcoming latest version of the software packages, you'll be required to take an action to apply the following changes.

If you are **NOT** planning to update the software by apt-get, you can stop reading this email here.

If you have any questions regarding this, please contact us by either one of the followings:
 * (Recommended) Post comment on this ticket.
 * Ask on the [rtm-ros-robotics mailinglist](https://groups.google.com/forum/#!forum/rtm-ros-robotics|rtm-ros-robotics@googlegroups.com) as usual.
 * Sned a direct email to support[ a-t ]opensource-robotics.tokyo.jp to respect privacy

Users affected
====================

Affected users are those who fit into **BOTH** of these two conditions:

* use either one of the following packages (users usually use both):
 * nextage_ros_bridge 0.6.0 or higher
 * nextage_moveit_config 0.6.0 or higher

* haven't updated the ControllerBox (QNX) software to 315.2.8 or higher
 * You're likely to fit here if you haven't asked any professional service for the update work.

Modifications required
============================

* **Modification-a**. Command to run ROS server on your Ubuntu machine needs an extra argument.
 * BEFORE ver. 0.6.0 `roslaunch nextage_ros_bridge nextage_ros_bridge_real.launch`
 * AFTER ver. 0.6.0  `roslaunch nextage_ros_bridge nextage_ros_bridge_real.launch MODEL:=/opt/jsk/etc/HIRONX/modes/main.wrl`

 Note that you should keep using `HIRONX` for your NEXTAGE robot in this particular command.

* **Modification-b**. If your application uses any of the following method from `NextageClient` class in Python (which is introduced as "`robot`" varibale on `ipython` console in the tutorials), make sure you add an argument `frame_name=None`.

 * [getCurrentPose](http://docs.ros.org/hydro/api/hironx_ros_bridge/html/classhironx__ros__bridge_1_1hironx__client_1_1HIRONX.html#a17afb9d8c83fb9d6b03565b27b7929de)
 * [getCurrentPosition](http://docs.ros.org/hydro/api/hironx_ros_bridge/html/classhironx__ros__bridge_1_1hironx__client_1_1HIRONX.html#a5d3d77ba14a1a38a3cd1763a3dc19f61)
 * [getCurrentRotation](http://docs.ros.org/hydro/api/hironx_ros_bridge/html/classhironx__ros__bridge_1_1hironx__client_1_1HIRONX.html#a07902654de9c39b38d34e45f065c5063)
 * [getCurrentRPY](http://docs.ros.org/hydro/api/hironx_ros_bridge/html/classhironx__ros__bridge_1_1hironx__client_1_1HIRONX.html#ac7c2db48019da907edb0c8dc39c44308)
 * [getReferencePose](http://docs.ros.org/hydro/api/hironx_ros_bridge/html/classhironx__ros__bridge_1_1hironx__client_1_1HIRONX.html#a5c720c0dd42cb572b53ac00c73f2d02d)
 * [getReferencePosition](http://docs.ros.org/hydro/api/hironx_ros_bridge/html/classhironx__ros__bridge_1_1hironx__client_1_1HIRONX.html#ab332daf349e68a2d3932af3ebe3e77da)
 * [getReferenceRotation](http://docs.ros.org/hydro/api/hironx_ros_bridge/html/classhironx__ros__bridge_1_1hironx__client_1_1HIRONX.html#aa550e5acc6aabf6da569f3e63124704c)
 * [getReferenceRPY](http://docs.ros.org/hydro/api/hironx_ros_bridge/html/classhironx__ros__bridge_1_1hironx__client_1_1HIRONX.html#aa6b6616f212c0c7fc74611ecc5ca0655)

 Example: 

  * Output from earlier versions:

   ```
In [18]: robot.getCurrentPosition('RARM_JOINT5')
Out[19]: [0.32484858569580044, -0.1820186913915894, 1.0368028692158924]
```

  * Output from version 0.6.0 or higher:

   ```
In [18]: robot.getCurrentPosition('RARM_JOINT5')
Out[18]: [0.32484858569580044, -0.1820186913915894, 0.0668028692158924]

In [1]: robot.getCurrentPosition('RARM_JOINT5', frame_name='WAIST')
Out[18]: [0.32484858569580044, -0.1820186913915894, 0.0668028692158924]

In [19]: robot.getCurrentPosition('RARM_JOINT5', frame_name=None)
Out[19]: [0.32484858569580044, -0.1820186913915894, 1.0368028692158924]
```

The affecting changes were primarilly made in order to compensate the different height between robots (in their model files, `Hiro` has no base while `NEXTAGE` has) so that uses who have both types of robots couldn't use same code for different robots. Actually these changes provide more solid API in general for Hiro and NEXTAGE, so we hope all users benefit from this. Please do not hesitate, however, to give any feedback.

Isaac from TORK

longjie, 2015-03-13:
I cannot find /opt/jsk/etc/HIRONX/modes/main.wrl . Which package does include this file?
I'm using following packages with NEXTAGE OPEN.

ii  ros-hydro-nextage-description              0.6.1-0precise-20150310-0207-+0000 
ii  ros-hydro-nextage-moveit-config            0.6.1-0precise-20150311-1844-+0000
ii  ros-hydro-nextage-ros-bridge               0.6.1-0precise-20150311-1752-+0000 

ii  ros-hydro-hironx-calibration               1.0.28-0precise-20150206-2221-+0000 
ii  ros-hydro-hironx-moveit-config             1.0.28-0precise-20150311-1752-+0000  
ii  ros-hydro-hironx-ros-bridge                1.0.28-0precise-20150311-1725-+0000                 
ii  ros-hydro-hironx-tutorial                  1.0.27-0precise-20150204-0535-+0000

longjie, 2015-03-13:
OK,  /opt/jsk/etc/HIRONX/models/main.wrl sould be the file in the QNX controller.
Anyway I got enomous error after launch such as:

omniORB: ERROR -- the application attempted to invoke an operation
 on a nil reference.
CORBA::SystemException raised by ModelLoader: IDL:omg.org/CORBA/INV_OBJREF:1.0
CORBA::SystemException raised by ModelLoader: IDL:omg.org/CORBA/INV_OBJREF:1.0
omniORB: ERROR -- the application attempted to invoke an operation
 on a nil reference.

130s, 2015-03-14:
> OK, /opt/jsk/etc/HIRONX/models/main.wrl sould be the file in the QNX controller.

Thanks for clarification, and yes that's correct. I've updated [my original post](https://github.com/tork-a/rtmros_nextage/issues/153#issue-60846562) to clarify this.

> Anyway I got enomous error after launch such as:
> 
> omniORB: ERROR -- the application attempted to invoke an operation on a nil reference.
> CORBA::SystemException raised by ModelLoader: IDL:omg.org/CORBA/INV_OBJREF:1.0

Are you still able to operate the robot as intended despite the errors?

Also, could you run the following script, and send back the resulted tarball to support[a-t]opensource-robotics.tokyo.jp? This script shouldn't be harmful to any corporate network (please take a look at the code in the URL if necessary).
```curl -sL https://raw.githubusercontent.com/start-jsk/rtmros_common/master/hrpsys_tools/scripts/rtmroswtf.sh | bash -```

130s, 2015-03-23:
I'm very sorry that there was a critical mistake.

- WRONG: `AFTER ver. 0.6.0 roslaunch nextage_ros_bridge nextage_ros_bridge_real.launch MODEL:=/opt/jsk/etc/HIRONX/modes/main.wrl`
- CORRECT: `AFTER ver. 0.6.0 roslaunch nextage_ros_bridge nextage_ros_bridge_real.launch MODEL_FILE:=/opt/jsk/etc/HIRONX/model/main.wrl`

longjie, 2015-03-25:
Sorry for late response.
I tried correct command. The robot looks working fine.
However, starting nextage_ros_bridge_real.launch, I got error messages below. Some node dies. 

I have e-mailed diagnosis file which previously you asked.
Thank you.
```
tajima@nextage1:~$ roslaunch nextage_ros_bridge nextage_ros_bridge_real.launch MODEL_FILE:=/opt/jsk/etc/HIRONX/model/main.wrl
... logging to /home/tajima/.ros/log/18a71b3c-d2cb-11e4-b248-000bab77d7b5/roslaunch-nextage1-3622.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://nextage1:55450/

SUMMARY
========

PARAMETERS
 * /HrpsysSeqStateROSBridge/publish_sensor_transforms
 * /collision_state/comp_name
 * /controller_configuration
 * /diagnostic_aggregator/analyzers/computers/contains
 * /diagnostic_aggregator/analyzers/computers/path
 * /diagnostic_aggregator/analyzers/computers/type
 * /diagnostic_aggregator/analyzers/hrpsys/contains
 * /diagnostic_aggregator/analyzers/hrpsys/path
 * /diagnostic_aggregator/analyzers/hrpsys/type
 * /diagnostic_aggregator/analyzers/mode/contains
 * /diagnostic_aggregator/analyzers/mode/path
 * /diagnostic_aggregator/analyzers/mode/type
 * /diagnostic_aggregator/analyzers/motor/contains
 * /diagnostic_aggregator/analyzers/motor/path
 * /diagnostic_aggregator/analyzers/motor/type
 * /diagnostic_aggregator/base_path
 * /diagnostic_aggregator/pub_rate
 * /robot_description
 * /robot_pose_ekf/debug
 * /robot_pose_ekf/freq
 * /robot_pose_ekf/imu_used
 * /robot_pose_ekf/odom_used
 * /robot_pose_ekf/output_frame
 * /robot_pose_ekf/self_diagnose
 * /robot_pose_ekf/sensor_timeout
 * /robot_pose_ekf/vo_used
 * /rosdistro
 * /rosversion

NODES
  /
    CollisionDetectorComp (hrpsys/CollisionDetectorComp)
    DataLoggerServiceROSBridge (hrpsys_ros_bridge/DataLoggerServiceROSBridgeComp)
    ForwardKinematicsServiceROSBridge (hrpsys_ros_bridge/ForwardKinematicsServiceROSBridgeComp)
    HrpsysJointTrajectoryBridge (hrpsys_ros_bridge/HrpsysJointTrajectoryBridge)
    HrpsysSeqStateROSBridge (hrpsys_ros_bridge/HrpsysSeqStateROSBridge)
    RobotHardwareServiceROSBridge (hrpsys_ros_bridge/RobotHardwareServiceROSBridgeComp)
    SequencePlayerServiceROSBridge (hrpsys_ros_bridge/SequencePlayerServiceROSBridgeComp)
    StateHolderServiceROSBridge (hrpsys_ros_bridge/StateHolderServiceROSBridgeComp)
    base_footprint_rootlink (tf/static_transform_publisher)
    collision_state (hrpsys_ros_bridge/collision_state.py)
    diagnostic_aggregator (diagnostic_aggregator/aggregator_node)
    hrpsys_profile (hrpsys_ros_bridge/hrpsys_profile.py)
    hrpsys_py (hironx_ros_bridge/hironx.py)
    hrpsys_ros_diagnostics (hrpsys_ros_bridge/diagnostics.py)
    hrpsys_state_publisher (robot_state_publisher/state_publisher)
    robot_pose_ekf (robot_pose_ekf/robot_pose_ekf)
    rtmlaunch_collision_detector (openrtm_tools/rtmlaunch.py)
    rtmlaunch_hrpsys_ros_bridge (openrtm_tools/rtmlaunch.py)
    sensor_ros_bridge_connect (hrpsys_ros_bridge/sensor_ros_bridge_connect.py)

[... 170 lines cut ...]
already serialized
already serialized
already serialized
already serialized
Traceback (most recent call last):
  File "/opt/ros/hydro/share/hironx_ros_bridge/scripts/hironx.py", line 75, in <module>
    robot.init(robotname=args.robot, url=args.modelfile)
  File "/opt/ros/hydro/lib/python2.7/dist-packages/hironx_ros_bridge/hironx_client.py", line 124, in init
    HrpsysConfigurator.init(self, robotname=robotname, url=url)
  File "/opt/ros/hydro/lib/python2.7/dist-packages/hrpsys/hrpsys_config.py", line 1748, in init
    self.activateComps()
  File "/opt/ros/hydro/lib/python2.7/dist-packages/hrpsys/hrpsys_config.py", line 444, in activateComps
    rtm.serializeComponents(rtcList)
  File "/opt/ros/hydro/lib/python2.7/dist-packages/hrpsys/rtm.py", line 424, in serializeComponents
    if not ec._is_equivalent(rtc.ec):
AttributeError: 'NoneType' object has no attribute 'ec'
[rtmlaunch] Wait for  /nextage:15005/co.rtc:qCurrent   3 /30
[hrpsys_py-1] process has died [pid 3640, exit code 1, cmd /opt/ros/hydro/share/hironx_ros_bridge/scripts/hironx.py --host nextage --port 15005 --modelfile /opt/jsk/etc/HIRONX/model/main.wrl --robot RobotHardware __name:=hrpsys_py __log:=/home/tajima/.ros/log/18a71b3c-d2cb-11e4-b248-000bab77d7b5/hrpsys_py-1.log].
log file: /home/tajima/.ros/log/18a71b3c-d2cb-11e4-b248-000bab77d7b5/hrpsys_py-1*.log
[WARN] [WallTime: 1427273305.259821] Could not found CollisionDetector(co), waiting...
[rtmlaunch] Connected from nextage:15005/HrpsysJointTrajectoryBridge0.rtc:SequencePlayerService
[rtmlaunch]             to nextage:15005/seq.rtc:SequencePlayerService
[rtmlaunch]           with {'id': '', 'properties': {'dataport.publisher.push_policy': 'all', 'dataport.subscription_type': 'new'}, 'name': None, 'verbose': False}
[rtmlaunch] Wait for  /nextage:15005/co.rtc:qCurrent   4 /30
```

534o, 2015-03-25:
Sorry for inconvenience, It seems we have bugs need to be fixed, please try this
```
./hot_fix.sh start-jsk/rtmros_hironx 318
./hot_fix.sh start-jsk/rtmros_hironx 343
./hot_fix.sh fkanehiro/hrpsys-base 534 /opt/ros/hydro/lib/python2.7/dist-packages/hrpsys 2
```
to obrain `hot_fix.sh` program
```
wget https://gist.githubusercontent.com/534o/de86e3d4ebce1f9e932a/raw/b83618d6a94c2c083cdf22a1fd02d86b20023eb6/hot_fix.sh -O hot_fix.sh
chmod u+x hot_fix.sh
```

you can revert these patches (either intentionally  or accidentally) 
```
sudo apt-get install --reinstall ros-hydro-hrpsys
sudo apt-get install --reinstall ros-hydro-hironx-ros-bridge
```

-- your friendly neighborhood  TORK dev team

longjie, 2015-03-30:
Some warnings and errors (red text in console) remains, but are'nt they critical?
If not so, I believe the hotfixes solved the error. Thank you!

```
tajima@nextage1:~/Codes/ros/hydro$ roslaunch nextage_ros_bridge nextage_ros_brid
ge_real.launch MODEL_FILE:=/opt/jsk/etc/HIRONX/model/main.wrl
```
First warnings:
```
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtactivate
WARNING: unrecognized tag rtactivate
WARNING: unrecognized tag rtactivate
WARNING: unrecognized tag rtactivate
WARNING: unrecognized tag rtactivate
WARNING: unrecognized tag rtactivate
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
```
Second fail:
```
[hrpsys.py]    eval :  [self.rmfo, self.rmfo_svc, self.rmfo_version] = self.crea
teComp("AbsoluteForceSensor","rmfo")
[hrpsys.py]  Fail to createComps 'NoneType' object has no attribute 'ref' 
[hrpsys.py]  connecting components
```
Third fails:
```
[rtm.py]    Failed to connect None to [None]
sh.basePosOut and seq.basePosInit are already connected
sh.basePosOut and fk.basePosRef are already connected
sh.baseRpyOut and seq.baseRpyInit are already connected
sh.baseRpyOut and fk.baseRpyRef are already connected
sh.qOut and seq.qInit are already connected
process[base_footprint_rootlink-17]: started with pid [9585]
[rtm.py]    Failed to connect sh.zmpOut to None
RobotHardware0.q and ic.qCurrent are already connected
[rtm.py]    Failed to connect sh.basePosOut to None
[sensor_ros_bridge_connect.py]  wait for RTCmanager :  nextage
[rtm.py]    Failed to connect sh.baseRpyOut to None
RobotHardware0.q and el.qCurrent are already connected
[hrpsys.py]  activating components
[sensor_ros_bridge_connect.py]  wait for RobotHardware0  :  <hrpsys.rtm.RTcompon
ent instance at 0x23f7440> (timeout  0  < 10)
[sensor_ros_bridge_connect.py]  findComps -> RobotHardware :  <hrpsys.rtm.RTcomp
onent instance at 0x23f7440> isActive? =  True
[hrpsys.py]  Fail to find instance (['rmfo', 'AbsoluteForceSensor']) for getRTCI
nstanceList
```
Forth fail is:
```
[hrpsys.py]    setupLogger : record type = TimedLongSeqSeq , name =  servoState 
 to  RobotHardware0_servoState
[rtm.py]    Failed to connect RobotHardware0.servoState to None
[hrpsys.py]  setup joint groups
```

534o, 2015-03-30:
Warnings and errors are expected and critical for robots have force sensors, but if you does not have one, it is ok.
Does you robot working fine, after this patch?
-- your friendly neighborhood TORK dev team

longjie, 2015-03-30:
Yes, it's working fine. Thank you for your effort.

130s, 2015-05-11:
Closing since no major dispute has been heard for awhile. Thank you for the attention and feedback (esp. @longjie).

closed 2015-05-11 by 130s

k-okada, 2016-10-18:
```
BEFORE ver. 0.6.0 roslaunch nextage_ros_bridge nextage_ros_bridge_real.launch
AFTER ver. 0.6.0 roslaunch nextage_ros_bridge nextage_ros_bridge_real.launch MODEL_FILE:=/opt/jsk/etc/HIRONX/model/main.wrl
```
why we have to change this? why we did not set default MODEL_FILE values?