tork-a/rtmros_nextage #157: Release into ROS Indigo

130s, 2015-03-23:
Hydro will be End-Of-Lifeed this year 

(I realized that I was misunderstanding the ROS support cycle but it's said:  http://answers.ros.org/question/200960/how-long-is-support-for-ros-hydro/)

We'll have to **officially** move on to Indigo this year. Given [J-turtle is expected to come out within the 1st half of 2015](http://wiki.ros.org/Distributions/#Next_Distribution_Release), official announce should happen around that time too. Maybe by the beginning of July 2015.

Possible TODOs:
- [x] DEB release
- [ ] Call for early adopter 
- [ ] Add travis
- [ ] Tests with real hardware

k-okada, 2015-03-25:
please also checks
- https://github.com/rdiankov/collada-dom/pull/13
- https://github.com/jsk-ros-pkg/jsk_model_tools/issues/99
for indigo release, although nextage itself did not depends on collada, 
other hrpsys robots does.

130s, 2015-04-01:
Delegated to https://github.com/start-jsk/rtmros_hironx/issues/202. Unless there will be NEXTAGE specific tasks, let's continue there for Indigo release.

130s, 2015-05-21:
We might need to separately work on "port to ROS Indigo" task from Hironx. 
With https://github.com/tork-a/rtmros_nextage/issues/173 I noticed that NXO depends on a specific version of ueye SDK (latest 4.6.1 is not compatible according to @emijah).

emijah, 2015-05-21:
There is a problem with the ueye drivers. The best version right now is version 4.4

130s, 2015-05-26:
We have to take care of both new and existing owners of this robot regarding ROS Indigo.

- For new owners: It's fine as long as the manufacturer or someone else installs the confirmed version of ueye SDK (Btw I've confirmed that 4.4.0 works with 14.04, [same as 12.04](https://github.com/tork-a/rtmros_nextage/issues/173#issuecomment-104084251)).
- For existing users, how the manufacturer provides the confirmed version of the driver might be a question (I know it's none of the business in opensource community though).

130s, 2015-05-26:
I confirmed that ueye cam SDK 4.61 version (that's available online as of today) works on 14.04, with its proprietary manager software. I haven't tested with other software (OpenCV, ROS etc.)

130s, 2015-10-16:
All checked.

closed 2015-10-16 by 130s