tork-a/rtmros_nextage #160: setTargetPoseRelative always fails

130s, 2015-03-29:
Moved from https://github.com/start-jsk/rtmros_hironx/issues/333
All `setTargetPoseRelative` command trials fail on simulator as far as I've tried today (giving separately dx, dy, dz, dr, dp, dw, one-by-one).

```
$ dpkg -p ros-indigo-hrpsys ros-indigo-openhrp3 ros-indigo-nextage-ros-bridge | grep Ver
Version: 315.3.0-1trusty-20150323-1010-+0000
Version: 3.1.7-15trusty-20150213-1223-+0000
Version: 0.6.0-0trusty-20150324-0008-+0000
```

k-okada, 2015-03-29:
> Moved from start-jsk/rtmros_hironx#333

that's because in setTargetPoseRelative, it uses
ret, tds = self.fk_svc.getCurrentPose(eename) (without frame)
to get current pose but use
ret = self.setTargetPose(gname, list(posRef), list(rpyRef), tm)
which implicitly set frame at
https://github.com/start-jsk/rtmros_hironx/blob/hydro-devel/hironx_ros_bridge/src/hironx_ros_bridge/hironx_client.py#L740
 https://github.com/fkanehiro/hrpsys-base/blob/master/python/hrpsys_config.py#L1275

so use  self.getCurrentPose(eename) instead of
self.fk_svc.getCurrentPose(eename) may solve problem.

130s, 2015-03-30:
Hm, the question is why does this happen ONLY with NEXTAGE. Do you have any idea on this?

In terms of setTargetPoseRelative, [nextage_client.py](https://github.com/tork-a/rtmros_nextage/blob/hydro-devel/nextage_ros_bridge/src/nextage_ros_bridge/nextage_client.py) does nothing but just inherits `HIRONX` class, which is a direct child class of `HrpsysConfigurator`. So there shouldn't be any special info added on to NEXTAGE. Only difference I think of is 3D model where NEXTAGE has bigger grippers.

k-okada, 2015-03-31:
that's because the position of 'RARM_JOINT5' wrt WAIST is different from
'RARM_OJINT' wrt None, in Nextage case
```
In [8]: hcf.getCurrentPosition('RARM_JOINT5', 'WAIST')
Out[8]: [0.32484858569580044, -0.1820186913915894, 0.0668028692158924]

In [9]: hcf.getCurrentPosition('RARM_JOINT5')
Out[9]: [0.32484858569580044, -0.1820186913915894, 1.0368028692158924]
```

130s, 2015-03-31:
That makes very much sense.

PR in the upstream https://github.com/fkanehiro/hrpsys-base/issues/543 should address this issue.

closed 2015-04-01 by 130s