tork-a/rtmros_nextage #161: eef axis may not be intuitive

130s, 2015-04-02:
#Colors are RBG : XYZ.

While arms are facing down, x-axis faces up.
[image]

I'm still new to humanoid but is this the common frame orientation? [REP-0120: Coordinate Frames for Humanoids Robots](http://www.ros.org/reps/rep-0120.html) doesn't specify this detail.

PR2's hand is intuitive: x faces forward, z is up (and thus y is left-bound).
[image]

k-okada, 2015-04-04:
since there are so many variation on hands/gripper, so I believe it is not 
able to define common convention of gripper orientation.
As for figures, you see joint/link coordinates on  nextage and see tip link 
for PR2 robots, At this moment we do not define tip link in hrpsys robot.

130s, 2015-04-04:
> since there are so many variation on hands/gripper, so I believe it is not
able to define common convention of gripper orientation.

I see.

I'm sorry if I wasn't clear enough; my point was the "direction" of the axis end effector for NXO (I just referred to PR2 as an example). Since the hand looks like facing down at its "goInitial" pose, I thought the roll-axis being facing down toward the ground as well would be the most intuitive (that's the purpose I cited PR2, where the roll-axis is facing forward as the grippers "look like" facing toward the front too).

k-okada, 2015-04-04:
Yes direction is also ambiguous. So each robot may have different rule. We define our local rule at JSK such as:
- Assuming there are a cup on the table and the z orientation of the cup is up direction
- If we solve IK using tip coordinates, it is convenient that z direction of a top and a cup is aligned when the robot grasp the cup naturally 

And even if we apply this rule to the nextage robot, we still have question  that what is the natural grasping for this robot? As you said, this robot is designed to grasp the cup from top side, and obviously the nominal gripper directions is different from pr2. So we have no clear answer for that. May be I would choose z axis is eef to up, when the robot is initial pose, but you can choose any direction you want.

Please not that what you seen in rviz is joint direction and not eef direction and eef is not defined in wrl yet. In urdf if is defined as fixed link. And direction of joint is usually choose z to rotation axis, but wrl/collada/urdf is designed to choose any direction for rotation axis.

130s, 2015-04-06:
Thank you for the clarification. I [added a note for the sake of the programmers](http://wiki.ros.org/action/info/rtmros_nextage/Tutorials/Programming?action=diff&rev2=13&rev1=12).

closed 2015-04-06 by 130s