tork-a/rtmros_nextage #163: What is the BODY frame and how can I set it?

longjie, 2015-04-08:
The NEXTAGE Open's root link is WAIST.
I'm getting warning that some frame has no transformation from WAIST.

* BODY
* odom
* base_footprint

odom and base_footprint is OK, they are for mobile robots. They should be somewhere on the ground.
What is the BODY frame and how can I set it?

k-okada, 2015-04-08:
could you paste error message?
and also is it possible to reproduce this with simulator?

longjie, 2015-04-10:
It's not errors, just warnings. I have ignored this until moveit_planning_execution.launch produce massive waning messages to the console (as the SS I attached). I guess it is no problem to publish some tf to eliminate them. I just want to know what is the correct transform from WAIST to BODY and what BODY frame means.

And yes, this warning happens on simulator. Doing:

```
$ rtmlaunch nextage_ros_bridge nextage_ros_bridge_simulation.launch
```
and running rviz, we get same warnings "No transform from [BODY] to frame [WAIST]".
Version is 0.6.1

Thank you.

[image]

k-okada, 2015-04-14:
Although I did not get so many warnings, I have noticed that odom-> base_footprint -> BODY is orphaned . I guess it comes from https://github.com/start-jsk/rtmros_common/blob/master/hrpsys_ros_bridge/launch/hrpsys_ros_bridge.launch#L234 and we have to specify `BASE_LINK` when we include hrpsys_ros_bridge.launch form nextage/hiro.

[image]
[image]

longjie, 2015-04-15:
I see. The BODY is for humanoids, to estimate the floating coordinate from gyro-sensors.
I simply added hironx_ros_bridge.launch 
```
    <arg name="BASE_LINK" value="WAIST"/>
```
and tf is now completely healthy. Thank you!

k-okada, 2015-04-16:
great! could you commit that fix if you have time?
Thanks in advance

closed 2015-07-30 by 130s