tork-a/rtmros_nextage #165: collada_to_urdf error: "Target Node visual1/node_joint0_axis0 NOT found!!!"

130s, 2015-05-11:
Referring to [hironx_moveit_config/CMakeLists.txt](https://github.com/start-jsk/rtmros_hironx/blob/c861995a7df8f6240afec4e6b2bac21ced49bc3a/hironx_moveit_config/CMakeLists.txt), I'm trying to manually re-generate urdf based on the discussion in https://github.com/start-jsk/rtmros_hironx/issues/202#issuecomment-100402902. 
Ran on Ubuntu Trusty with ROS Indigo.

```
$ roscd nextage_description
$ rosrun collada_urdf collada_to_urdf models/main.dae -A --mesh_output_dir ./urdf/meshes --mesh_prefix 'package://nextage_description/urdf/meshes'
;; Use assimp export
;; mesh_prefix is: package://nextage_description/urdf/meshes
;; Mesh output directory is: ./urdf/meshes
;; Input file is: models/main.dae
[ WARN] [1431336983.379608621]: COLLADA warning: The DOM was unable to create an element named copyright at line 10. Probably a schema violation.

[ERROR] [1431336983.453536589]: Target Node visual1/node_joint0_axis0 NOT found!!!
[ WARN] [1431336983.457185252]: could not find binding for axis: kmodel1/jointsid1000/axis0, axis0
[ WARN] [1431336983.459746735]: could not find binding for axis: kmodel1/jointsid1000/axis0, axis0
[ WARN] [1431336983.539906089]: could not find openrave joint WAIST!

$ roscd collada_urdf
$ git branch
* indigo-devel
$ git log
commit 6534424a2ce67353db339059ee0c3be3272e897e
Merge: 04f7637 3c55217
Author: Ioan A Sucan <isucan@users.noreply.github.com>
Date:   Sat May 2 07:46:58 2015 -0700

    Merge pull request #108 from k-okada/fix_pcrecpp2
    
    link libpcrecpp before other libraries
```
#Once this error will be gone I'll improve CMakeLists.txt for auto re-generation.

130s, 2015-05-11:
`urdf/meshes/` folder was re-generated. And `HiroNx.urdf` file was generated at the current directory. So at least `collada_to_urdf` did generate something. 
Why a file was named as `HiroNx.urdf` is another question.

closed 2015-05-11 by 130s

k-okada, 2015-05-11:
> I'm trying to manually re-generate urdf based on the discussion in start-jsk/rtmros_hironx#202
> (comment)

even if it said error, but may be ok, please check if this generate urdf 
and the urdf is ok. 
the file name is dtermined from 
```

			<node id="visual1" name="HiroNX" sid="visual1">

```
of collada file