tork-a/rtmros_nextage #168: Unable to set camera info for calibration

130s, 2015-05-12:
By following the tutorial http://wiki.ros.org/rtmros_nextage/Tutorials/Setup%20stereo%20camera I got an error. Asked on http://answers.ros.org/question/209058/unable-to-set-camera-info-for-calibration/

130s, 2015-07-15:
Although got [an answer](http://answers.ros.org/question/209058/unable-to-set-camera-info-for-calibration/?answer=213198#post-id-213198) from the developer of ueye_cam, I don't have an access to the device and can't verify if the suggested solution works. But it seems valid so I close it for now. Feel free to reopen when this happens again.

closed 2015-07-15 by 130s

antgarmun, 2017-06-13:
I followed the same tutorial, I did the calibration process and I got the same error: [ERROR] [WallTime: 1497355764.396146] Unable to set camera info for calibration. Failure message: failed to write to file

I can't find information about how to solve this. Can you help me?

130s, 2017-06-13:
@antgarmun I think this issue is not specific to NEXTAGE Open software, but instead it's suitable to be discussed on camera driver's side. 
Follow the link to a question forum above, and if that won't help, then ask a new question there is my suggestion (be sure to tag `ueye_cam` so that people intested in the package get poked).

k-okada, 2017-06-14:
@antgarmun 
if you ave errors like http://answers.ros.org/question/209058/unable-to-set-camera-info-for-calibration/?answer=213198#post-id-213198
```
[ERROR] Unable to open camera calibration file [/home/rospasta/.ros/camera_info/left.yaml] for writing
```
please make sure that you have `${HOME}/.ros/camera_info` directory.

antgarmun, 2017-06-14:
Finally the problem was the directory. I created the ${HOME}/.ros/camera_info directory and all correct!

Thank you so much!