tork-a/rtmros_nextage #172: On real robot "[rtmlaunch] SIMULATOR_NAME RobotHardware0"

130s, 2015-05-19:
Why does `rtmlaunch` get started and why we're seeing "simulator" when we work with a real robot?
```
[rtmlaunch] SIMULATOR_NAME RobotHardware0
[rtmlaunch] check connection/activation
process[StateHolderServiceROSBridge-14]: started with pid [18321]
[rtmlaunch] Connected from nextage:15005/RobotHardware0.rtc:q
[rtmlaunch]             to nextage:15005/HrpsysSeqStateROSBridge0.rtc:rsangle
```

130s, 2015-05-19:
```
$ dpkg -p ros-hydro-nextage-ros-bridge ros-hydro-hrpsys-ros-bridge ros-hydro-hrpsys ros-hydro-openhrp3 |grep Ver
Version: 0.6.1-0precise-20150311-1752-+0000
Version: 1.2.8-0precise-20150311-1659-+0000
Version: 315.3.0-1precise-20150311-1614-+0000
Version: 3.1.7-3precise-20150213-1403-+0000
```

k-okada, 2015-05-19:
this just showing SIMULATOR_NAME environment variables 
https://github.com/start-jsk/rtmros_common/blob/master/openrtm_tools/src/openrtm_tools/rtmlaunch.py#L223
and used in 
https://github.com/start-jsk/rtmros_common/blob/master/hrpsys_tools/launch/hrpsys.launch#L118 
as RobotHardware name (On real robot it is RobotHardware but on simulaiton 
it is HiroNX(Robot)0 or something)

130s, 2015-07-10:
> On real robot it is RobotHardware but on simulaiton it is HiroNX(Robot)0 or something)

Understood, thank you.

closed 2015-07-10 by 130s