tork-a/rtmros_nextage #173: Could not start free-run live video mode on UEye camera 'right_hand_ueye' (IS_NO_SUCCESS)

130s, 2015-05-20:
On the same NXO robot where ueye camera on the right hand worked fine https://github.com/tork-a/rtmros_nextage/pull/162#issuecomment-101181101, now this error happens when the topic `/right_hand_ueye/image_raw` is tried being subscribed to.
```
$ roslaunch nextage_ros_bridge hands_ueye.launch 
:
SUMMARY
========

PARAMETERS
 * /left_hand_ueye_nodelet/auto_exposure
 * /left_hand_ueye_nodelet/auto_frame_rate
 * /left_hand_ueye_nodelet/auto_gain
 * /left_hand_ueye_nodelet/auto_white_balance
 * /left_hand_ueye_nodelet/binning
 * /left_hand_ueye_nodelet/blue_gain
 * /left_hand_ueye_nodelet/camera_id
 * /left_hand_ueye_nodelet/camera_intrinsics_file
 * /left_hand_ueye_nodelet/camera_name
 * /left_hand_ueye_nodelet/camera_parameters_file
 * /left_hand_ueye_nodelet/camera_topic
 * /left_hand_ueye_nodelet/color_mode
 * /left_hand_ueye_nodelet/exposure
 * /left_hand_ueye_nodelet/ext_trigger_mode
 * /left_hand_ueye_nodelet/flash_delay
 * /left_hand_ueye_nodelet/flash_duration
 * /left_hand_ueye_nodelet/frame_rate
 * /left_hand_ueye_nodelet/gain_boost
 * /left_hand_ueye_nodelet/green_gain
 * /left_hand_ueye_nodelet/image_height
 * /left_hand_ueye_nodelet/image_left
 * /left_hand_ueye_nodelet/image_top
 * /left_hand_ueye_nodelet/image_width
 * /left_hand_ueye_nodelet/master_gain
 * /left_hand_ueye_nodelet/pixel_clock
 * /left_hand_ueye_nodelet/red_gain
 * /left_hand_ueye_nodelet/sensor_scaling
 * /left_hand_ueye_nodelet/subsampling
 * /left_hand_ueye_nodelet/white_balance_blue_offset
 * /left_hand_ueye_nodelet/white_balance_red_offset
 * /right_hand_ueye_nodelet/auto_exposure
 * /right_hand_ueye_nodelet/auto_frame_rate
 * /right_hand_ueye_nodelet/auto_gain
 * /right_hand_ueye_nodelet/auto_white_balance
 * /right_hand_ueye_nodelet/binning
 * /right_hand_ueye_nodelet/blue_gain
 * /right_hand_ueye_nodelet/camera_id
 * /right_hand_ueye_nodelet/camera_intrinsics_file
 * /right_hand_ueye_nodelet/camera_name
 * /right_hand_ueye_nodelet/camera_parameters_file
 * /right_hand_ueye_nodelet/camera_topic
 * /right_hand_ueye_nodelet/color_mode
 * /right_hand_ueye_nodelet/exposure
 * /right_hand_ueye_nodelet/ext_trigger_mode
 * /right_hand_ueye_nodelet/flash_delay
 * /right_hand_ueye_nodelet/flash_duration
 * /right_hand_ueye_nodelet/frame_rate
 * /right_hand_ueye_nodelet/gain_boost
 * /right_hand_ueye_nodelet/green_gain
 * /right_hand_ueye_nodelet/image_height
 * /right_hand_ueye_nodelet/image_left
 * /right_hand_ueye_nodelet/image_top
 * /right_hand_ueye_nodelet/image_width
 * /right_hand_ueye_nodelet/master_gain
 * /right_hand_ueye_nodelet/pixel_clock
 * /right_hand_ueye_nodelet/red_gain
 * /right_hand_ueye_nodelet/sensor_scaling
 * /right_hand_ueye_nodelet/subsampling
 * /right_hand_ueye_nodelet/white_balance_blue_offset
 * /right_hand_ueye_nodelet/white_balance_red_offset
 * /rosdistro
 * /rosversion

NODES
  /right_hand_ueye/
    image_proc (image_proc/image_proc)
  /left_hand_ueye/
    image_proc (image_proc/image_proc)
  /
    hands_ueye_nodelet_manager (nodelet/nodelet)
    left_hand_ueye_nodelet (nodelet/nodelet)
    right_hand_ueye_nodelet (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [6750]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 65e0b5e4-fee4-11e4-9aa5-60a44caf77a8
process[rosout-1]: started with pid [6763]
started core service [/rosout]
process[hands_ueye_nodelet_manager-2]: started with pid [6775]
process[right_hand_ueye_nodelet-3]: started with pid [6776]
[ INFO] [1432121764.932069036]: Initializing nodelet with 8 worker threads.
process[left_hand_ueye_nodelet-4]: started with pid [6815]
[ INFO] [1432121765.040968971]: Unable to open camera calibration file [/home/rospasta/.ros/camera_info/right_hand_ueye.yaml]
process[right_hand_ueye/image_proc-5]: started with pid [6839]
process[left_hand_ueye/image_proc-6]: started with pid [6921]
[ WARN] [1432121766.386764563]: Could not load UEye camera 'right_hand_ueye' sensor parameters file /home/rospasta/.ros/camera_conf/right_hand_ueye.ini (IS_FILE_WRITE_OPEN_ERROR)
[ WARN] [1432121766.386823463]: Camera configuration loaded into an unsupported color mode; switching to MONO8.
[ WARN] [1432121766.407556133]: Internal image scaling is not supported by camera
[ INFO] [1432121766.431445966]: UEye camera 'right_hand_ueye' initialized on topic /right_hand_ueye/image_raw
Width:			752
Height:			480
Left Pos.:		-1
Top Pos.:		-1
Color Mode:		rgb8
Subsampling:		1
Binning:		1
Sensor Scaling:		1
Auto Gain:		1
Master Gain:		0
Red Gain:		0
Green Gain:		1
Blue Gain:		16
Gain Boost:		0
Auto Exposure:		0
Exposure (ms):		32.9818
Auto White Balance:	1
WB Red Offset:		0
WB Blue Offset:		0
Flash Delay (us):	30000
Flash Duration (us):	1000
Ext Trigger Mode:	0
Auto Frame Rate:	0
Frame Rate (Hz):	30.0321
Pixel Clock (MHz):	20
Mirror Image Upside Down:	0
Mirror Image Left Right:	0

[ INFO] [1432121766.478340004]: Unable to open camera calibration file [/home/rospasta/.ros/camera_info/left_hand_ueye.yaml]
[ERROR] [1432121766.706133310]: Could not open UEye camera ID 3 (IS_CANT_OPEN_DEVICE)
[ERROR] [1432121766.706192759]: Failed to initialize UEye camera 'left_hand_ueye'
[ERROR] [1432121832.303717666]: Could not start free-run live video mode on UEye camera 'right_hand_ueye' (IS_NO_SUCCESS)
[ERROR] [1432121832.303767123]: Shutting down UEye camera interface...
[hands_ueye_nodelet_manager-2] process has finished cleanly
log file: /home/rospasta/.ros/log/65e0b5e4-fee4-11e4-9aa5-60a44caf77a8/hands_ueye_nodelet_manager-2*.log
[right_hand_ueye_nodelet-3] process has finished cleanly
log file: /home/rospasta/.ros/log/65e0b5e4-fee4-11e4-9aa5-60a44caf77a8/right_hand_ueye_nodelet-3*.log
[left_hand_ueye_nodelet-4] process has finished cleanly
log file: /home/rospasta/.ros/log/65e0b5e4-fee4-11e4-9aa5-60a44caf77a8/left_hand_ueye_nodelet-4*.log
^C[left_hand_ueye/image_proc-6] killing on exit
[right_hand_ueye/image_proc-5] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

k-okada, 2015-05-20:
> On the same NXO robot where ueye camera on the right hand worked fine #162 (comment), now this error happens when the topic /right_hand_ueye/image_raw is tried being subscribed to.

> Could not open UEye camera ID 3 (IS_CANT_OPEN_DEVICE)

Please check camera is

130s, 2015-05-20:
`ueyecameramanager` (proprietary driver software) isn't receiving images either.

closed 2015-05-20 by 130s

130s, 2015-05-21:
- With ueye SDK 4.1 and kernel 3.13 (released this week) this issue occurred.
- With ueye SDK 4.4 and the same kernel version we started getting images again.

130s, 2016-01-15:
@RyuYamamoto FYI. Please try SDK version 4.4.

> With ueye SDK 4.4 and the same kernel version we started getting images again.