tork-a/rtmros_nextage #174: Image from hand camera not displayed on RViz

130s, 2015-05-21:
Image topics are received but RViz camera plugin says no image received.
```
$ rostopic hz /right_hand_ueye/image_raw 
subscribed to [/right_hand_ueye/image_raw]
average rate: 0.294
	min: 3.397s max: 3.397s std dev: 0.00000s window: 2
no new messages
average rate: 0.388
	min: 1.763s max: 3.397s std dev: 0.81670s window: 3
average rate: 0.609
	min: 0.561s max: 3.397s std dev: 1.10667s window: 5
average rate: 0.673
	min: 0.561s max: 3.397s std dev: 1.03712s window: 6
average rate: 0.725
	min: 0.561s max: 3.397s std dev: 0.97604s window: 7
^Caverage rate: 0.857
	min: 0.480s max: 3.397s std dev: 0.92278s window: 9
$ rostopic hz /right_hand_ueye/image_raw/compressed
subscribed to [/right_hand_ueye/image_raw/compressed]
average rate: 18.862
	min: 0.017s max: 0.104s std dev: 0.02188s window: 17
average rate: 16.666
	min: 0.017s max: 0.169s std dev: 0.03315s window: 31
average rate: 16.893
	min: 0.014s max: 0.169s std dev: 0.03296s window: 48
average rate: 19.188
	min: 0.000s max: 0.198s std dev: 0.03587s window: 75
^Caverage rate: 19.874
	min: 0.000s max: 0.198s std dev: 0.03437s window: 89
$ rostopic hz /right_hand_ueye/image_mono
subscribed to [/right_hand_ueye/image_mono]
average rate: 4.316
	min: 0.154s max: 0.309s std dev: 0.07721s window: 3
average rate: 3.166
	min: 0.154s max: 0.409s std dev: 0.10049s window: 5
average rate: 3.390
	min: 0.154s max: 0.409s std dev: 0.09118s window: 8
```

k-okada, 2015-05-21:
please try "image" plugin , not "camera" plugin
camera plugin requires both image and camera_info

130s, 2015-05-21:
That did it. Thank you!

closed 2015-05-21 by 130s

k-okada, 2015-05-21:
I think ar tools also depends on camera_info, so you need to publish them
to use ar pose