tork-a/rtmros_nextage #175: natto view is lost with the real robot

130s, 2015-05-21:
`For frame [WAIST_Link]: Frame [WAIST_Link] does not exist`

```
$ dpkg -p ros-indigo-nextage-ros-bridge ros-indigo-hironx-ros-bridge ros-indigo-hrpsys-ros-bridge ros-indigo-hrpsys ros-indigo-openhrp3 |grep Ver
Version: 0.6.2-0trusty-20150519-2228-+0000
Version: 1.0.30-0trusty-20150519-1246-+0000
Version: 1.2.12-0trusty-20150516-1626-+0000
Version: 315.4.0-0trusty-20150507-0829-+0000
Version: 3.1.8-0trusty-20150424-0521-+0000
```

k-okada, 2015-05-23:
this may because of https://github.com/tork-a/rtmros_nextage/pull/105

check
https://github.com/start-jsk/rtmros_hironx/blob/hydro-devel/hironx_ros_bridge/launch/hironx_ros_bridge.launch#L42
and
https://github.com/start-jsk/rtmros_common/blob/master/hrpsys_ros_bridge/scripts/collision_state.py#L63

you need to fix model fine robot computer ad tork-a/rtmros_nextage/pull/105 
or pass model in user'side model in collision_state.py or use ROS model 
(robot_description) for collision_state.py

k-okada, 2016-09-05:
please re-open if you still have this problem, I think if it works on hrpsys simulator, then it should work on real robot.

closed 2016-09-05 by k-okada