tork-a/rtmros_nextage #177: 実機が動かせない

RyuYamamoto, 2015-05-26:
実機にてプログラムを実行しようとしたところ，move itのpythonプログラム側のデバッグメッセージにABORTED: Timeout reachedと出ていて実機が動作しませんでした．以下に動かしたプログラムとその時のデバッグメッセージを冗長ですが添付します.

```roslaunch nextage_moveit_config moveit_planning_execution.launch``` 
> [ WARN] [1432629448.160562004]: The root link WAIST has an inertia specified in the URDF, but KDL does not support a root link with an inertia.  As a workaround, you can add an extra dummy link to your URDF.
[ WARN] [1432629448.160931786]: The root link WAIST has an inertia specified in the URDF, but KDL does not support a root link with an inertia.  As a workaround, you can add an extra dummy link to your URDF.
[ INFO] [1432629448.169248446]: Constructing new MoveGroup connection for group 'right_arm' in namespace ''
[ INFO] [1432629448.947314701]: Ready to take MoveGroup commands for group right_arm.
[ INFO] [1432629448.947748816]: Looking around: no
[ INFO] [1432629448.948477102]: Replanning: no
[ WARN] [1432629486.982809084]: No transform available between frame 'BODY' and planning frame '/WAIST' (Could not find a connection between 'WAIST' and 'BODY' because they are not part of the same tree.Tf has two or more unconnected trees.)
[ WARN] [1432629486.983050251]: No transform available between frame 'base_footprint' and planning frame '/WAIST' (Could not find a connection between 'WAIST' and 'base_footprint' because they are not part of the same tree.Tf has two or more unconnected trees.)
[ WARN] [1432629486.983352431]: No transform available between frame 'odom' and planning frame '/WAIST' (Could not find a connection between 'WAIST' and 'odom' because they are not part of the same tree.Tf has two or more unconnected trees.)
[ INFO] [1432629486.983643140]: Combined planning and execution request received for MoveGroup action. Forwarding to planning and execution pipeline.
[ INFO] [1432629486.985013201]: Planning attempt 1 of at most 1
[ INFO] [1432629487.004372558]: No planner specified. Using default.
[ INFO] [1432629487.005180598]: LBKPIECE1: Attempting to use default projection.
[ INFO] [1432629487.008145752]: LBKPIECE1: Starting with 1 states
[ INFO] [1432629488.149794234]: LBKPIECE1: Created 105 (47 start + 58 goal) states in 91 cells (45 start (45 on boundary) + 46 goal (46 on boundary))
[ INFO] [1432629488.149908252]: Solution found in 1.144098 seconds
[ INFO] [1432629488.155554627]: Path simplification took 0.005454 seconds
[ WARN] [1432629488.207055302]: No transform available between frame 'BODY' and planning frame '/WAIST' (Could not find a connection between 'WAIST' and 'BODY' because they are not part of the same tree.Tf has two or more unconnected trees.)
[ WARN] [1432629488.207162958]: No transform available between frame 'base_footprint' and planning frame '/WAIST' (Could not find a connection between 'WAIST' and 'base_footprint' because they are not part of the same tree.Tf has two or more unconnected trees.)
[ WARN] [1432629488.207250637]: No transform available between frame 'odom' and planning frame '/WAIST' (Could not find a connection between 'WAIST' and 'odom' because they are not part of the same tree.Tf has two or more unconnected trees.)
[ERROR] [1432629492.448162651]: Controller is taking too long to execute trajectory (the expected upper bound for the trajectory execution was 4.024098 seconds). Stopping trajectory.
[ INFO] [1432629492.448438973]: Cancelling execution for rarm_controller
[ WARN] [1432629493.591332107]: No transform available between frame 'BODY' and planning frame '/WAIST' (Could not find a connection between 'WAIST' and 'BODY' because they are not part of the same tree.Tf has two or more unconnected trees.)
[ WARN] [1432629493.591486891]: No transform available between frame 'base_footprint' and planning frame '/WAIST' (Could not find a connection between 'WAIST' and 'base_footprint' because they are not part of the same tree.Tf has two or more unconnected trees.)
[ WARN] [1432629493.591664470]: No transform available between frame 'odom' and planning frame '/WAIST' (Could not find a connection between 'WAIST' and 'odom' because they are not part of the same tree.Tf has two or more unconnected trees.)
[ INFO] [1432629493.591792200]: Combined planning and execution request received for MoveGroup action. Forwarding to planning and execution pipeline.
[ INFO] [1432629493.592133290]: Planning attempt 1 of at most 1
[ INFO] [1432629493.600464275]: No planner specified. Using default.
[ INFO] [1432629493.600591249]: LBKPIECE1: Attempting to use default projection.
[ INFO] [1432629493.603227282]: LBKPIECE1: Starting with 1 states
[ INFO] [1432629494.063985136]: LBKPIECE1: Created 48 (12 start + 36 goal) states in 36 cells (10 start (10 on boundary) + 26 goal (26 on boundary))
[ INFO] [1432629494.064089160]: Solution found in 0.463341 seconds
[ INFO] [1432629494.069704393]: Path simplification took 0.005504 seconds
[ WARN] [1432629494.111956799]: No transform available between frame 'BODY' and planning frame '/WAIST' (Could not find a connection between 'WAIST' and 'BODY' because they are not part of the same tree.Tf has two or more unconnected trees.)
[ WARN] [1432629494.112084585]: No transform available between frame 'base_footprint' and planning frame '/WAIST' (Could not find a connection between 'WAIST' and 'base_footprint' because they are not part of the same tree.Tf has two or more unconnected trees.)
[ WARN] [1432629494.112161812]: No transform available between frame 'odom' and planning frame '/WAIST' (Could not find a connection between 'WAIST' and 'odom' because they are not part of the same tree.Tf has two or more unconnected trees.)
[ERROR] [1432629498.357926602]: Controller is taking too long to execute trajectory (the expected upper bound for the trajectory execution was 4.024754 seconds). Stopping trajectory.
[ INFO] [1432629498.358140347]: Cancelling execution for larm_controller
[ WARN] [1432629499.504703284]: No transform available between frame 'BODY' and planning frame '/WAIST' (Could not find a connection between 'WAIST' and 'BODY' because they are not part of the same tree.Tf has two or more unconnected trees.)
[ WARN] [1432629499.504844590]: No transform available between frame 'base_footprint' and planning frame '/WAIST' (Could not find a connection between 'WAIST' and 'base_footprint' because they are not part of the same tree.Tf has two or more unconnected trees.)
[ WARN] [1432629499.505017935]: No transform available between frame 'odom' and planning frame '/WAIST' (Could not find a connection between 'WAIST' and 'odom' because they are not part of the same tree.Tf has two or more unconnected trees.)
[ INFO] [1432629499.505137254]: Combined planning and execution request received for MoveGroup action. Forwarding to planning and execution pipeline.
[ INFO] [1432629499.505475093]: Planning attempt 1 of at most 1
[ INFO] [1432629499.515658725]: LBKPIECE1: Starting with 1 states
[ INFO] [1432629500.350831875]: LBKPIECE1: Created 95 (33 start + 62 goal) states in 81 cells (30 start (30 on boundary) + 51 goal (51 on boundary))
[ INFO] [1432629500.351043467]: Solution found in 0.837720 seconds
[ INFO] [1432629500.357247218]: Path simplification took 0.006008 seconds
[ WARN] [1432629500.420898203]: No transform available between frame 'BODY' and planning frame '/WAIST' (Could not find a connection between 'WAIST' and 'BODY' because they are not part of the same tree.Tf has two or more unconnected trees.)
[ WARN] [1432629500.421098748]: No transform available between frame 'base_footprint' and planning frame '/WAIST' (Could not find a connection between 'WAIST' and 'base_footprint' because they are not part of the same tree.Tf has two or more unconnected trees.)
[ WARN] [1432629500.421229986]: No transform available between frame 'odom' and planning frame '/WAIST' (Could not find a connection between 'WAIST' and 'odom' because they are not part of the same tree.Tf has two or more unconnected trees.)
[ERROR] [1432629504.801996244]: Controller is taking too long to execute trajectory (the expected upper bound for the trajectory execution was 4.404246 seconds). Stopping trajectory.
[ INFO] [1432629504.802186220]: Cancelling execution for rarm_controller
[ WARN] [1432629506.582994094]: No transform available between frame 'BODY' and planning frame '/WAIST' (Could not find a connection between 'WAIST' and 'BODY' because they are not part of the same tree.Tf has two or more unconnected trees.)
[... cut ...]
  w: 0.706656500482
==========  Printing Left Hand initial pose: 
position: 
  x: 0.325556524492
  y: 0.182431716648
  z: 0.0676662591281
orientation: 
  x: -0.000473123916785
  y: -0.708087381263
  z: -3.77361922096e-05
  w: 0.706124659834
==========  plan1 ...
[ INFO] [1432629492.589286292]: ABORTED: Timeout reached
==========  plan2 ...
[ INFO] [1432629498.502715261]: ABORTED: Timeout reached
==========  plan3...
[ INFO] [1432629504.998800322]: ABORTED: Timeout reached
^C========== Initial pose ...

k-okada, 2015-05-26:
> [hrpsys.py] create Comp -> ImpedanceController :  (315.2.8)
> [hrpsys.py] create Comp -> SoftErrorLimiter : (315.2.8)
> [hrpsys.py] create Comp -> ServoController : (315.2.8)
> [hrpsys.py] create Comp -> DataLogger : (315.2.8)
> Exception AttributeError: "objref_PortService instance has no attribute 'Object__release'" in <bound method _objref_PortService.__del of > ignored

https://github.com/tork-a/rtmros_nextage/issues/170

はやられていますのでしょうか
--
◉ Kei Okada

130s, 2015-10-19:
> #170 はやられていますのでしょうか

Looking at the log, the default planner is chosen, so most likely the workaround in #170 isn't active.

>> [ INFO] [1432629487.004372558]: No planner specified. Using default.
>> [ INFO] [1432629487.005180598]: LBKPIECE1: Attempting to use default projection.

So I'll close this for now. Please re-open if workaround in #170 doesn't work.

closed 2015-10-19 by 130s