tork-a/rtmros_nextage #178: ikfast with nxo

130s, 2015-06-15:
I'm just wondering if there is any reason why nextage_moveit_config does not use ikfast? 
@k-okada

k-okada, 2015-06-15:
no

emijah, 2015-06-16:
I've tried generating the IK using ikfast recently, but it was unsuccessful. So I'm posting here.
I used the main.dae model distributed in nextage_description with the following command.

`/opt/openrave/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/ikfast.py --robot=/opt/ros/hydro/share/nextage_description/models/main.dae --iktype=transform6d --baselink=1 --eelink=9 --savefile=nextage-right.cpp`

The output from this was

`/opt/openrave/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/misc.py:17: RuntimeWarning: to-Python converter for boost::multi_array<float, 1ul, std::allocator<float> > already registered; second conversion method ignored.
  import openravepy_int
/opt/openrave/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/misc.py:17: RuntimeWarning: to-Python converter for boost::multi_array<float, 2ul, std::allocator<float> > already registered; second conversion method ignored.
  import openravepy_int
/opt/openrave/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/misc.py:17: RuntimeWarning: to-Python converter for boost::multi_array<float, 3ul, std::allocator<float> > already registered; second conversion method ignored.
  import openravepy_int
/opt/openrave/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/misc.py:17: RuntimeWarning: to-Python converter for boost::multi_array<double, 1ul, std::allocator<double> > already registered; second conversion method ignored.
  import openravepy_int
/opt/openrave/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/misc.py:17: RuntimeWarning: to-Python converter for boost::multi_array<double, 2ul, std::allocator<double> > already registered; second conversion method ignored.
  import openravepy_int
/opt/openrave/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/misc.py:17: RuntimeWarning: to-Python converter for boost::multi_array<double, 3ul, std::allocator<double> > already registered; second conversion method ignored.
  import openravepy_int
/opt/openrave/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/misc.py:17: RuntimeWarning: to-Python converter for boost::multi_array<int, 1ul, std::allocator<int> > already registered; second conversion method ignored.
  import openravepy_int
/opt/openrave/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/misc.py:17: RuntimeWarning: to-Python converter for boost::multi_array<int, 2ul, std::allocator<int> > already registered; second conversion method ignored.
  import openravepy_int
/opt/openrave/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/misc.py:17: RuntimeWarning: to-Python converter for openravepy::PyVoidHandle already registered; second conversion method ignored.
  import openravepy_int
/opt/openrave/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/misc.py:17: RuntimeWarning: to-Python converter for boost::shared_ptr<openravepy::PyVoidHandle> already registered; second conversion method ignored.
  import openravepy_int
/opt/openrave/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/misc.py:17: RuntimeWarning: to-Python converter for OpenRAVE::openrave_exception already registered; second conversion method ignored.
  import openravepy_int
/opt/openrave/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/misc.py:17: RuntimeWarning: to-Python converter for std::runtime_error already registered; second conversion method ignored.
  import openravepy_int
/opt/openrave/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/misc.py:17: RuntimeWarning: to-Python converter for boost::bad_function_call already registered; second conversion method ignored.
  import openravepy_int
/opt/openrave/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/misc.py:17: RuntimeWarning: to-Python converter for openravepy::PyEnvironmentBase already registered; second conversion method ignored.
  import openravepy_int
/opt/openrave/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/misc.py:17: RuntimeWarning: to-Python converter for boost::shared_ptr<openravepy::PyEnvironmentBase> already registered; second conversion method ignored.
  import openravepy_int
/opt/openrave/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/misc.py:17: RuntimeWarning: to-Python converter for openravepy::PyInterfaceBase already registered; second conversion method ignored.
  import openravepy_int
/opt/openrave/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/misc.py:17: RuntimeWarning: to-Python converter for boost::shared_ptr<openravepy::PyInterfaceBase> already registered; second conversion method ignored.
  import openravepy_int
/opt/openrave/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/misc.py:17: RuntimeWarning: to-Python converter for OpenRAVE::EnvironmentBase::SelectionOptions already registered; second conversion method ignored.
  import openravepy_int
/opt/openrave/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/misc.py:17: RuntimeWarning: to-Python converter for openravepy::DummyStruct already registered; second conversion method ignored.
  import openravepy_int
/opt/openrave/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/misc.py:17: RuntimeWarning: to-Python converter for OpenRAVE::OpenRAVEErrorCode already registered; second conversion method ignored.
  import openravepy_int
/opt/openrave/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/misc.py:17: RuntimeWarning: to-Python converter for OpenRAVE::DebugLevel already registered; second conversion method ignored.
  import openravepy_int
[... cut ...]
  File "/opt/openrave/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/ikfast.py", line 4667, in solveLiWoernleHiller
    A = zeros((len(neweqs_test),len(allmonoms)))
  File "/usr/local/lib/python2.7/dist-packages/sympy/matrices/dense.py", line 1229, in zeros
    return cls.zeros(r, c)
  File "/usr/local/lib/python2.7/dist-packages/sympy/matrices/dense.py", line 513, in zeros
    r = as_int(r)
  File "/usr/local/lib/python2.7/dist-packages/sympy/core/compatibility.py", line 389, in as_int
    raise ValueError('%s is not an integer' % n)
TypeError: not all arguments converted during string formatting`

emijah, 2015-06-16:
よく見るとsympyの問題でしたので、インタフェースを合わせて現在計算中です。

130s, 2015-06-16:
I saw a possibly relevant PR https://github.com/rdiankov/openrave/pull/351 so the issue might be already addressed. 
But just wondering; have you tried also on Ubuntu 14.04 with ROS Indigo?

emijah, 2015-06-16:
Actually, I just send that pull request.
However, the IK itself is taking ages to put together so there may be another problem.

emijah, 2015-07-10:
I've emailed Rosen and he says that some restraints he added recently may be the culprit. He's working on it in his spare time.

markusgft, 2015-08-17:
Hi all, are there by any chance any news on this issue?

k-okada, 2015-08-20:
see #187, please check if that works.

emijah, 2015-08-24:
@markusgft I've tested the newest code in the "production" line and it seems to be working.

130s, 2015-10-02:
Addressed by #187. Please not hesitate to reopen if this issue remains.

closed 2015-10-02 by 130s