tork-a/rtmros_nextage #179: Enable AR tag recognition

130s, 2015-06-16:
@RyuYamamoto is working on this.

- [ ] Use http://wiki.ros.org/rtmros_nextage/Tutorials/Setup%20stereo%20camera#Cameras_at_hand_tip to run cameras at handtip. Output1: On rqt_image_view (or similar tool) you can see camera image topic. Output2: Fill the blank topic names on the wiki
- [ ] Create a launch file under nextage_ros_bridge/launch that passes the topics above to ar_track_alvar pkg. Don't forget to work on a branch. Output1: nextage_ros_bridge/launch/ar.launch
- [x] Print AR markers http://wiki.ros.org/ar_track_alvar?action=AttachFile&do=get&target=markers0to8.png --> I asked someone.
- [ ] Show any of the printed AR marker to either hand camera. Output1: Take a screenshot of the treee on `rqt_tf_tree` that shows the marker.

RyuYamamoto, 2015-06-16:
作業しようとした途中ros_bridgeを立ち上げたら以下のようなエラーで接続ができなくなりました.

... logging to /home/marsryu/.ros/log/e715b862-13cf-11e5-a01b-100ba90b744c/roslaunch-marsryu-CF-S10EWHDS-23591.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://marsryu-CF-S10EWHDS:33655/

SUMMARY
========-

PARAMETERS
 * /HrpsysSeqStateROSBridge/publish_sensor_transforms
 * /collision_state/comp_name
 * /controller_configuration
 * /diagnostic_aggregator/analyzers/computers/contains
 * /diagnostic_aggregator/analyzers/computers/path
 * /diagnostic_aggregator/analyzers/computers/type
 * /diagnostic_aggregator/analyzers/hrpsys/contains
 * /diagnostic_aggregator/analyzers/hrpsys/path
 * /diagnostic_aggregator/analyzers/hrpsys/type
 * /diagnostic_aggregator/analyzers/mode/contains
 * /diagnostic_aggregator/analyzers/mode/path
 * /diagnostic_aggregator/analyzers/mode/type
 * /diagnostic_aggregator/analyzers/motor/contains
 * /diagnostic_aggregator/analyzers/motor/path
 * /diagnostic_aggregator/analyzers/motor/type
 * /diagnostic_aggregator/base_path
 * /diagnostic_aggregator/pub_rate
 * /robot_description
 * /robot_pose_ekf/debug
 * /robot_pose_ekf/freq
 * /robot_pose_ekf/imu_used
 * /robot_pose_ekf/odom_used
 * /robot_pose_ekf/output_frame
 * /robot_pose_ekf/self_diagnose
 * /robot_pose_ekf/sensor_timeout
 * /robot_pose_ekf/vo_used
 * /rosdistro
 * /rosversion

NODES
  /
    CollisionDetectorComp (hrpsys/CollisionDetectorComp)
    DataLoggerServiceROSBridge (hrpsys_ros_bridge/DataLoggerServiceROSBridgeComp)
    ForwardKinematicsServiceROSBridge (hrpsys_ros_bridge/ForwardKinematicsServiceROSBridgeComp)
    HrpsysJointTrajectoryBridge (hrpsys_ros_bridge/HrpsysJointTrajectoryBridge)
    HrpsysSeqStateROSBridge (hrpsys_ros_bridge/HrpsysSeqStateROSBridge)
    RobotHardwareServiceROSBridge (hrpsys_ros_bridge/RobotHardwareServiceROSBridgeComp)
    SequencePlayerServiceROSBridge (hrpsys_ros_bridge/SequencePlayerServiceROSBridgeComp)
    StateHolderServiceROSBridge (hrpsys_ros_bridge/StateHolderServiceROSBridgeComp)
    base_footprint_rootlink (tf/static_transform_publisher)
    collision_state (hrpsys_ros_bridge/collision_state.py)
    diagnostic_aggregator (diagnostic_aggregator/aggregator_node)
    hrpsys_profile (hrpsys_ros_bridge/hrpsys_profile.py)
    hrpsys_py (hironx_ros_bridge/hironx.py)
    hrpsys_ros_diagnostics (hrpsys_ros_bridge/diagnostics.py)
    hrpsys_state_publisher (robot_state_publisher/state_publisher)
    robot_pose_ekf (robot_pose_ekf/robot_pose_ekf)
    rtmlaunch_collision_detector (openrtm_tools/rtmlaunch.py)
    rtmlaunch_hrpsys_ros_bridge (openrtm_tools/rtmlaunch.py)
    sensor_ros_bridge_connect (hrpsys_ros_bridge/sensor_ros_bridge_connect.py)

WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
[... 74 lines cut ...]
[rtmlaunch] starting...  /opt/ros/hydro/share/hrpsys_ros_bridge/launch/collision_detector.launch
[rtmlaunch] RTCTREE_NAMESERVERS nextage:15005 nextage:15005
[rtmlaunch] SIMULATOR_NAME RobotHardware0
[rtmlaunch] check connection/activation
[rtmlaunch] [ERROR] Could not Connect ( /nextage:15005/RobotHardware0.rtc:q , /nextage:15005/CollisionDetector0.rtc:qCurrent ):  Invalid CORBA naming service: nextage:15005 
[rtmlaunch] Could not Activate ( nextage:15005/$(arg COLLISIONDETECTOR_INSTANCE_NAME).rtc ) :  Invalid CORBA naming service: nextage:15005 
configuration ORB with nextage:15005
[ERROR] Connection Failed with the Nameserver (hostname=nextage port=15005).
Make sure the hostname is correct and the Nameserver is running.
CORBA.TRANSIENT(omniORB.TRANSIENT_ConnectFailed, CORBA.COMPLETED_NO)
[collision_state-20] process has died [pid 24012, exit code 1, cmd /opt/ros/hydro/share/hrpsys_ros_bridge/scripts/collision_state.py /opt/jsk/etc/NEXTAGE/model/main.wrl -ORBInitRef NameService=corbaloc:iiop:nextage:15005/NameService __name:=collision_state __log:=/home/marsryu/.ros/log/e715b862-13cf-11e5-a01b-100ba90b744c/collision_state-20.log].
log file: /home/marsryu/.ros/log/e715b862-13cf-11e5-a01b-100ba90b744c/collision_state-20*.log
[HrpsysSeqStateROSBridge-3] process has died [pid 23631, exit code -11, cmd /opt/ros/hydro/lib/hrpsys_ros_bridge/HrpsysSeqStateROSBridge -o corba.nameservers:nextage:15005 -o naming.formats:%n.rtc -o exec_cxt.periodic.type:PeriodicExecutionContext -o exec_cxt.periodic.rate:2000 -o logger.file_name:/tmp/rtc%p.log -o model:/opt/jsk/etc/NEXTAGE/model/main.wrl -o example.HrpsysSeqStateROSBridge.config_file: __name:=HrpsysSeqStateROSBridge __log:=/home/marsryu/.ros/log/e715b862-13cf-11e5-a01b-100ba90b744c/HrpsysSeqStateROSBridge-3.log].
log file: /home/marsryu/.ros/log/e715b862-13cf-11e5-a01b-100ba90b744c/HrpsysSeqStateROSBridge-3*.log
[HrpsysJointTrajectoryBridge-4] process has died [pid 23659, exit code -11, cmd /opt/ros/hydro/lib/hrpsys_ros_bridge/HrpsysJointTrajectoryBridge -o corba.nameservers:nextage:15005 -o naming.formats:%n.rtc -o exec_cxt.periodic.type:PeriodicExecutionContext -o exec_cxt.periodic.rate:2000 -o logger.file_name:/tmp/rtc%p.log -o model:/opt/jsk/etc/NEXTAGE/model/main.wrl __name:=HrpsysJointTrajectoryBridge __log:=/home/marsryu/.ros/log/e715b862-13cf-11e5-a01b-100ba90b744c/HrpsysJointTrajectoryBridge-4.log].
[... cut ...]
[rtmlaunch] [ERROR] Could not Connect ( /nextage:15005/RobotHardware0.rtc:q , /nextage:15005/CollisionDetector0.rtc:qCurrent ):  Invalid CORBA naming service: nextage:15005 
[rtmlaunch] Could not Activate ( nextage:15005/$(arg COLLISIONDETECTOR_INSTANCE_NAME).rtc ) :  Invalid CORBA naming service: nextage:15005 
[rtmlaunch] check connection/activation
[rtmlaunch] [ERROR] Could not Connect ( /nextage:15005/RobotHardware0.rtc:q , /nextage:15005/HrpsysSeqStateROSBridge0.rtc:rsangle ):  Invalid CORBA naming service: nextage:15005 
[rtmlaunch] Could not Activate ( nextage:15005/HrpsysSeqStateROSBridge0.rtc ) :  Invalid CORBA naming service: nextage:15005

130s, 2015-06-16:
Aside from the real issue, this particular task doesn't require ros_bridge to be running, until you want to work with `tf`.

RyuYamamoto, 2015-06-16:
> roslaunch nextage_ros_bridge hands_ueye.launch 
で手先カメラの画像が見れることをrqtで確認しました．

wikiの方も空欄になっているtopicを以下のようにしました.
> rostopic echo /left_hand_ueye/image_raw
> rostopic echo /right_hand_ueye/image_raw

130s, 2015-06-16:
The issue reported in https://github.com/tork-a/rtmros_nextage/issues/179#issuecomment-112263834 is not generic; turned out it's only due to the specific lab's LAN environment.

130s, 2015-06-16:
> Add static tf publisher to combine the 2 tf trees: the one of the robot, and the one of cameras. See this line of this launch file as an example.

On this robot, I think hand-tip cameras can be statically attached to the tf frame of eef (I forgot the name). So the node you have to add would look something like (numbers and the frame names can be incorrect):

```
<node pkg="tf" type="static_transform_publisher" name="handcamera_l" args="0.080 0.120 0.150 0.0 1.5708 1.5708 /camera_l /larm_joint5 100" />
<node pkg="tf" type="static_transform_publisher" name="handcamera_r" args="0.080 0.120 0.150 0.0 1.5708 1.5708 /camera_r /rarm_joint5 100" />
```