tork-a/rtmros_nextage #181: Is the simulation working?

emijah, 2015-07-09:
I'm using a recently updated Ubuntu 12.04 with the latest ros-hydro packages and I seem to have lost the ability to run the simulation with 
```
prompt> roslaunch nextage_ros_bridge nextage_ros_bridge_simulation.launch
```
Has this command changed? Please tell me what I'm doing wrong. The list of packages I have installed are as follows.
--------------------------------
ii  python-rosdep                                     0.11.2-1                                                                rosdep package manager abstrction tool for ROS
ii  python-rosdistro                                  0.4.2-1                                                                 A tool to work with rosdistro files
ii  ros-hydro-actionlib                               1.10.3-0precise-20150329-2021-+0000                                     The actionlib stack provides a standardized interface for interfacing with preemptable tasks. Examples of this include moving the base to a target location, performing a laser scan and returning the resulting point cloud, detecting the handle of a door, etc.
ii  ros-hydro-actionlib-msgs                          1.10.6-0precise-20150327-1850-+0000                                     actionlib_msgs defines the common messages to interact with an action server and an action client. For full documentation of the actionlib API see the actionlib package.
ii  ros-hydro-actionlib-tutorials                     0.1.7-0precise-20150329-2025-+0000                                      The actionlib_tutorials package
ii  ros-hydro-amcl                                    1.11.16-0precise-20150501-0906-+0000                                    amcl is a probabilistic localization system for a robot moving in 2D. It implements the adaptive (or KLD-sampling) Monte Carlo localization approach (as described by Dieter Fox), which uses a particle filter to track the pose of a robot against a known map. This node is derived, with thanks, from Andrew Howard's excellent 'amcl' Player driver.
ii  ros-hydro-angles                                  1.9.9-0precise-20141231-2216-+0000                                      This package provides a set of simple math utilities to work with angles. The utilities cover simple things like normalizing an angle and conversion between degrees and radians. But even if you're trying to calculate things like the shortest angular distance between two joinst space positions of your robot, but the joint motion is constrained by joint limits, this package is what you need. The code in this packge is stable and well tested. There are no plans for major changes in the near future.
ii  ros-hydro-base-local-planner                      1.11.16-0precise-20150501-0923-+0000                                    This package provides implementations of the Trajectory Rollout and Dynamic Window approaches to local robot navigation on a plane. Given a plan to follow and a costmap, the controller produces velocity commands to send to a mobile base. This package supports both holonomic and non-holonomic robots, any robot footprint that can be represented as a convex polygon or circle, and exposes its configuration as ROS parameters that can be set in a launch file. This package's ROS wrapper adheres to the BaseLocalPlanner interface specified in the nav_core package.
ii  ros-hydro-bfl                                     0.7.0-3precise-20141231-2216-+0000                                      This package contains a recent version of the Bayesian Filtering Library (BFL), distributed by the Orocos Project. For stability reasons, this package is currently locked to revision 31655 (April 19, 2010), but this revision will be updated on a regular basis to the latest available BFL trunk. This ROS package does not modify BFL in any way, it simply provides a convenient way to download and compile the library, because BFL is not available from an OS package manager. This ROS package compiles BFL with the Boost library for matrix operations and random number generation.
ii  ros-hydro-bond                                    1.7.16-0precise-20150327-1852-+0000                                     A bond allows two processes, A and B, to know when the other has terminated, either cleanly or by crashing. The bond remains connected until it is either broken explicitly or until a heartbeat times out.
ii  ros-hydro-bond-core                               1.7.16-0precise-20150327-1902-+0000                                     A bond allows two processes, A and B, to know when the other has terminated, either cleanly or by crashing. The bond remains connected until it is either broken explicitly or until a heartbeat times out.
ii  ros-hydro-bondcpp                                 1.7.16-0precise-20150327-1858-+0000                                     C++ implementation of bond, a mechanism for checking when another process has terminated.
ii  ros-hydro-bondpy                                  1.7.16-0precise-20150327-1859-+0000                                     Python implementation of bond, a mechanism for checking when another process has terminated.
ii  ros-hydro-calibration-estimation                  0.10.13-0precise-20150331-0024-+0000                                    Runs an optimization to estimate the a robot's kinematic parameters. This package is a generic rewrite of pr2_calibration_estimation.
ii  ros-hydro-calibration-launch                      0.10.13-0precise-20150518-1128-+0000                                    This package contains a collection of launch files that can be helpful in configuring the calibration stack to run on your robot.
ii  ros-hydro-calibration-msgs                        0.10.13-0precise-20150327-1938-+0000                                    This package defines messages for storing calibration samples to be used in full robot calibration procedures. This package is still unstable. Expect the messages to change.
ii  ros-hydro-camera-calibration                      1.11.11-0precise-20150517-0844-+0000                                    camera_calibration allows easy calibration of monocular or stereo cameras using a checkerboard calibration target.
ii  ros-hydro-camera-calibration-parsers              1.11.5-0precise-20150514-1628-+0000                                     camera_calibration_parsers contains routines for reading and writing camera calibration parameters.
ii  ros-hydro-camera-info-manager                     1.11.5-0precise-20150514-1646-+0000                                     This package provides a C++ interface for camera calibration information. It provides CameraInfo, and handles SetCameraInfo service requests, saving and restoring the camera calibration data.
ii  ros-hydro-carrot-planner                          1.11.16-0precise-20150501-0930-+0000                                    This planner attempts to find a legal place to put a carrot for the robot to follow. It does this by moving back along the vector between the robot and the goal point.
ii  ros-hydro-catkin                                  0.5.90-0precise-20141231-1837-+0000                                     Low-level build system macros and infrastructure for ROS.
ii  ros-hydro-class-loader                            0.2.5-0precise-20141231-2224-+0000                                      The class_loader package is a ROS-independent package for loading plugins during runtime and the foundation of the higher level ROS &quot;pluginlib&quot; library. class_loader utilizes the host operating system's runtime loader to open runtime libraries (e.g. .so/.dll files), introspect the library for exported plugin classes, and allows users to instantiate objects of said exported classes without the explicit declaration (i.e. header file) for those classes.
ii  ros-hydro-clear-costmap-recovery                  1.11.16-0precise-20150501-0923-+0000                                    This package provides a recovery behavior for the navigation stack that attempts to clear space by reverting the costmaps used by the navigation stack to the static map outside of a given area.
ii  ros-hydro-cmake-modules                           0.3.3-0precise-20141231-2220-+0000                                      A common repository for CMake Modules which are not distributed with CMake but are commonly used by ROS packages.
[... cut ...]
ii  ros-hydro-xacro                                   1.8.6-0precise-20150328-1151-+0000                                      Xacro (XML Macros) Xacro is an XML macro language. With xacro, you can construct shorter and more readable XML files by using macros that expand to larger XML expressions.
ii  ros-hydro-xmlrpcpp                                1.10.12-0precise-20141231-2227-+0000                                    XmlRpc++ is a C++ implementation of the XML-RPC protocol. This version is heavily modified from the package available on SourceForge in order to support roscpp's threading model. As such, we are maintaining our own fork.

k-okada, 2015-07-09:
please provide error message.

--
◉ Kei Okada

On Thu, Jul 9, 2015 at 3:09 PM, Hajime SAITO <notifications@github.com> 
wrote:

> I'm using a recently updated Ubuntu 12.04 with the latest ros-hydro 
> packages and I seem to have lost the ability to run the simulation with 
>
> prompt> roslaunch nextage_ros_bridge nextage_ros_bridge_simulation.launch
>
> Has this command changed? Please tell me what I'm doing wrong. The list of 
> packages I have installed are as follows. 
>
> ii python-rosdep 0.11.2-1 rosdep package manager abstrction tool for ROS
> ii python-rosdistro 0.4.2-1 A tool to work with rosdistro files
> ii ros-hydro-actionlib 1.10.3-0precise-20150329-2021-+0000 The actionlib 
> stack provides a standardized interface for interfacing with preemptable 
> tasks. Examples of this include moving the base to a target location, 
> performing a laser scan and returning the resulting point cloud, detecting 
> the handle of a door, etc.
> ii ros-hydro-actionlib-msgs 1.10.6-0precise-20150327-1850-+0000 
> actionlib_msgs defines the common messages to interact with an action 
> server and an action client. For full documentation of the actionlib API 
> see the actionlib package.
> ii ros-hydro-actionlib-tutorials 0.1.7-0precise-20150329-2025-+0000 The 
> actionlib_tutorials package
> ii ros-hydro-amcl 1.11.16-0precise-20150501-0906-+0000 amcl is a 
> probabilistic localization system for a robot moving in 2D. It implements 
> the adaptive (or KLD-sampling) Monte Carlo localization approach (as 
> described by Dieter Fox), which uses a particle filter to track the pose of 
> a robot against a known map. This node is derived, with thanks, from Andrew 
> Howard's excellent 'amcl' Player driver.
> ii ros-hydro-angles 1.9.9-0precise-20141231-2216-+0000 This package 
> provides a set of simple math utilities to work with angles. The utilities 
> cover simple things like normalizing an angle and conversion between 
> degrees and radians. But even if you're trying to calculate things like the 
> shortest angular distance between two joinst space positions of your robot, 
> but the joint motion is constrained by joint limits, this package is what 
> you need. The code in this packge is stable and well tested. There are no 
> plans for major changes in the near future.
> ii ros-hydro-base-local-planner 1.11.16-0precise-20150501-0923-+0000 This 
> package provides implementations of the Trajectory Rollout and Dynamic 
> Window approaches to local robot navigation on a plane. Given a plan to 
> follow and a costmap, the controller produces velocity commands to send to 
> a mobile base. This package supports both holonomic and non-holonomic 
> robots, any robot footprint that can be represented as a convex polygon or 
> circle, and exposes its configuration as ROS parameters that can be set in 
> a launch file. This package's ROS wrapper adheres to the BaseLocalPlanner 
> interface specified in the nav_core package.
> ii ros-hydro-bfl 0.7.0-3precise-20141231-2216-+0000 This package contains 
> a recent version of the Bayesian Filtering Library (BFL), distributed by 
> the Orocos Project. For stability reasons, this package is currently locked 
> to revision 31655 (April 19, 2010), but this revision will be updated on a 
> regular basis to the latest available BFL trunk. This ROS package does not 
> modify BFL in any way, it simply provides a convenient way to download and 
> compile the library, because BFL is not available from an OS package 
> manager. This ROS package compiles BFL with the Boost library for matrix 
> operations and random number generation.
> ii ros-hydro-bond 1.7.16-0precise-20150327-1852-+0000 A bond allows two 
> processes, A and B, to know when the other has terminated, either cleanly 
> or by crashing. The bond remains connected until it is either broken 
> explicitly or until a heartbeat times out.
> ii ros-hydro-bond-core 1.7.16-0precise-20150327-1902-+0000 A bond allows 
> two processes, A and B, to know when the other has terminated, either 
> cleanly or by crashing. The bond remains connected until it is either 
> broken explicitly or until a heartbeat times out.
> ii ros-hydro-bondcpp 1.7.16-0precise-20150327-1858-+0000 C++ 
[... 919 lines cut ...]
> including using messages, servers, parameters, etc. These tutorials are 
> compatible with the nodes in roscpp_tutorial.
> ii ros-hydro-rosserial-arduino 0.5.6-0precise-20150330-2334-+0000 
> Libraries and examples for ROSserial usage on Arduino/AVR Platforms.
> ii ros-hydro-rosserial-client 0.5.6-0precise-20150330-2329-+0000 
> Generalized client side source for rosserial.
> ii ros-hydro-rosserial-leonardo-cmake 0.1.3-0precise-20150331-0841-+0000 
> CMake scripts for building Arduino Leonardo (atmega32u4) Firmwares
> ii ros-hydro-rosserial-msgs 0.5.6-0precise-20150326-2051-+0000 Messages 
> for automatic topic configuration using rosserial.
> ii ros-hydro-rosservice 1.10.12-0precise-20150330-0703-+0000 rosservice 
> contains the rosservice command-line tool for listing and querying ROS 
> Services. It also contains a Python library for retrieving information 
> about Services and dynamically invoking them. The Python library is 
> experimental and is for internal-use only.
> ii ros-hydro-rostest 1.10.12-0precise-20150328-1035-+0000 Integration test 
> suite based on roslaunch that is compatible with xUnit frameworks.
> ii ros-hydro-rostime 0.4.4-0precise-20141231-2227-+0000 Time and Duration 
> implementations for C++ libraries, including roscpp.
> ii ros-hydro-rostopic 1.10.12-0precise-20150328-1115-+0000 rostopic 
> contains the rostopic command-line tool for displaying debug information 
> about ROS Topics, including publishers, subscribers, publishing rate, and 
> ROS Messages. It also contains an experimental Python library for getting 
> information about and interacting with topics dynamically. This library is 
> for internal-use onl

130s, 2015-07-09:
For simulation you have to use `rtmlaunch`, not `roslaunch`. Please make sure you go through [the tutorial for simulation](http://wiki.ros.org/rtmros_nextage/Tutorials/Run%20simulation).

emijah, 2015-07-10:
The thing is, rtmlaunch doesn't seem to exist. When I type rtmlaunch on the command line, I get 
```
bash: /opt/ros/hydro/share/hrpsys_ros_bridge/scripts/rtmlaunch: No such file or directory
```
When I look in the directory, 
```
$ ls /opt/ros/hydro/share/hrpsys_ros_bridge/scripts/
collision_state.py              default_robot_nosim.launch.in         diagnostics.py            hrpsys_dashboard
controller_config_converter.py  default_robot_ros_bridge.launch.in    diagnostics.yaml          hrpsys_profile.py
default-robot-interface.l.in    default_robot_startup.launch.in       emergence_stop.py         motor_states.py
default_robot.launch.in         default_unstable_hrpsys_config.py.in  hrpsys-bridge-connect.py  sensor_ros_bridge_connect.py

```

So it seems like it doesn't exist. Or is it located somewhere else now...?
A quick search shows,

```
$ find /opt/ros/hydro/ -name 'rtmlaunch'
/opt/ros/hydro/bin/rtmlaunch
/opt/ros/hydro/share/openrtm_tools/scripts/rtmlaunch
```
A look in my .bashrc file shows that I have
```
alias rtmlaunch='`rospack find hrpsys_ros_bridge`/scripts/rtmlaunch'
```
After deleting this and another alias for rtmtest, I can run
```
$ rtmlaunch nextage_ros_bridge nextage_ros_bridge_simulation.launch
```
now works. I think the aliases need to be removed from the tutorial. How about you?

130s, 2015-07-10:
Thank you for the catch. I [updated the wiki](http://wiki.ros.org/action/info/rtmros_nextage/Tutorials/Run%20simulation?action=diff&rev2=4&rev1=2).

closed 2015-07-10 by 130s