tork-a/rtmros_nextage #182: Script nextage.py emits errors related "loading robot model 'HiroNX'"

emijah, 2015-07-10:
Seems to be one of those days...

After getting the simulator going, I tried to connect to it using the nextage.py script, but I get errors.
The exact command I tried is,
```
ipython -i `rospack find nextage_ros_bridge`/script/nextage.py
```
After this I get 
```
[ WARN] [1436498663.381013222]: COLLADA warning: The DOM was unable to create an element named copyright at line 10. Probably a schema violation.

[ERROR] [1436498663.460565989]: Target Node visual1/node_joint0_axis0 NOT found!!!

[ WARN] [1436498663.464479123]: could not find binding for axis: kmodel1/jointsid1000/axis0, axis0

[ WARN] [1436498663.467305983]: could not find binding for axis: kmodel1/jointsid1000/axis0, axis0

[ WARN] [1436498663.550656987, 381.509999999]: could not find openrave joint WAIST!

[ERROR] [1436498663.555955954, 381.509999999]: Link 'RHAND_JOINT0_Link' declared as part of group 'right_gripper' is not known to the URDF
[ERROR] [1436498663.556065274, 381.509999999]: Link 'RHAND_JOINT1_Link' declared as part of group 'right_gripper' is not known to the URDF
[ERROR] [1436498663.556150493, 381.509999999]: Link 'RHAND_JOINT2_Link' declared as part of group 'right_gripper' is not known to the URDF
[ERROR] [1436498663.556225175, 381.509999999]: Link 'RHAND_JOINT3_Link' declared as part of group 'right_gripper' is not known to the URDF
[ WARN] [1436498663.556299044, 381.509999999]: Group 'right_gripper' is empty.
[ERROR] [1436498663.556375101, 381.509999999]: Link 'LHAND_JOINT0_Link' declared as part of group 'left_gripper' is not known to the URDF
[ERROR] [1436498663.556448279, 381.509999999]: Link 'LHAND_JOINT1_Link' declared as part of group 'left_gripper' is not known to the URDF
[ERROR] [1436498663.556520394, 381.509999999]: Link 'LHAND_JOINT2_Link' declared as part of group 'left_gripper' is not known to the URDF
[ERROR] [1436498663.556591339, 381.509999999]: Link 'LHAND_JOINT3_Link' declared as part of group 'left_gripper' is not known to the URDF
[ WARN] [1436498663.556660233, 381.509999999]: Group 'left_gripper' is empty.
[ERROR] [1436498663.556807872, 381.509999999]: Link 'LHAND_JOINT0_Link' is not known to URDF. Cannot disable collisons.
[ERROR] [1436498663.556882752, 381.509999999]: Link 'LHAND_JOINT1_Link' is not known to URDF. Cannot disable collisons.
[ERROR] [1436498663.556968415, 381.509999999]: Link 'LHAND_JOINT2_Link' is not known to URDF. Cannot disable collisons.
[ERROR] [1436498663.557039422, 381.509999999]: Link 'LHAND_JOINT3_Link' is not known to URDF. Cannot disable collisons.
[ERROR] [1436498663.557119065, 381.509999999]: Link 'RHAND_JOINT0_Link' is not known to URDF. Cannot disable collisons.
[ERROR] [1436498663.557190336, 381.509999999]: Link 'RHAND_JOINT1_Link' is not known to URDF. Cannot disable collisons.
[ERROR] [1436498663.557261781, 381.509999999]: Link 'RHAND_JOINT2_Link' is not known to URDF. Cannot disable collisons.
[ERROR] [1436498663.557332880, 381.509999999]: Link 'RHAND_JOINT3_Link' is not known to URDF. Cannot disable collisons.
[ERROR] [1436498663.557415003, 381.509999999]: Link 'LHAND_JOINT0_Link' is not known to URDF. Cannot disable collisons.
[ERROR] [1436498663.557485469, 381.509999999]: Link 'LHAND_JOINT1_Link' is not known to URDF. Cannot disable collisons.
[ERROR] [1436498663.557556236, 381.509999999]: Link 'LHAND_JOINT2_Link' is not known to URDF. Cannot disable collisons.
[ERROR] [1436498663.557626437, 381.509999999]: Link 'LHAND_JOINT3_Link' is not known to URDF. Cannot disable collisons.
[ERROR] [1436498663.557716518, 381.509999999]: Link 'LHAND_JOINT0_Link' is not known to URDF. Cannot disable collisons.
[ERROR] [1436498663.557787357, 381.509999999]: Link 'LHAND_JOINT1_Link' is not known to URDF. Cannot disable collisons.
[ERROR] [1436498663.557858672, 381.509999999]: Link 'LHAND_JOINT2_Link' is not known to URDF. Cannot disable collisons.
[ERROR] [1436498663.557929533, 381.509999999]: Link 'LHAND_JOINT3_Link' is not known to URDF. Cannot disable collisons.
[ERROR] [1436498663.558003302, 381.509999999]: Link 'LHAND_JOINT0_Link' is not known to URDF. Cannot disable collisons.
[ERROR] [1436498663.558073426, 381.509999999]: Link 'LHAND_JOINT1_Link' is not known to URDF. Cannot disable collisons.
[ERROR] [1436498663.558144233, 381.509999999]: Link 'LHAND_JOINT2_Link' is not known to URDF. Cannot disable collisons.
[ERROR] [1436498663.558214700, 381.509999999]: Link 'LHAND_JOINT3_Link' is not known to URDF. Cannot disable collisons.
[ERROR] [1436498663.558285504, 381.509999999]: Link 'LHAND_JOINT0_Link' is not known to URDF. Cannot disable collisons.
[ERROR] [1436498663.558355969, 381.509999999]: Link 'LHAND_JOINT1_Link' is not known to URDF. Cannot disable collisons.
[ERROR] [1436498663.558425278, 381.509999999]: Link 'LHAND_JOINT2_Link' is not known to URDF. Cannot disable collisons.
[ERROR] [1436498663.558495144, 381.509999999]: Link 'LHAND_JOINT3_Link' is not known to URDF. Cannot disable collisons.
[ERROR] [1436498663.558564774, 381.509999999]: Link 'LHAND_JOINT0_Link' is not known to URDF. Cannot disable collisons.
[ERROR] [1436498663.558633229, 381.509999999]: Link 'LHAND_JOINT0_Link' is not known to URDF. Cannot disable collisons.
[ERROR] [1436498663.558702063, 381.509999999]: Link 'LHAND_JOINT0_Link' is not known to URDF. Cannot disable collisons.
[ERROR] [1436498663.558770640, 381.509999999]: Link 'LHAND_JOINT0_Link' is not known to URDF. Cannot disable collisons.
[ERROR] [1436498663.558839358, 381.509999999]: Link 'LHAND_JOINT0_Link' is not known to URDF. Cannot disable collisons.
[ERROR] [1436498663.558909051, 381.509999999]: Link 'LHAND_JOINT0_Link' is not known to URDF. Cannot disable collisons.
[ERROR] [1436498663.558978035, 381.509999999]: Link 'LHAND_JOINT1_Link' is not known to URDF. Cannot disable collisons.
[ERROR] [1436498663.559047536, 381.509999999]: Link 'LHAND_JOINT1_Link' is not known to URDF. Cannot disable collisons.
[ERROR] [1436498663.559117396, 381.509999999]: Link 'LHAND_JOINT2_Link' is not known to URDF. Cannot disable collisons.
[ERROR] [1436498663.559187639, 381.509999999]: Link 'LHAND_JOINT2_Link' is not known to URDF. Cannot disable collisons.
[ERROR] [1436498663.559257496, 381.509999999]: Link 'LHAND_JOINT3_Link' is not known to URDF. Cannot disable collisons.
[ERROR] [1436498663.559326179, 381.509999999]: Link 'LHAND_JOINT3_Link' is not known to URDF. Cannot disable collisons.
[ERROR] [1436498663.559412020, 381.509999999]: Link 'RHAND_JOINT0_Link' is not known to URDF. Cannot disable collisons.
[ERROR] [1436498663.559482413, 381.509999999]: Link 'RHAND_JOINT1_Link' is not known to URDF. Cannot disable collisons.
[ERROR] [1436498663.559551049, 381.509999999]: Link 'RHAND_JOINT2_Link' is not known to URDF. Cannot disable collisons.
[ERROR] [1436498663.559619837, 381.509999999]: Link 'RHAND_JOINT3_Link' is not known to URDF. Cannot disable collisons.
[ERROR] [1436498663.559698778, 381.509999999]: Link 'RHAND_JOINT0_Link' is not known to URDF. Cannot disable collisons.
[... 12 lines cut ...]
[ERROR] [1436498663.560658401, 381.509999999]: Link 'RHAND_JOINT0_Link' is not known to URDF. Cannot disable collisons.
[ERROR] [1436498663.560727372, 381.509999999]: Link 'RHAND_JOINT0_Link' is not known to URDF. Cannot disable collisons.
[ERROR] [1436498663.560798850, 381.509999999]: Link 'RHAND_JOINT1_Link' is not known to URDF. Cannot disable collisons.
[ERROR] [1436498663.560994738, 381.514999999]: Link 'RHAND_JOINT2_Link' is not known to URDF. Cannot disable collisons.
[ INFO] [1436498663.561183871, 381.514999999]: Loading robot model 'HiroNX'...
[ WARN] [1436498663.593298824, 381.554999999]: Group 'right_gripper' must have at least one valid joint
[ WARN] [1436498663.593501186, 381.554999999]: Failed to add group 'right_gripper'
[ WARN] [1436498663.593587790, 381.554999999]: Group 'left_gripper' must have at least one valid joint
[ WARN] [1436498663.593664076, 381.554999999]: Failed to add group 'left_gripper'

[ERROR] [WallTime: 1436498668.669839] [381.230000] Unable to connect to move_group action server within allotted time (2)
Make sure you've launched MoveGroup (e.g. by launching moveit_planning_execution.launch)
```
After I get the ipython prompt back, I can issue commands like,
```
In [1]: robot
Out[1]: <nextage_ros_bridge.nextage_client.NextageClient at 0x540aa10>

In [2]: robot.getCurrentPosition('LARM_JOINT5')
Out[2]: [0.3255751432930864, 0.1823632083754606, 0.06762475210283836]

In [3]: robot.getCurrentPosition('RARM_JOINT5')
Out[3]: [0.3255751432930864, -0.1823632083754606, 0.06762475210283836]
```
The Z position seems consistent with Hiro.

130s, 2015-07-10:
I can't reproduce the issue on my Indigo machine. Could you:

- send the full list of commands you ran?
- run the following script and send back the generated tarball to support[atto]opensource-robotics.tokyo.jp so that we can get better idea of your system?

```
curl -sL https://raw.githubusercontent.com/130s/rtmros_common/impr/rtmroswtf/hrpsys_tools/scripts/rtmroswtf.sh | bash -
```

emijah, 2015-07-10:
The full list of commands would be 
```
rtmlaunch nextage_ros_bridge nextage_ros_bridge_simulation.launch
```
and then in another terminal
```
ipython -i `rospack find nextage_ros_bridge`/script/nextage.py
```
and after getting all the error messages I did the following and found that despite the errors, the system seemed to be responsive.
```
In [1]: robot
Out[1]: <nextage_ros_bridge.nextage_client.NextageClient at 0x540aa10>

In [2]: robot.getCurrentPosition('LARM_JOINT5')
Out[2]: [0.3255751432930864, 0.1823632083754606, 0.06762475210283836]

In [3]: robot.getCurrentPosition('RARM_JOINT5')
Out[3]: [0.3255751432930864, -0.1823632083754606, 0.06762475210283836]
```
I'll send you the tarball in a few moments.

k-okada, 2016-09-05:
no information updated recently, please re-open if you still have trouble

closed 2016-09-05 by k-okada