tork-a/rtmros_nextage #183: tf tree is sparse on simulation on Indigo

130s, 2015-08-01:
[image]

```
term-1$ rtmlaunch nextage_ros_bridge nextage_ros_bridge_simulation.launch 
term-2$ roslaunch nextage_moveit_config moveit_planning_execution.launch 

$ rospack find nextage_moveit_config 
/opt/ros/indigo/share/nextage_moveit_config
$ dpkg -p ros-indigo-robot-model ros-indigo-nextage-moveit-config |grep Ver
Version: 1.11.7-0trusty-20150523-1006-+0000
Version: 0.6.2-0trusty-20150730-1059-+0000
$ dpkg -p ros-indigo-robot-model ros-indigo-hrpsys ros-indigo-hrpsys-ros-bridge ros-indigo-nextage-moveit-config |grep Ver
Version: 1.11.7-0trusty-20150523-1006-+0000
Version: 315.6.0-0trusty-20150716-0711-+0000
Version: 1.2.14-0trusty-20150730-0547-+0000
Version: 0.6.2-0trusty-20150730-1059-+0000
```

k-okada, 2015-08-01:
check if robot_state_publisher works

130s, 2015-10-16:
Can't reproduce the issue again. 

`$ rtmlaunch nextage_ros_bridge nextage_ros_bridge_simulation.launch`
[image]

closed 2015-10-16 by 130s