tork-a/rtmros_nextage #185: how to write continuous trajectory command

k-okada, 2015-08-04:
question from Yang-san @ waseda

k-okada, 2015-08-04:
you can use
```
 hcf.playPatternOfGroup('LARM', [[0.010,0.0,-1.745,-0.265,0.164,0.06],[0.010,-0.2,-2.045,-0.265,0.164,0.06],[0.010,-0.4,-2.245,-0.265,0.164,0.06],[0.010,-0.6,-2.445,-0.265,0.164,0.06],[0.010,-0.8,-2.645,-0.265,0.164,0.06]],[1,1,1,1,1])
```

from hrpsys 0.6.0, you can use setJointAnglesSequenceOfGroup https://github.com/fkanehiro/hrpsys-base/blob/master/python/hrpsys_config.py#L1027

130s, 2015-10-26:
This is added to the [tutorial](http://wiki.ros.org/action/info/rtmros_nextage/Tutorials/Programming_Hiro_NEXTAGE_OPEN_RTM?action=diff&rev2=13&rev1=12).

> from hrpsys 0.6.0, you can use setJointAnglesSequenceOfGroup

I assume that is hrpsys 315.6.0.

closed 2015-10-26 by 130s