tork-a/rtmros_nextage #188: About real time joystick control

kuochiyoug, 2015-09-18:
Hi! I am now doing work on joystick controlling on Nextage Open Robot.
I had tried use moveit_commander in python code but it not works very well.
Seems moveit only provide target pose solution but no trajectory speed control.
I can only do is keep sending following command in every loop:
{{{
while not rospy.is_shutdown():	
  [rarm_current_pose,larm_current_pose] = self.GetAllCurrentPose()
  target_pose_right.position.x = rarm_current_pose.position.x + self.mouse.x
  target_pose_right.position.y = rarm_current_pose.position.y + self.mouse.y
  target_pose_right.position.z = rarm_current_pose.position.z + self.mouse.z
  target_pose_right.orientation.x = rarm_current_pose.orientation.x
  target_pose_right.orientation.y = rarm_current_pose.orientation.y
  target_pose_right.orientation.z = rarm_current_pose.orientation.z
  target_pose_right.orientation.w = rarm_current_pose.orientation.w
  rarm.set_pose_target(target_pose_right)
  rarm.go(wait = True)
}}}
I actually want a smooth movement so that I can do my following research.
Can anyone give me a good advice?

Thanks you very much!

k-okada, 2015-09-25:
if you'd like to use realtime-control, you'd better to call via ros/hrpsys interface, not move interface.

http://wiki.ros.org/rtmros_nextage/Tutorials/Programming may help.

kuochiyoug, 2015-09-26:
Oh!
Thanks you very much!
It works with it.!

130s, 2015-09-27:
I'm closing @kuochiyoug since I assume your original issue is resolved. Please feel free to reopen (or even better to open as new ticket) if you come up with specific questions.

closed 2015-09-27 by 130s