tork-a/rtmros_nextage #215: indigo版でシミュレータが起動できない

RyuYamamoto, 2016-01-08:
wikiより（http://wiki.ros.org/rtmros_nextage/Tutorials/Install%20NEXTAGE%20OPEN%20software%20on%20your%20machine）
indigo版を以下のようにインストール後，
> $ rtmlaunch nextage_ros_bridge nextage_ros_bridge_simulation.launch

シミュレータを起動しようとしたところ，以下のエラーで起動できませんでした．
>Traceback (most recent call last):
  File "/opt/ros/indigo/bin/rtmlaunch", line 28, in <module>
    from openrtm_tools import rtmstart, rtmlaunch
  File "/opt/ros/indigo/lib/python2.7/dist-packages/openrtm_tools/rtmlaunch.py", line 14, in <module>
    from rtshell import rtcon
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rtshell/rtcon.py", line 26, in <module>
    import rtctree.tree
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rtctree/tree.py", line 30, in <module>
    from rtctree.directory import Directory
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rtctree/directory.py", line 26, in <module>
    from rtctree.component import Component
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rtctree/component.py", line 30, in <module>
    import rtctree.sdo
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rtctree/sdo.py", line 26, in <module>
    class RTCObserver(OpenRTM__POA.ComponentObserver):
AttributeError: 'module' object has no attribute 'ComponentObserver'

この場合の対処はどのようにすればよいのでしょうか．

k-okada, 2016-01-08:
I think you have installed openrtm other than ros repository, 
please try
```
sudo apt-get install --reinstall ros-indigo-openrtm*
```
also check 
```
aptitude search openrtm
```
and
```
python -v /opt/ros/indigo/bin/rtmlaunch -v nextage_ros_bridge nextage_ros_bridge_simulation.launch
```
may help

RyuYamamoto, 2016-01-08:
起動することが出来ました．ありがとうございます．

closed 2016-09-05 by k-okada