tork-a/rtmros_nextage #216: [rtmlaunch] Could not Activate ( nextage:15005/HrpsysSeqStateROSBridge0.rtc ) : Invalid CORBA naming service: nextage:15005

RyuYamamoto, 2016-01-08:
> roslaunch nextage_ros_bridge nextage_ros_bridge_real.launch

で実機に接続できずに以下のエラーがでます．

```
 roslaunch... logging to /home/haze/.ros/log/4f770006-b5c3-11e5-b5d9-705812de5ae5/roslaunch-haze-CF-S10EWHDS-22978.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is < 1GB.
started roslaunch server http://haze-CF-S10EWHDS:43145/

SUMMARY
========-

PARAMETERS
 * /HrpsysSeqStateROSBridge/publish_sensor_transforms: True
 * /collision_state/comp_name: CollisionDetector0
 * /controller_configuration: [{'group_name': '...
 * /diagnostic_aggregator/analyzers/computers/contains: ['HD Temp', 'CPU ...
 * /diagnostic_aggregator/analyzers/computers/path: Computers
 * /diagnostic_aggregator/analyzers/computers/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/hrpsys/contains: ['hrpEC']
 * /diagnostic_aggregator/analyzers/hrpsys/path: Hrpsys
 * /diagnostic_aggregator/analyzers/hrpsys/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/mode/contains: ['Operating Mode'...
 * /diagnostic_aggregator/analyzers/mode/path: Mode
 * /diagnostic_aggregator/analyzers/mode/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/motor/contains: ['Motor']
 * /diagnostic_aggregator/analyzers/motor/path: Motor
 * /diagnostic_aggregator/analyzers/motor/type: diagnostic_aggreg...
 * /diagnostic_aggregator/base_path: 
 * /diagnostic_aggregator/pub_rate: 1.0
 * /robot_description: <?xml version="1....
 * /rosdistro: indigo
 * /rosversion: 1.11.13

NODES
  /
    CollisionDetectorComp (hrpsys/CollisionDetectorComp)
    DataLoggerServiceROSBridge (hrpsys_ros_bridge/DataLoggerServiceROSBridgeComp)
    ForwardKinematicsServiceROSBridge (hrpsys_ros_bridge/ForwardKinematicsServiceROSBridgeComp)
    HrpsysJointTrajectoryBridge (hrpsys_ros_bridge/HrpsysJointTrajectoryBridge)
    HrpsysSeqStateROSBridge (hrpsys_ros_bridge/HrpsysSeqStateROSBridge)
    RobotHardwareServiceROSBridge (hrpsys_ros_bridge/RobotHardwareServiceROSBridgeComp)
    SequencePlayerServiceROSBridge (hrpsys_ros_bridge/SequencePlayerServiceROSBridgeComp)
    StateHolderServiceROSBridge (hrpsys_ros_bridge/StateHolderServiceROSBridgeComp)
    collision_state (hrpsys_ros_bridge/collision_state.py)
    diagnostic_aggregator (diagnostic_aggregator/aggregator_node)
    hrpsys_profile (hrpsys_ros_bridge/hrpsys_profile.py)
    hrpsys_py (hironx_ros_bridge/hironx.py)
    hrpsys_ros_diagnostics (hrpsys_ros_bridge/diagnostics.py)
    hrpsys_state_publisher (robot_state_publisher/state_publisher)
    rtmlaunch_collision_detector (openrtm_tools/rtmlaunch.py)
    rtmlaunch_hrpsys_ros_bridge (openrtm_tools/rtmlaunch.py)
    sensor_ros_bridge_connect (hrpsys_ros_bridge/sensor_ros_bridge_connect.py)

WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtactivate
WARNING: unrecognized tag rtactivate
WARNING: unrecognized tag rtactivate
WARNING: unrecognized tag rtactivate
[... 112 lines cut ...]
[rtmlaunch_collision_detector-16] process has died [pid 23938, exit code 1, cmd /opt/ros/indigo/share/openrtm_tools/scripts/rtmlaunch.py /opt/ros/indigo/share/hrpsys_ros_bridge/launch/collision_detector.launch __name:=rtmlaunch_collision_detector __log:=/home/haze/.ros/log/4f770006-b5c3-11e5-b5d9-705812de5ae5/rtmlaunch_collision_detector-16.log].
log file: /home/haze/.ros/log/4f770006-b5c3-11e5-b5d9-705812de5ae5/rtmlaunch_collision_detector-16*.log
/opt/ros/indigo/share/hrpsys_ros_bridge/scripts/collision_state.py:181: SyntaxWarning: The publisher should be created with an explicit keyword argument 'queue_size'. Please see http://wiki.ros.org/rospy/Overview/Publishers%20and%20Subscribers for more information.
  pub_diagnostics = rospy.Publisher('diagnostics', DiagnosticArray)
/opt/ros/indigo/share/hrpsys_ros_bridge/scripts/collision_state.py:182: SyntaxWarning: The publisher should be created with an explicit keyword argument 'queue_size'. Please see http://wiki.ros.org/rospy/Overview/Publishers%20and%20Subscribers for more information.
  pub_collision = rospy.Publisher('collision_detector_marker_array', MarkerArray)
configuration ORB with nextage:15005
[ERROR] Connection Failed with the Nameserver (hostname=nextage port=15005).
Make sure the hostname is correct and the Nameserver is running.
CORBA.TRANSIENT(omniORB.TRANSIENT_ConnectFailed, CORBA.COMPLETED_NO)
[collision_state-17] process has died [pid 23941, exit code 1, cmd /opt/ros/indigo/share/hrpsys_ros_bridge/scripts/collision_state.py /opt/jsk/etc/NEXTAGE/model/main.wrl -ORBInitRef NameService=corbaloc:iiop:nextage:15005/NameService __name:=collision_state __log:=/home/haze/.ros/log/4f770006-b5c3-11e5-b5d9-705812de5ae5/collision_state-17.log].
log file: /home/haze/.ros/log/4f770006-b5c3-11e5-b5d9-705812de5ae5/collision_state-17*.log
^C[RobotHardwareServiceROSBridge-14] killing on exit
[StateHolderServiceROSBridge-13] killing on exit
[ForwardKinematicsServiceROSBridge-12] killing on exit
[diagnostic_aggregator-6] killing on exit
[DataLoggerServiceROSBridge-11] killing on exit
[hrpsys_ros_diagnostics-5] killing on exit
[SequencePlayerServiceROSBridge-10] killing on exit
[HrpsysJointTrajectoryBridge-3] killing on exit
[HrpsysSeqStateROSBridge-2] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

RyuYamamoto, 2016-01-08:
以下、使用してるバージョンです．

```
Version: 0.7.2-0trusty-20151117-0351-+0000
0.7.2
/opt/ros/indigo/share/nextage_ros_bridge
```

k-okada, 2016-01-08:
apt-get update
apt-get install --reinstall collada-dom*
apt-get install --reinstall ros-indigo-openhrp*

and try again

RyuYamamoto, 2016-01-08:
```
$ sudo apt-get install --reinstall collada-dom*
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'collada-dom2.4-sp-base' for regex 'collada-dom*'
Note, selecting 'collada-dom2.4-sp' for regex 'collada-dom*'
Note, selecting 'collada-dom2.4-dp-dev' for regex 'collada-dom*'
Note, selecting 'collada-dom2.4-dp-base' for regex 'collada-dom*'
Note, selecting 'collada-dom-dev' for regex 'collada-dom*'
Note, selecting 'collada-dom2.4-sp-dev' for regex 'collada-dom*'
Note, selecting 'collada-dom2.4-dp' for regex 'collada-dom*'
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 collada-dom2.4-dp : Conflicts: collada-dom2.4-sp but 2.4.4.4-ubuntu1~trusty1 is to be installed
 collada-dom2.4-dp-base : Conflicts: collada-dom2.4-sp-base but 2.4.4.4-ubuntu1~trusty1 is to be installed
 collada-dom2.4-dp-dev : Conflicts: collada-dom2.4-sp-dev but 2.4.4.4-ubuntu1~trusty1 is to be installed
 collada-dom2.4-sp : Conflicts: collada-dom2.4-dp but 2.4.4.4-ubuntu1~trusty1 is to be installed
 collada-dom2.4-sp-base : Conflicts: collada-dom2.4-dp-base but 2.4.4.4-ubuntu1~trusty1 is to be installed
 collada-dom2.4-sp-dev : Conflicts: collada-dom2.4-dp-dev but 2.4.4.4-ubuntu1~trusty1 is to be installed
E: Unable to correct problems, you have held broken packages
```

k-okada, 2016-01-08:
please try
```
apt-get remove collada-dom2.4-sp*
apt-get install --reinstall collada-dom2.4-dp*
```

RyuYamamoto, 2016-01-08:
うまく行きましたが、ros_birdgeは起動しないです．
```
... logging to /home/haze/.ros/log/a2950c2a-b5cc-11e5-acba-705812de5ae5/roslaunch-haze-CF-S10EWHDS-8999.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://haze-CF-S10EWHDS:32940/

SUMMARY
========

PARAMETERS
 * /HrpsysSeqStateROSBridge/publish_sensor_transforms: True
 * /collision_state/comp_name: CollisionDetector0
 * /controller_configuration: [{'group_name': '...
 * /diagnostic_aggregator/analyzers/computers/contains: ['HD Temp', 'CPU ...
 * /diagnostic_aggregator/analyzers/computers/path: Computers
 * /diagnostic_aggregator/analyzers/computers/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/hrpsys/contains: ['hrpEC', 'Emerge...
 * /diagnostic_aggregator/analyzers/hrpsys/path: Hrpsys
 * /diagnostic_aggregator/analyzers/hrpsys/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/mode/contains: ['Operating Mode'...
 * /diagnostic_aggregator/analyzers/mode/path: Mode
 * /diagnostic_aggregator/analyzers/mode/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/motor/contains: ['Motor']
 * /diagnostic_aggregator/analyzers/motor/path: Motor
 * /diagnostic_aggregator/analyzers/motor/type: diagnostic_aggreg...
 * /diagnostic_aggregator/base_path: 
 * /diagnostic_aggregator/pub_rate: 1.0
 * /robot_description: <?xml version="1....
 * /rosdistro: indigo
 * /rosversion: 1.11.13

NODES
  /
    CollisionDetectorComp (hrpsys/CollisionDetectorComp)
    DataLoggerServiceROSBridge (hrpsys_ros_bridge/DataLoggerServiceROSBridgeComp)
    ForwardKinematicsServiceROSBridge (hrpsys_ros_bridge/ForwardKinematicsServiceROSBridgeComp)
    HrpsysJointTrajectoryBridge (hrpsys_ros_bridge/HrpsysJointTrajectoryBridge)
    HrpsysSeqStateROSBridge (hrpsys_ros_bridge/HrpsysSeqStateROSBridge)
    RobotHardwareServiceROSBridge (hrpsys_ros_bridge/RobotHardwareServiceROSBridgeComp)
    SequencePlayerServiceROSBridge (hrpsys_ros_bridge/SequencePlayerServiceROSBridgeComp)
    StateHolderServiceROSBridge (hrpsys_ros_bridge/StateHolderServiceROSBridgeComp)
    collision_state (hrpsys_ros_bridge/collision_state.py)
    diagnostic_aggregator (diagnostic_aggregator/aggregator_node)
    hrpsys_profile (hrpsys_ros_bridge/hrpsys_profile.py)
    hrpsys_py (hironx_ros_bridge/hironx.py)
    hrpsys_ros_diagnostics (hrpsys_ros_bridge/diagnostics.py)
    hrpsys_state_publisher (robot_state_publisher/state_publisher)
    rtmlaunch_collision_detector (openrtm_tools/rtmlaunch.py)
    rtmlaunch_hrpsys_ros_bridge (openrtm_tools/rtmlaunch.py)
    sensor_ros_bridge_connect (hrpsys_ros_bridge/sensor_ros_bridge_connect.py)

WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtactivate
WARNING: unrecognized tag rtactivate
WARNING: unrecognized tag rtactivate
WARNING: unrecognized tag rtactivate
WARNING: unrecognized tag rtactivate
[... 113 lines cut ...]
configuration ORB with nextage:15005
[ERROR] Connection Failed with the Nameserver (hostname=nextage port=15005).
Make sure the hostname is correct and the Nameserver is running.
CORBA.TRANSIENT(omniORB.TRANSIENT_ConnectFailed, CORBA.COMPLETED_NO)
[collision_state-18] process has died [pid 9365, exit code 1, cmd /home/haze/catkin_ws/src/rtmros_common/hrpsys_ros_bridge/scripts/collision_state.py /opt/jsk/etc/NEXTAGE/model/main.wrl -ORBInitRef NameService=corbaloc:iiop:nextage:15005/NameService __name:=collision_state __log:=/home/haze/.ros/log/a2950c2a-b5cc-11e5-acba-705812de5ae5/collision_state-18.log].
log file: /home/haze/.ros/log/a2950c2a-b5cc-11e5-acba-705812de5ae5/collision_state-18*.log
[HrpsysSeqStateROSBridge-3] process has finished cleanly
log file: /home/haze/.ros/log/a2950c2a-b5cc-11e5-acba-705812de5ae5/HrpsysSeqStateROSBridge-3*.log
[HrpsysJointTrajectoryBridge-4] process has died [pid 9080, exit code -11, cmd /home/haze/catkin_ws/devel/lib/hrpsys_ros_bridge/HrpsysJointTrajectoryBridge -o corba.nameservers:nextage:15005 -o naming.formats:%n.rtc -o exec_cxt.periodic.type:PeriodicExecutionContext -o exec_cxt.periodic.rate:2000 -o logger.file_name:/tmp/rtc%p.log -o model:/opt/jsk/etc/NEXTAGE/model/main.wrl __name:=HrpsysJointTrajectoryBridge __log:=/home/haze/.ros/log/a2950c2a-b5cc-11e5-acba-705812de5ae5/HrpsysJointTrajectoryBridge-4.log].
log file: /home/haze/.ros/log/a2950c2a-b5cc-11e5-acba-705812de5ae5/HrpsysJointTrajectoryBridge-4*.log
^C[RobotHardwareServiceROSBridge-15] killing on exit
[StateHolderServiceROSBridge-14] killing on exit
[ForwardKinematicsServiceROSBridge-13] killing on exit
[DataLoggerServiceROSBridge-12] killing on exit
[SequencePlayerServiceROSBridge-11] killing on exit
[hrpsys_ros_diagnostics-6] killing on exit
[hrpsys_state_publisher-5] killing on exit
[diagnostic_aggregator-7] killing on exit
^[[A[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

```

k-okada, 2016-01-08:
are you able to connect to  nextage? are you sure that they have omni nameserver at port 15005?
```
ping nextage
rtls
rtls nextage:15005
```

130s, 2016-01-08:
Looks similar to https://github.com/tork-a/rtmros_nextage/issues/179. And as said in https://github.com/tork-a/rtmros_nextage/issues/179#issuecomment-112315984, #179 was environment specific issue (, which I can't recall how we fixed that).

And #179 and this #216 are happening at the same lab.

RyuYamamoto, 2016-01-08:
```
haze@haze-CF-S10EWHDS:~$ rtls
rtls: Invalid CORBA naming service: localhost:15005
haze@haze-CF-S10EWHDS:~$ rtls nextage:15005
rtls: Cannot list ports.
```

130s, 2016-01-08:
What about `rtls nextage:15005/` (with slash)?

RyuYamamoto, 2016-01-08:
```
haze@haze-CF-S10EWHDS:~$ rtls nextage:15005/
rtls: Invalid CORBA naming service: nextage:15005
```

130s, 2016-01-08:
`ping nextage`?

RyuYamamoto, 2016-01-08:
pingは通りました．

130s, 2016-01-08:
From https://github.com/tork-a/rtmros_nextage/issues/216#issuecomment-169908237, corba name server is not running on the target host. Please make sure that your robot is started properly, or you're trying to connect the right host.

closed 2016-01-08 by 130s

k-okada, 2016-01-10:
これは実機のnamesrverが上がっていなかったということでしょうか？
実機のnameserverは実機を立ち上げえたら動くとはずですが，何故かリブートしたら治った？，そもそもnameserverが走っていなかった？のどちらかでしょうか？
あまり解決方法になりませんが，少しでも混乱をなくすためにエラーメッセージを少し親切にしておきました．こういう風に言ってくれたら，困らなかったのに，というのがあれば，コメントを追加しておいて下さい．
https://github.com/start-jsk/rtmros_common/pull/904

130s, 2016-01-11:
On the host OP was targetting, nameserver was not running
(turned out there were 2 QNX hosts in the same network but nameserver was running on one host only, and we did not realize the OP was connecting to the wrong one).

k-okada, 2016-01-12:
what is `OP` ?

130s, 2016-01-12:
Sorry I meant [Original Poster](https://en.wikipedia.org/wiki/Internet_forum#Post).

k-okada, 2016-01-12:
I see, so using `nextage` as target hostname was wrong, is it correct?

130s, 2016-01-13:
No, we initially assigned `nextage` in `/etc/hosts` to a wrong host where necessary service was not running.
Using `nextage` for an argument `--host` worked perfectly fine for us.

k-okada, 2016-01-13:
I see,
```
$ roslaunch nextage_ros_bridge_real.launch --ros-args
Optional Arguments:
  CONF_FILE_COLLISIONDETECT (default
"/opt/ros/indigo/share/hironx_ros_bridge/conf/hironx_realrobot_fixedpath.conf"):
undocumented
  MODEL_FILE (default "/opt/jsk/etc/NEXTAGE/model/main.wrl"): undocumented
  USE_SERVOCONTROLLER (default "false"): undocumented
  corbaport (default "15005"): undocumented
  nameserver (default "nextage"): undocumented
```

`--nameserver` instead of `--host` ??? ;-)