tork-a/rtmros_nextage #217: The error before refreshing the cache was: Failed to load library /opt/ros/indigo/lib//libueye_cam_nodelet.so

RyuYamamoto, 2016-01-08:
> $ roslaunch nextage_ros_bridge nextage_ueye_stereo.launch

でカメラ画像が取得できずに以下のエラーが出ます．カメラは接続されています．

``` 
... logging to /home/haze/.ros/log/ff7f1b04-b5d8-11e5-80f9-705812de5ae5/roslaunch-haze-CF-S10EWHDS-19231.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://haze-CF-S10EWHDS:47764/

SUMMARY
========

PARAMETERS
 * /rosdistro: indigo
 * /rosversion: 1.11.13
 * /ueye_cam_nodelet_master/auto_exposure: False
 * /ueye_cam_nodelet_master/auto_frame_rate: False
 * /ueye_cam_nodelet_master/auto_gain: True
 * /ueye_cam_nodelet_master/auto_white_balance: True
 * /ueye_cam_nodelet_master/binning: 2
 * /ueye_cam_nodelet_master/blue_gain: 16
 * /ueye_cam_nodelet_master/camera_id: 1
 * /ueye_cam_nodelet_master/camera_intrinsics_file: 
 * /ueye_cam_nodelet_master/camera_name: right
 * /ueye_cam_nodelet_master/camera_parameters_file: 
 * /ueye_cam_nodelet_master/camera_topic: image_raw
 * /ueye_cam_nodelet_master/color_mode: rgb8
 * /ueye_cam_nodelet_master/exposure: 33
 * /ueye_cam_nodelet_master/ext_trigger_mode: False
 * /ueye_cam_nodelet_master/flash_delay: 30000
 * /ueye_cam_nodelet_master/flash_duration: 1000
 * /ueye_cam_nodelet_master/frame_rate: 30.0
 * /ueye_cam_nodelet_master/gain_boost: False
 * /ueye_cam_nodelet_master/green_gain: 1
 * /ueye_cam_nodelet_master/image_height: 1200
 * /ueye_cam_nodelet_master/image_left: -1
 * /ueye_cam_nodelet_master/image_top: -1
 * /ueye_cam_nodelet_master/image_width: 1600
 * /ueye_cam_nodelet_master/master_gain: 0
 * /ueye_cam_nodelet_master/pixel_clock: 24
 * /ueye_cam_nodelet_master/red_gain: 0
 * /ueye_cam_nodelet_master/sensor_scaling: 1.0
 * /ueye_cam_nodelet_master/subsampling: 1
 * /ueye_cam_nodelet_master/white_balance_blue_offset: 0
 * /ueye_cam_nodelet_master/white_balance_red_offset: 0
 * /ueye_cam_nodelet_slave1/auto_exposure: False
 * /ueye_cam_nodelet_slave1/auto_frame_rate: False
 * /ueye_cam_nodelet_slave1/auto_gain: True
 * /ueye_cam_nodelet_slave1/auto_white_balance: True
 * /ueye_cam_nodelet_slave1/binning: 2
 * /ueye_cam_nodelet_slave1/blue_gain: 16
 * /ueye_cam_nodelet_slave1/camera_id: 2
 * /ueye_cam_nodelet_slave1/camera_intrinsics_file: 
 * /ueye_cam_nodelet_slave1/camera_name: left
 * /ueye_cam_nodelet_slave1/camera_parameters_file: 
 * /ueye_cam_nodelet_slave1/camera_topic: image_raw
 * /ueye_cam_nodelet_slave1/color_mode: rgb8
 * /ueye_cam_nodelet_slave1/exposure: 33
 * /ueye_cam_nodelet_slave1/ext_trigger_mode: False
 * /ueye_cam_nodelet_slave1/flash_delay: 0
 * /ueye_cam_nodelet_slave1/flash_duration: 1000
 * /ueye_cam_nodelet_slave1/frame_rate: 30.0
 * /ueye_cam_nodelet_slave1/gain_boost: False
 * /ueye_cam_nodelet_slave1/green_gain: 1
 * /ueye_cam_nodelet_slave1/image_height: 1200
 * /ueye_cam_nodelet_slave1/image_left: -1
 * /ueye_cam_nodelet_slave1/image_top: -1
 * /ueye_cam_nodelet_slave1/image_width: 1600
 * /ueye_cam_nodelet_slave1/master_gain: 0
 * /ueye_cam_nodelet_slave1/pixel_clock: 24
 * /ueye_cam_nodelet_slave1/red_gain: 0
 * /ueye_cam_nodelet_slave1/sensor_scaling: 1.0
 * /ueye_cam_nodelet_slave1/subsampling: 1
 * /ueye_cam_nodelet_slave1/white_balance_blue_offset: 0
 * /ueye_cam_nodelet_slave1/white_balance_red_offset: 0

NODES
  /left/
    image_proc (image_proc/image_proc)
  /right/
    image_proc (image_proc/image_proc)
  /
    head_link_to_stereo_link (tf/static_transform_publisher)
    left_force_subscribe (rostopic/rostopic)
    nodelet_manager (nodelet/nodelet)
    right_force_subscribe (rostopic/rostopic)
    stereo_image_proc (stereo_image_proc/stereo_image_proc)
    stereo_link_to_stereo_frame (tf/static_transform_publisher)
    ueye_cam_nodelet_master (nodelet/nodelet)
    ueye_cam_nodelet_slave1 (nodelet/nodelet)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[nodelet_manager-1]: started with pid [19249]
process[ueye_cam_nodelet_master-2]: started with pid [19250]
[ERROR] [1452239845.399739675]: Skipping XML Document "/opt/ros/indigo/share/hector_pose_estimation/hector_pose_estimation_nodelets.xml" which had no Root Element.  This likely means the XML is malformed or missing.
[ INFO] [1452239845.411970605]: Initializing nodelet with 4 worker threads.
process[ueye_cam_nodelet_slave1-3]: started with pid [19285]
[ERROR] [1452239845.771483228]: Skipping XML Document "/opt/ros/indigo/share/hector_pose_estimation/hector_pose_estimation_nodelets.xml" which had no Root Element.  This likely means the XML is malformed or missing.
process[left_force_subscribe-4]: started with pid [19301]
[ERROR] [1452239845.930619060]: Failed to load nodelet [/ueye_cam_nodelet_master] of type [ueye_cam/ueye_cam_nodelet] even after refreshing the cache: Failed to load library /opt/ros/indigo/lib//libueye_cam_nodelet.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libueye_api.so: cannot open shared object file: No such file or directory)
[ERROR] [1452239845.930921245]: The error before refreshing the cache was: Failed to load library /opt/ros/indigo/lib//libueye_cam_nodelet.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libueye_api.so: cannot open shared object file: No such file or directory)
[FATAL] [1452239845.945136572]: Service call failed!
process[right_force_subscribe-5]: started with pid [19304]
[ueye_cam_nodelet_master-2] process has died [pid 19250, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load ueye_cam/ueye_cam_nodelet nodelet_manager __name:=ueye_cam_nodelet_master __log:=/home/haze/.ros/log/ff7f1b04-b5d8-11e5-80f9-705812de5ae5/ueye_cam_nodelet_master-2.log].
log file: /home/haze/.ros/log/ff7f1b04-b5d8-11e5-80f9-705812de5ae5/ueye_cam_nodelet_master-2*.log
process[right/image_proc-6]: started with pid [19311]
[ERROR] [1452239846.334148959]: Skipping XML Document "/opt/ros/indigo/share/hector_pose_estimation/hector_pose_estimation_nodelets.xml" which had no Root Element.  This likely means the XML is malformed or missing.
[ERROR] [1452239846.402874685]: Skipping XML Document "/opt/ros/indigo/share/hector_pose_estimation/hector_pose_estimation_nodelets.xml" which had no Root Element.  This likely means the XML is malformed or missing.
process[left/image_proc-7]: started with pid [19339]
subscribed to [/left/camera_info]
[ERROR] [1452239846.552183481]: Failed to load nodelet [/ueye_cam_nodelet_slave1] of type [ueye_cam/ueye_cam_nodelet] even after refreshing the cache: Failed to load library /opt/ros/indigo/lib//libueye_cam_nodelet.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libueye_api.so: cannot open shared object file: No such file or directory)
[ERROR] [1452239846.552562327]: The error before refreshing the cache was: Failed to load library /opt/ros/indigo/lib//libueye_cam_nodelet.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libueye_api.so: cannot open shared object file: No such file or directory)
[FATAL] [1452239846.570856937]: Service call failed!
[ERROR] [1452239846.636243784]: Skipping XML Document "/opt/ros/indigo/share/hector_pose_estimation/hector_pose_estimation_nodelets.xml" which had no Root Element.  This likely means the XML is malformed or missing.
process[stereo_image_proc-8]: started with pid [19366]
[ueye_cam_nodelet_slave1-3] process has died [pid 19285, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load ueye_cam/ueye_cam_nodelet nodelet_manager __name:=ueye_cam_nodelet_slave1 __log:=/home/haze/.ros/log/ff7f1b04-b5d8-11e5-80f9-705812de5ae5/ueye_cam_nodelet_slave1-3.log].
log file: /home/haze/.ros/log/ff7f1b04-b5d8-11e5-80f9-705812de5ae5/ueye_cam_nodelet_slave1-3*.log
[ERROR] [1452239846.917792147]: Skipping XML Document "/opt/ros/indigo/share/hector_pose_estimation/hector_pose_estimation_nodelets.xml" which had no Root Element.  This likely means the XML is malformed or missing.
process[head_link_to_stereo_link-9]: started with pid [19386]
process[stereo_link_to_stereo_frame-10]: started with pid [19431]
subscribed to [/right/camera_info]
```

130s, 2016-01-08:
```
ls /usr/local/share/ueye/bin/ueyedemo
```
?

RyuYamamoto, 2016-01-08:
```
$ ls /usr/local/share/ueye/bin/ueyedemo
ls: cannot access /usr/local/share/ueye/bin/ueyedemo: No such file or directory
```

130s, 2016-01-08:
> ls: cannot access /usr/local/share/ueye/bin/ueyedemo: No such file or directory

This sounds like the camera's proprietary driver is no installed.
I'll share the best instruction to install it.

k-okada, 2016-01-08:
https://jp.ids-imaging.com/ueye-software-archive.html#4.61_Linux
this one?

130s, 2016-01-08:
I thought `ueye_cam` package may have taken care of installation of the library, but I was wrong (e.g. looking at https://github.com/anqixu/ueye_cam/issues/21). 

> https://jp.ids-imaging.com/ueye-software-archive.html#4.61_Linux

So, yes. We need to manually install it for now.

@RyuYamamoto I'm closing this one but please reopen if the issue persists even with the library installed.

closed 2016-01-08 by 130s