tork-a/rtmros_nextage #218: [ERROR] [1452821520.942674221]: No UEye cameras are connected

RyuYamamoto, 2016-01-15:
ueye SDK 4.4をインストール後
> $roslaunch nextage_ros_bridge nextage_ueye_stereo.launch

すると以下のようなエラーが出ます
```
... logging to /home/haze/.ros/log/8584ed24-bb28-11e5-be67-705812de5ae5/roslaunch-haze-CF-S10EWHDS-30607.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://haze-CF-S10EWHDS:37987/

SUMMARY
========

PARAMETERS
 * /rosdistro: indigo
 * /rosversion: 1.11.13
 * /ueye_cam_nodelet_master/auto_exposure: False
 * /ueye_cam_nodelet_master/auto_frame_rate: False
 * /ueye_cam_nodelet_master/auto_gain: True
 * /ueye_cam_nodelet_master/auto_white_balance: True
 * /ueye_cam_nodelet_master/binning: 2
 * /ueye_cam_nodelet_master/blue_gain: 16
 * /ueye_cam_nodelet_master/camera_id: 1
 * /ueye_cam_nodelet_master/camera_intrinsics_file: 
 * /ueye_cam_nodelet_master/camera_name: right
 * /ueye_cam_nodelet_master/camera_parameters_file: 
 * /ueye_cam_nodelet_master/camera_topic: image_raw
 * /ueye_cam_nodelet_master/color_mode: rgb8
 * /ueye_cam_nodelet_master/exposure: 33
 * /ueye_cam_nodelet_master/ext_trigger_mode: False
 * /ueye_cam_nodelet_master/flash_delay: 30000
 * /ueye_cam_nodelet_master/flash_duration: 1000
 * /ueye_cam_nodelet_master/frame_rate: 30.0
 * /ueye_cam_nodelet_master/gain_boost: False
 * /ueye_cam_nodelet_master/green_gain: 1
 * /ueye_cam_nodelet_master/image_height: 1200
 * /ueye_cam_nodelet_master/image_left: -1
 * /ueye_cam_nodelet_master/image_top: -1
 * /ueye_cam_nodelet_master/image_width: 1600
 * /ueye_cam_nodelet_master/master_gain: 0
 * /ueye_cam_nodelet_master/pixel_clock: 24
 * /ueye_cam_nodelet_master/red_gain: 0
 * /ueye_cam_nodelet_master/sensor_scaling: 1.0
 * /ueye_cam_nodelet_master/subsampling: 1
 * /ueye_cam_nodelet_master/white_balance_blue_offset: 0
 * /ueye_cam_nodelet_master/white_balance_red_offset: 0
 * /ueye_cam_nodelet_slave1/auto_exposure: False
 * /ueye_cam_nodelet_slave1/auto_frame_rate: False
 * /ueye_cam_nodelet_slave1/auto_gain: True
 * /ueye_cam_nodelet_slave1/auto_white_balance: True
 * /ueye_cam_nodelet_slave1/binning: 2
 * /ueye_cam_nodelet_slave1/blue_gain: 16
 * /ueye_cam_nodelet_slave1/camera_id: 2
 * /ueye_cam_nodelet_slave1/camera_intrinsics_file: 
 * /ueye_cam_nodelet_slave1/camera_name: left
 * /ueye_cam_nodelet_slave1/camera_parameters_file: 
 * /ueye_cam_nodelet_slave1/camera_topic: image_raw
 * /ueye_cam_nodelet_slave1/color_mode: rgb8
 * /ueye_cam_nodelet_slave1/exposure: 33
 * /ueye_cam_nodelet_slave1/ext_trigger_mode: False
 * /ueye_cam_nodelet_slave1/flash_delay: 0
 * /ueye_cam_nodelet_slave1/flash_duration: 1000
 * /ueye_cam_nodelet_slave1/frame_rate: 30.0
 * /ueye_cam_nodelet_slave1/gain_boost: False
 * /ueye_cam_nodelet_slave1/green_gain: 1
 * /ueye_cam_nodelet_slave1/image_height: 1200
 * /ueye_cam_nodelet_slave1/image_left: -1
 * /ueye_cam_nodelet_slave1/image_top: -1
 * /ueye_cam_nodelet_slave1/image_width: 1600
 * /ueye_cam_nodelet_slave1/master_gain: 0
 * /ueye_cam_nodelet_slave1/pixel_clock: 24
 * /ueye_cam_nodelet_slave1/red_gain: 0
 * /ueye_cam_nodelet_slave1/sensor_scaling: 1.0
 * /ueye_cam_nodelet_slave1/subsampling: 1
 * /ueye_cam_nodelet_slave1/white_balance_blue_offset: 0
 * /ueye_cam_nodelet_slave1/white_balance_red_offset: 0

NODES
  /left/
    image_proc (image_proc/image_proc)
  /right/
    image_proc (image_proc/image_proc)
  /
    head_link_to_stereo_link (tf/static_transform_publisher)
    left_force_subscribe (rostopic/rostopic)
    nodelet_manager (nodelet/nodelet)
    right_force_subscribe (rostopic/rostopic)
    stereo_image_proc (stereo_image_proc/stereo_image_proc)
    stereo_link_to_stereo_frame (tf/static_transform_publisher)
    ueye_cam_nodelet_master (nodelet/nodelet)
    ueye_cam_nodelet_slave1 (nodelet/nodelet)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[nodelet_manager-1]: started with pid [31736]
[ERROR] [1452821876.510326540]: Skipping XML Document "/opt/ros/indigo/share/hector_pose_estimation/hector_pose_estimation_nodelets.xml" which had no Root Element.  This likely means the XML is malformed or missing.
[ INFO] [1452821876.549133141]: Initializing nodelet with 4 worker threads.
process[ueye_cam_nodelet_master-2]: started with pid [31962]
process[ueye_cam_nodelet_slave1-3]: started with pid [32179]
[ INFO] [1452821877.825481474]: Unable to open camera calibration file [/home/haze/.ros/camera_info/right.yaml]
process[left_force_subscribe-4]: started with pid [32476]
WARNING: topic [/left/camera_info] does not appear to be published yet
process[right_force_subscribe-5]: started with pid [32681]
[ERROR] [1452821880.001279699]: No UEye cameras are connected

[ERROR] [1452821880.001485576]: Hint: make sure that the IDS camera daemon (/etc/init.d/ueyeusbdrc) is running

[ERROR] [1452821880.001649075]: Failed to initialize [right]
ERROR: cannot launch node of type [image_proc/image_proc]: can't locate node [image_proc] in package [image_proc]
[ INFO] [1452821880.951174180]: Unable to open camera calibration file [/home/haze/.ros/camera_info/left.yaml]
subscribed to [/right/camera_info]
ERROR: cannot launch node of type [image_proc/image_proc]: can't locate node [image_proc] in package [image_proc]
ERROR: cannot launch node of type [stereo_image_proc/stereo_image_proc]: can't locate node [stereo_image_proc] in package [stereo_image_proc]
[ERROR] [1452821883.109873335]: No UEye cameras are connected

[ERROR] [1452821883.110497462]: Hint: make sure that the IDS camera daemon (/etc/init.d/ueyeusbdrc) is running

[ERROR] [1452821883.110921294]: Failed to initialize [left]
process[head_link_to_stereo_link-9]: started with pid [1125]
subscribed to [/left/camera_info]
process[stereo_link_to_stereo_frame-10]: started with pid [1361]
```

RyuYamamoto, 2016-01-15:
ros_bridge側のデバッグメッセージです
正常に起動していないということでしょうか？

```
[rtmlaunch] [ERROR] Could not Connect ( /nextage:15005/StateHolderServiceROSBridge.rtc:StateHolderService , /nextage:15005/sh.rtc:StateHolderService ):  No such object: /nextage:15005/sh.rtc:StateHolderService 
```

130s, 2016-01-15:
> Hint: make sure that the IDS camera daemon (/etc/init.d/ueyeusbdrc) is running

?

130s, 2016-01-15:
> Hint: make sure that the IDS camera daemon (/etc/init.d/ueyeusbdrc) is running

Can you check this?

RyuYamamoto, 2016-01-15:
ueyeusbd is not running.

RyuYamamoto, 2016-01-15:
```
$ sudo /etc/init.d/ueyeusbdrc status 
ueyeusbd is not running.
```

130s, 2016-01-15:
> [ERROR] [1452821880.001279699]: No UEye cameras are connected
> [ERROR] [1452821880.001485576]: Hint: make sure that the IDS camera daemon (/etc/init.d/ueyeusbdrc) is running

BTW, ros_bridge not running is a separate issue and shouldn't matter to this particular problem.

RyuYamamoto, 2016-01-15:
uninstall ueye SDK 4.4,
```
$ sudo ./ueyed_install-eth uninstall
./ueyed_install-ethueyed installer v1.1.14
Selecting uninstall...
Stopping daemon...
Killing ueyeethd... Done.
Remove service start entry for ueyeethd... Done.
Remove runlevel control script...
Remove ueyed configuration file...
Remove binary directory...
Remove ueye library configuration files...
Remove firmware directory...
Remove icon directory...
Remove ueye directory...
Remove system user...
Remove system group...
Remove runtime directory...
Remove firmware directory...
Remove header...
Remove library...
Run ldconfig to update library cache...
sh: 0: getcwd() failed: No such file or directory
Remove links... /usr/bin/ueyesetid /usr/bin/ueyesetip /usr/bin/ueyecameramanager /usr/bin/ueyedemo /etc/NetworkManager/dispatcher.d/ueyeethdnotify /etc/network/if-post-up.d/ueyeethdnotify /etc/network/if-pre-down.d/ueyeethdnotify /etc/wicd/scripts/postconnect//ueyeethdnotify /etc/wicd/scripts/predisconnect//ueyeethdnotify Done.
Remove sources directory...
Remove documentation directory...
Successfully uninstalled ueyed.
```

install ueye SDK 4.6(Latest version)，
```
$ sudo ./ueyesdk-setup-4.61-eth-amd64.gz.run
./ueyesdk-setup-4.61-eth-amd64.gz.runVerifying archive integrity... All good.
Uncompressing uEye SDK Linux (eth) v4.61 amd64........................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................
ueyed installer v1.1.17
Selecting install variant...
Will install eth 64bit
Installing eth driver (64bit)...
Checking prerequisites...
Create group 'ueye'...
Create user 'ueyed'...
Install documentation...
Install manual...
Create runtime directory...
Create binary directories...
Create library directory...
Install installer and tools to binary directory...
Install binary...
Install library...
Install header...
Install firmwares...
Install icons...
Install ueyed configuration file...
Creating configuration, output file is ./ueyeethd.conf.
Install ueye library configuration files...
Install runlevel control script...
Install sources...
Creating service start entry for ueyeethd... Done.
Run ldconfig to update library cache...
***************************************************************************
Successfully installed ueyeethd. Start (as root) with:
     /etc/init.d/ueyeethdrc start
 
By default, ueyeethd will work with all interfaces found during
installation.
To use dedicated interfaces only, edit (as root) the option:
		[Parameters] - Interfaces
at /usr/local/share/ueye/ueyeethd/ueyeethd.conf
one must stop the daemon before editing the configuration file. Alternatively,
the uEye Camera Manager may be used to configure network interfaces.
 
If ueyeethd hangs and does not terminate on rc 'stop' command, run (as root)
     /etc/init.d/ueyeethdrc force-stop
 
NOTE that the daemon refuses to terminate if there are clients connected!
To uninstall, run '/usr/local/share/ueye/bin/ueyed_install-eth uninstall'.
 
If a graphical desktop environment is available, one may use the
uEye Camera Manager to configure cameras and services. To start
it, type idscameramanager on the command line or in the graphical application
launcher. If idscameramanager shall be used for daemon configuration and service
control, it _must_ be started as root.
```

130s, 2016-01-15:
> Successfully installed ueyeethd. Start (as root) with:
>      /etc/init.d/ueyeethdrc start

Did you run this?

RyuYamamoto, 2016-01-15:
はい、実行しました

RyuYamamoto, 2016-01-15:
もう一度実行してみましたが駄目でした

```
roslaunch... logging to /home/haze/.ros/log/a8bef91a-bb2d-11e5-be14-705812de5ae5/roslaunch-haze-CF-S10EWHDS-20207.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://haze-CF-S10EWHDS:34377/

SUMMARY
========

PARAMETERS
 * /rosdistro: indigo
 * /rosversion: 1.11.13
 * /ueye_cam_nodelet_master/auto_exposure: False
 * /ueye_cam_nodelet_master/auto_frame_rate: False
 * /ueye_cam_nodelet_master/auto_gain: True
 * /ueye_cam_nodelet_master/auto_white_balance: True
 * /ueye_cam_nodelet_master/binning: 2
 * /ueye_cam_nodelet_master/blue_gain: 16
 * /ueye_cam_nodelet_master/camera_id: 1
 * /ueye_cam_nodelet_master/camera_intrinsics_file: 
 * /ueye_cam_nodelet_master/camera_name: right
 * /ueye_cam_nodelet_master/camera_parameters_file: 
 * /ueye_cam_nodelet_master/camera_topic: image_raw
 * /ueye_cam_nodelet_master/color_mode: rgb8
 * /ueye_cam_nodelet_master/exposure: 33
 * /ueye_cam_nodelet_master/ext_trigger_mode: False
 * /ueye_cam_nodelet_master/flash_delay: 30000
 * /ueye_cam_nodelet_master/flash_duration: 1000
 * /ueye_cam_nodelet_master/frame_rate: 30.0
 * /ueye_cam_nodelet_master/gain_boost: False
 * /ueye_cam_nodelet_master/green_gain: 1
 * /ueye_cam_nodelet_master/image_height: 1200
 * /ueye_cam_nodelet_master/image_left: -1
 * /ueye_cam_nodelet_master/image_top: -1
 * /ueye_cam_nodelet_master/image_width: 1600
 * /ueye_cam_nodelet_master/master_gain: 0
 * /ueye_cam_nodelet_master/pixel_clock: 24
 * /ueye_cam_nodelet_master/red_gain: 0
 * /ueye_cam_nodelet_master/sensor_scaling: 1.0
 * /ueye_cam_nodelet_master/subsampling: 1
 * /ueye_cam_nodelet_master/white_balance_blue_offset: 0
 * /ueye_cam_nodelet_master/white_balance_red_offset: 0
 * /ueye_cam_nodelet_slave1/auto_exposure: False
 * /ueye_cam_nodelet_slave1/auto_frame_rate: False
 * /ueye_cam_nodelet_slave1/auto_gain: True
 * /ueye_cam_nodelet_slave1/auto_white_balance: True
 * /ueye_cam_nodelet_slave1/binning: 2
 * /ueye_cam_nodelet_slave1/blue_gain: 16
 * /ueye_cam_nodelet_slave1/camera_id: 2
 * /ueye_cam_nodelet_slave1/camera_intrinsics_file: 
 * /ueye_cam_nodelet_slave1/camera_name: left
 * /ueye_cam_nodelet_slave1/camera_parameters_file: 
 * /ueye_cam_nodelet_slave1/camera_topic: image_raw
 * /ueye_cam_nodelet_slave1/color_mode: rgb8
 * /ueye_cam_nodelet_slave1/exposure: 33
 * /ueye_cam_nodelet_slave1/ext_trigger_mode: False
 * /ueye_cam_nodelet_slave1/flash_delay: 0
 * /ueye_cam_nodelet_slave1/flash_duration: 1000
 * /ueye_cam_nodelet_slave1/frame_rate: 30.0
 * /ueye_cam_nodelet_slave1/gain_boost: False
 * /ueye_cam_nodelet_slave1/green_gain: 1
 * /ueye_cam_nodelet_slave1/image_height: 1200
 * /ueye_cam_nodelet_slave1/image_left: -1
 * /ueye_cam_nodelet_slave1/image_top: -1
 * /ueye_cam_nodelet_slave1/image_width: 1600
 * /ueye_cam_nodelet_slave1/master_gain: 0
[... 39 lines cut ...]
[ERROR] [1452824712.945601623]: Hint: make sure that the IDS camera daemon (/etc/init.d/ueyeusbdrc) is running

[ERROR] [1452824712.946605865]: Failed to initialize [right]
ERROR: cannot launch node of type [stereo_image_proc/stereo_image_proc]: can't locate node [stereo_image_proc] in package [stereo_image_proc]
[ INFO] [1452824714.040040190]: Unable to open camera calibration file [/home/haze/.ros/camera_info/left.yaml]
process[head_link_to_stereo_link-9]: started with pid [22444]
process[stereo_link_to_stereo_frame-10]: started with pid [22634]
[ERROR] [1452824716.402583372]: No UEye cameras are connected

[ERROR] [1452824716.402772871]: Hint: make sure that the IDS camera daemon (/etc/init.d/ueyeusbdrc) is running

[ERROR] [1452824716.402890388]: Failed to initialize [left]
^C[stereo_link_to_stereo_frame-10] killing on exit
[head_link_to_stereo_link-9] killing on exit
[left_force_subscribe-4] killing on exit
[right_force_subscribe-5] killing on exit
[ueye_cam_nodelet_master-2] killing on exit
[ueye_cam_nodelet_slave1-3] killing on exit
[nodelet_manager-1] killing on exit
/opt/ros/indigo/lib/nodelet/nodelet: symbol lookup error: /opt/ros/indigo/lib/libbondcpp.so: undefined symbol: _ZN3ros9WallTimer9setPeriodERKNS_12WallDurationE
/opt/ros/indigo/lib/nodelet/nodelet: symbol lookup error: /opt/ros/indigo/lib/libbondcpp.so: undefined symbol: _ZN3ros9WallTimer9setPeriodERKNS_12WallDurationE
shutting down processing monitor...
... shutting down processing monitor complete

```

RyuYamamoto, 2016-01-15:
```
$ sudo /etc/init.d/ueyeethdrc start
/etc/init.d/ueyeethdrcStarting ueyeethd... IDS ueyeethd  4.61.3.0 0.19.5 1.6.0 64bit build Mar 27 2015 14:20:45 ( cmdl cap hll scq )
ueyeethd is running.
```

130s, 2016-01-15:
Turned out the camera is not connected to the machine. So error messages (`No UEye cameras are connected` and such) were already providing sufficient info.

closed 2016-01-15 by 130s