tork-a/rtmros_nextage #219: [ERROR] [1452874562.044205758]: Failed to load nodelet [/ueye_cam_nodelet_master] of type [ueye_cam/ueye_cam_nodelet]: According to the loaded plugin descriptions the class ueye_cam/ueye_cam_nodelet with base class type nodelet::Nodelet does not exist.

RyuYamamoto, 2016-01-15:
>$ roslaunch nextage_ros_bridge nextage_ueye_stereo.launch 
を起動した際以下のエラーが生じてしまいます．

```
... logging to /home/tork/.ros/log/44b041d8-bba3-11e5-8f7f-000bab925b83/roslaunch-nxconsole-12682.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://nxconsole:42817/

SUMMARY
========

PARAMETERS
 * /rosdistro
 * /rosversion
 * /ueye_cam_nodelet_master/auto_exposure
 * /ueye_cam_nodelet_master/auto_frame_rate
 * /ueye_cam_nodelet_master/auto_gain
 * /ueye_cam_nodelet_master/auto_white_balance
 * /ueye_cam_nodelet_master/binning
 * /ueye_cam_nodelet_master/blue_gain
 * /ueye_cam_nodelet_master/camera_id
 * /ueye_cam_nodelet_master/camera_intrinsics_file
 * /ueye_cam_nodelet_master/camera_name
 * /ueye_cam_nodelet_master/camera_parameters_file
 * /ueye_cam_nodelet_master/camera_topic
 * /ueye_cam_nodelet_master/color_mode
 * /ueye_cam_nodelet_master/exposure
 * /ueye_cam_nodelet_master/ext_trigger_mode
 * /ueye_cam_nodelet_master/flash_delay
 * /ueye_cam_nodelet_master/flash_duration
 * /ueye_cam_nodelet_master/frame_rate
 * /ueye_cam_nodelet_master/gain_boost
 * /ueye_cam_nodelet_master/green_gain
 * /ueye_cam_nodelet_master/image_height
 * /ueye_cam_nodelet_master/image_left
 * /ueye_cam_nodelet_master/image_top
 * /ueye_cam_nodelet_master/image_width
 * /ueye_cam_nodelet_master/master_gain
 * /ueye_cam_nodelet_master/pixel_clock
 * /ueye_cam_nodelet_master/red_gain
 * /ueye_cam_nodelet_master/sensor_scaling
 * /ueye_cam_nodelet_master/subsampling
 * /ueye_cam_nodelet_master/white_balance_blue_offset
 * /ueye_cam_nodelet_master/white_balance_red_offset
 * /ueye_cam_nodelet_slave1/auto_exposure
 * /ueye_cam_nodelet_slave1/auto_frame_rate
 * /ueye_cam_nodelet_slave1/auto_gain
 * /ueye_cam_nodelet_slave1/auto_white_balance
 * /ueye_cam_nodelet_slave1/binning
 * /ueye_cam_nodelet_slave1/blue_gain
 * /ueye_cam_nodelet_slave1/camera_id
 * /ueye_cam_nodelet_slave1/camera_intrinsics_file
 * /ueye_cam_nodelet_slave1/camera_name
 * /ueye_cam_nodelet_slave1/camera_parameters_file
 * /ueye_cam_nodelet_slave1/camera_topic
 * /ueye_cam_nodelet_slave1/color_mode
 * /ueye_cam_nodelet_slave1/exposure
 * /ueye_cam_nodelet_slave1/ext_trigger_mode
 * /ueye_cam_nodelet_slave1/flash_delay
 * /ueye_cam_nodelet_slave1/flash_duration
 * /ueye_cam_nodelet_slave1/frame_rate
 * /ueye_cam_nodelet_slave1/gain_boost
 * /ueye_cam_nodelet_slave1/green_gain
 * /ueye_cam_nodelet_slave1/image_height
 * /ueye_cam_nodelet_slave1/image_left
 * /ueye_cam_nodelet_slave1/image_top
 * /ueye_cam_nodelet_slave1/image_width
 * /ueye_cam_nodelet_slave1/master_gain
 * /ueye_cam_nodelet_slave1/pixel_clock
 * /ueye_cam_nodelet_slave1/red_gain
 * /ueye_cam_nodelet_slave1/sensor_scaling
 * /ueye_cam_nodelet_slave1/subsampling
 * /ueye_cam_nodelet_slave1/white_balance_blue_offset
 * /ueye_cam_nodelet_slave1/white_balance_red_offset

NODES
  /left/
    image_proc (image_proc/image_proc)
  /right/
    image_proc (image_proc/image_proc)
  /
    head_link_to_stereo_link (tf/static_transform_publisher)
    left_force_subscribe (rostopic/rostopic)
    nodelet_manager (nodelet/nodelet)
    right_force_subscribe (rostopic/rostopic)
    stereo_image_proc (stereo_image_proc/stereo_image_proc)
    stereo_link_to_stereo_frame (tf/static_transform_publisher)
    ueye_cam_nodelet_master (nodelet/nodelet)
    ueye_cam_nodelet_slave1 (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [12696]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 44b041d8-bba3-11e5-8f7f-000bab925b83
process[rosout-1]: started with pid [12709]
started core service [/rosout]
process[nodelet_manager-2]: started with pid [12721]
process[ueye_cam_nodelet_master-3]: started with pid [12722]
[ INFO] [1452874562.009551626]: Initializing nodelet with 8 worker threads.
[ERROR] [1452874562.044205758]: Failed to load nodelet [/ueye_cam_nodelet_master] of type [ueye_cam/ueye_cam_nodelet]: According to the loaded plugin descriptions the class ueye_cam/ueye_cam_nodelet with base class type nodelet::Nodelet does not exist. Declared types are  depth_image_proc/convert_metric depth_image_proc/disparity depth_image_proc/point_cloud_xyz depth_image_proc/point_cloud_xyzrgb depth_image_proc/register image_proc/crop_decimate image_proc/debayer image_proc/rectify image_rotate/image_rotate image_view/disparity image_view/image nodelet_tutorial_math/Plus openni2_camera/OpenNI2DriverNodelet pcl/BAGReader pcl/BoundaryEstimation pcl/ConvexHull2D pcl/CropBox pcl/EuclideanClusterExtraction pcl/ExtractIndices pcl/ExtractPolygonalPrismData pcl/FPFHEstimation pcl/FPFHEstimationOMP pcl/MomentInvariantsEstimation pcl/MovingLeastSquares pcl/NodeletDEMUX pcl/NodeletMUX pcl/NormalEstimation pcl/NormalEstimationOMP pcl/NormalEstimationTBB pcl/PCDReader pcl/PCDWriter pcl/PFHEstimation pcl/PassThrough pcl/PointCloudConcatenateDataSynchronizer pcl/PointCloudConcatenateFieldsSynchronizer pcl/PrincipalCurvaturesEstimation pcl/ProjectInliers pcl/SACSegmentation pcl/SACSegmentationFromNormals pcl/SegmentDifferences pcl/StatisticalOutlierRemoval pcl/VFHEstimation pcl/VoxelGrid pointcloud_to_laserscan/pointcloud_to_laserscan_nodelet stereo_image_proc/disparity stereo_image_proc/point_cloud stereo_image_proc/point_cloud2
[FATAL] [1452874562.044318465]: Service call failed!
process[ueye_cam_nodelet_slave1-4]: started with pid [12762]
[ERROR] [1452874562.079751660]: Failed to load nodelet [/ueye_cam_nodelet_slave1] of type [ueye_cam/ueye_cam_nodelet]: According to the loaded plugin descriptions the class ueye_cam/ueye_cam_nodelet with base class type nodelet::Nodelet does not exist. Declared types are  depth_image_proc/convert_metric depth_image_proc/disparity depth_image_proc/point_cloud_xyz depth_image_proc/point_cloud_xyzrgb depth_image_proc/register image_proc/crop_decimate image_proc/debayer image_proc/rectify image_rotate/image_rotate image_view/disparity image_view/image nodelet_tutorial_math/Plus openni2_camera/OpenNI2DriverNodelet pcl/BAGReader pcl/BoundaryEstimation pcl/ConvexHull2D pcl/CropBox pcl/EuclideanClusterExtraction pcl/ExtractIndices pcl/ExtractPolygonalPrismData pcl/FPFHEstimation pcl/FPFHEstimationOMP pcl/MomentInvariantsEstimation pcl/MovingLeastSquares pcl/NodeletDEMUX pcl/NodeletMUX pcl/NormalEstimation pcl/NormalEstimationOMP pcl/NormalEstimationTBB pcl/PCDReader pcl/PCDWriter pcl/PFHEstimation pcl/PassThrough pcl/PointCloudConcatenateDataSynchronizer pcl/PointCloudConcatenateFieldsSynchronizer pcl/PrincipalCurvaturesEstimation pcl/ProjectInliers pcl/SACSegmentation pcl/SACSegmentationFromNormals pcl/SegmentDifferences pcl/StatisticalOutlierRemoval pcl/VFHEstimation pcl/VoxelGrid pointcloud_to_laserscan/pointcloud_to_laserscan_nodelet stereo_image_proc/disparity stereo_image_proc/point_cloud stereo_image_proc/point_cloud2
[FATAL] [1452874562.079876350]: Service call failed!
process[left_force_subscribe-5]: started with pid [12779]
WARNING: topic [/left/camera_info] does not appear to be published yet
process[right_force_subscribe-6]: started with pid [12788]
WARNING: topic [/right/camera_info] does not appear to be published yet
[ueye_cam_nodelet_master-3] process has died [pid 12722, exit code 255, cmd /opt/ros/hydro/lib/nodelet/nodelet load ueye_cam/ueye_cam_nodelet nodelet_manager __name:=ueye_cam_nodelet_master __log:=/home/tork/.ros/log/44b041d8-bba3-11e5-8f7f-000bab925b83/ueye_cam_nodelet_master-3.log].
log file: /home/tork/.ros/log/44b041d8-bba3-11e5-8f7f-000bab925b83/ueye_cam_nodelet_master-3*.log
[ueye_cam_nodelet_slave1-4] process has died [pid 12762, exit code 255, cmd /opt/ros/hydro/lib/nodelet/nodelet load ueye_cam/ueye_cam_nodelet nodelet_manager __name:=ueye_cam_nodelet_slave1 __log:=/home/tork/.ros/log/44b041d8-bba3-11e5-8f7f-000bab925b83/ueye_cam_nodelet_slave1-4.log].
log file: /home/tork/.ros/log/44b041d8-bba3-11e5-8f7f-000bab925b83/ueye_cam_nodelet_slave1-4*.log
process[right/image_proc-7]: started with pid [12794]
process[left/image_proc-8]: started with pid [12826]
process[stereo_image_proc-9]: started with pid [12888]
process[head_link_to_stereo_link-10]: started with pid [13000]
process[stereo_link_to_stereo_frame-11]: started with pid [13098]

```

130s, 2016-01-15:
FYI [error in this thread](http://answers.ros.org/question/98861/how-to-use-ueye_cam-to-drive-my-ui1221-m-camera/) looks similar.

130s, 2016-01-15:
Please run the same command as https://github.com/tork-a/rtmros_nextage/issues/217#issuecomment-169929334

130s, 2016-01-15:
Also `rospack find ueye_cam` and `rosversion ueye_cam`.

130s, 2016-01-15:
Original poster was using 2 ROS hosts; host-(1) where ueye camera is connected to, and host-(2) where OP was trying to subscribe camera topics from.
Issue was worked around by simply using host-(1) for topic subscription.

Sounds like the cause of issue is a general ROS network issue http://wiki.ros.org/ROS/NetworkSetup.

closed 2016-01-15 by 130s