tork-a/rtmros_nextage #227: Servo goes off at the end of checkEncoders

130s, 2016-02-02:
After the 2nd "bump" motion at the end of `checkEncoders`, servo goes off and the green light at the chest starts blinking. Servo can be turned on by servoOn command from `NextageOpenSupervisor` (Kawada's proprietary tool).

I'm familliar with this phonomenon but for now can't find a ticket. Will keep looking.

```
$ rosversion hironx_ros_bridge
1.1.3
```

Content of `rtcd.log`.

```
hrpExecutionContext is registered
pdgains.file_name: /opt/jsk/etc/HIRONX/hrprtc/PDgains.sav
dof = 15
open_iob - NEXTAGE OPEN  I/F v1 instance at 0x80c1f70
-- NEXTAGE::OpenV10::open_iob()
-- NEXTAGE::OpenV10::open_iob - at 0x80ca078

-- NEXTAGE::OpenV10::open_iob(): succeeded, lockower is -1
the number of gyros = 0
the number of accelerometers = 0
the number of force sensors = 0
open_iob - NEXTAGE OPEN I/F instance is not NULL, ignored
-- NEXTAGE::OpenV10::lock_iob()
ShMIF::lock()
ShmIF::lock(): IOB_OK owner: 675872, self: 675872
period = 5[ms], priority = 49
SequencePlayer::onInitialize()
open_iob - NEXTAGE OPEN I/F instance is not NULL, ignored
-- NEXTAGE::OpenV10::lock_iob()
ShMIF::lock()
ShmIF::lock(): IOB_OK owner: 675872, self: 675872
period = 5[ms], priority = 49
StateHolder: dt = 0.005
open_iob - NEXTAGE OPEN I/F instance is not NULL, ignored
-- NEXTAGE::OpenV10::lock_iob()
ShMIF::lock()
ShmIF::lock(): IOB_OK owner: 675872, self: 675872
period = 5[ms], priority = 49
fk: onInitialize()
open_iob - NEXTAGE OPEN I/F instance is not NULL, ignored
-- NEXTAGE::OpenV10::lock_iob()
ShMIF::lock()
ShmIF::lock(): IOB_OK owner: 675872, self: 675872
period = 5[ms], priority = 49
;; Opening /dev/console for writing.
;;
el: onInitialize()
dof = 15
open_iob - NEXTAGE OPEN I/F instance is not NULL, ignored
-- NEXTAGE::OpenV10::lock_iob()
ShMIF::lock()
ShmIF::lock(): IOB_OK owner: 675872, self: 675872
period = 5[ms], priority = 49
log: onInitialize()
open_iob - NEXTAGE OPEN I/F instance is not NULL, ignored
-- NEXTAGE::OpenV10::lock_iob()
ShMIF::lock()
ShmIF::lock(): IOB_OK owner: 675872, self: 675872
period = 5[ms], priority = 49
-- NEXTAGE::OpenV10::unlock_iob()
ShMIF::unlock()
ShmIF::unlock(): IOB_OK owner: -1, self: 675872
-- NEXTAGE::OpenV10::unlock_iob()
ShMIF::unlock()
iob is unlocked by 675872
ShmIF::unlock(): IOB_OK owner: -1, self: 675872
-- NEXTAGE::OpenV10::unlock_iob()
ShMIF::unlock()
iob is unlocked by 675872
ShmIF::unlock(): IOB_OK owner: -1, self: 675872
-- NEXTAGE::OpenV10::unlock_iob()
ShMIF::unlock()
iob is unlocked by 675872
ShmIF::unlock(): IOB_OK owner: -1, self: 675872
-- NEXTAGE::OpenV10::unlock_iob()
ShMIF::unlock()
iob is unlocked by 675872
ShmIF::unlock(): IOB_OK owner: -1, self: 675872
SequencePlayer::onActivated(1000)
fk: onActivated(1000)
el: onActivated(1000)
DataLogger: unsupported data type(TimedLongSeqSeq)
[addJointGroup] group name TORSO is already installed
[addJointGroup] group name HEAD is already installed
[addJointGroup] group name RARM is already installed
[addJointGroup] group name LARM is already installed
readActualAngle:0=-0.0736311
readActualAngle:1=3.06796e-05
readActualAngle:2=0
readActualAngle:3=0.263841
readActualAngle:4=-0.0613508
readActualAngle:5=-0.901969
readActualAngle:6=0.355878
readActualAngle:7=0.0368155
readActualAngle:8=-0.40496
readActualAngle:9=0.362016
readActualAngle:10=-0.0245418
readActualAngle:11=-0.742437
readActualAngle:12=-0.282241
readActualAngle:13=-0.515418
readActualAngle:14=-0.245437
writeCommandAngle:0=-0.0736311
writeCommandAngle:1=3.06796e-05
writeCommandAngle:2=0
writeCommandAngle:3=0.263841
writeCommandAngle:4=-0.0613508
writeCommandAngle:5=-0.901969
writeCommandAngle:6=0.355878
writeCommandAngle:7=0.0368155
writeCommandAngle:8=-0.40496
writeCommandAngle:9=0.362016
writeCommandAngle:10=-0.0245418
writeCommandAngle:11=-0.742437
writeCommandAngle:12=-0.282241
writeCommandAngle:13=-0.515418
writeCommandAngle:14=-0.245437
doCalibration_startC:0
doCalibration_startC:1
doCalibration_startC:2
doCalibration_startC:3
doCalibration_startC:4
doCalibration_startC:5
doCalibration_startC:6
doCalibration_startC:7
doCalibration_startC:8
doCalibration_startC:9
doCalibration_startC:10
doCalibration_startC:11
doCalibration_startC:12
doCalibration_startC:13
doCalibration_startC:14
stageCalibration_finishedS .............. done.
loop 1
1:stageCalibration[0] = 2
1:stageCalibration[1] = 2
1:stageCalibration[2] = 2
1:stageCalibration[3] = 2
1:stageCalibration[4] = 2
1:stageCalibration[5] = 2
1:stageCalibration[6] = 2
1:stageCalibration[7] = 2
1:stageCalibration[8] = 2
1:stageCalibration[9] = 2
1:stageCalibration[10] = 2
1:stageCalibration[11] = 2
1:stageCalibration[12] = 2
1:stageCalibration[13] = 2
1:stageCalibration[14] = 2
doCalibration[0] = 255
doCalibration[1] = 255
doCalibration[2] = 255
doCalibration[3] = 255
doCalibration[4] = 255
doCalibration[5] = 255
doCalibration[6] = 255
doCalibration[7] = 255
doCalibration[8] = 255
doCalibration[9] = 255
doCalibration[10] = 255
doCalibration[11] = 255
doCalibration[12] = 255
doCalibration[13] = 255
doCalibration[14] = 255
loop 1
2:stageCalibration[0] = 255
2:stageCalibration[1] = 255
2:stageCalibration[2] = 255
2:stageCalibration[3] = 255
2:stageCalibration[4] = 255
2:stageCalibration[5] = 255
2:stageCalibration[6] = 255
2:stageCalibration[7] = 255
2:stageCalibration[8] = 255
2:stageCalibration[9] = 255
2:stageCalibration[10] = 255
2:stageCalibration[11] = 255
2:stageCalibration[12] = 255
2:stageCalibration[13] = 255
2:stageCalibration[14] = 255
readServoState . done.
Timeover: processing time = 13025.0[ms]
13025.0,  0.0,  0.0,  0.0,  0.0,  0.0, 
StateHolder::goActual()
```

130s, 2016-02-02:
On Ubuntu side, error message reads:

> Turning servos ({}) failed. This is likely because of issues happening in lower level. Please check if the Kawada's proprietary tool NextageOpenSupervisor returns without issue or not. If the issue persists, contact the manufacturer

Through `NextageOpenSupervisor`, servoOn, `initPose`, `servoOff` work.

130s, 2016-02-02:
`HIRONX.servoOn` does NOT work right after this issue occurred. 
`NextageOpenSupervisor's servoOn` does work.

@emijah

130s, 2016-02-02:
The phenomenon looks similar to https://github.com/start-jsk/hironx-package/issues/107, but the solution from there isn't applicable (2 CPUs are enabled already).

emijah, 2016-02-02:
Hmm. I'm not sure what is going on. 
Did you turn the servo drives on when the arms were still being pulled down by gravity?
If so, the encoders will show motion so the system will refuse to turn on if the difference is excessive.

130s, 2016-02-02:
Sounds like I'm forced to get out (for the lunch at customer's site) so will be back soon but I now suspect https://github.com/tork-a/rtmros_nextage/issues/153, hrpsys version might be old.

130s, 2016-02-02:
Sorry, there's a correction:

- WRONG: **After** the 2nd "bump" motion at the end of checkEncoders, servo goes off and the green light at the chest starts blinking. 
- CORRECT: **Before** the 2nd "bump" motion at the end of checkEncoders, servo goes off and the green light at the chest starts blinking.

130s, 2016-02-02:
>  Did you turn the servo drives on when the arms were still being pulled down by gravity?

No. Arms are sitting on the pedestal whenever I've been trying to servoOn from `NextageOpenSupervisor`.

130s, 2016-02-02:
And #153 shouldn't be influencing this particular issue (since #153 is about the height of the robot model).

emijah, 2016-02-02:
Can you switch the servos on after waiting a while?

emijah, 2016-02-02:
Ah  just re-read this.

> HIRONX.servoOn does NOT work right after this issue occurred.
> NextageOpenSupervisor's servoOn does work.

emijah, 2016-02-02:
Is rtcd RobotHardware sitting on top of libiob.so or libNextageOpenCore.so?

130s, 2016-02-02:
>>HIRONX.servoOn does NOT work right after this issue occurred.

More precisely, the following's what happens. BUT, the chest lump turns to green, no blink.

```
In [2]: robot.servoOn()
Move to Actual State, Just a minute.
Turn on Hand Servo
hrpsys ServoController not found. You may want to restart QNX/ControllerBox afterward and run the command again.
Out[2]: -1
```

emijah, 2016-02-02:
Ah, so the robot servo is on?
If the robot doesn't have Futaba servo drive hands, the ServoController RTC cannot find anything to turn on. This will return an error.
If there are no hands or if the hands are not Futaba servo drive hands, you should use the NextageOpen controller instead.

130s, 2016-02-02:
Sorry, I'm confused. This is NEXTAGE Open, so it should have not be shipped with Futaba servo AFAIK.

emijah, 2016-02-02:
That's right. I think that is the reason the controller is complaining that the ServoController can't work. There are no servo drives for it to talk to, so the servo cannot be switched on. 
What if you use the NextageOpen class instead of HiroNX?

emijah, 2016-02-02:
The ServoController RTC is a dedicated software controller for Futaba servo drives connected with RS485 only.

130s, 2016-02-03:
This also happens with a Hironx, at a different location from the original post. 
```
$ rosversion hironx_ros_bridge
1.1.5
```

And, from `hironx_ros_bridge` 1.0.31 this seems like NOT occurring.

closed 2016-02-03 by 130s