tork-a/rtmros_nextage #229: IK in setTargetPoseRelative fails for head joints in certain configuration

harumo11, 2016-02-02:
I'm trying to move head (via joystick using #228 btw). In this particular case, robot faces toward left-bottom. IK failure occurs in the particular order of commands:
- Head down (giving negative pitch) then roll toward left (positive yaw) --> IK **failure**
- Opposite order (roll left then head down) --> **no failure**

[This link](http://1drv.ms/1WZhIzO) you can find the screenshot video and a rosbag file that includes `joy` topics so that you can hopefully replicate the issue.

k-okada, 2016-09-05:
sorry for not looking this, could re-enable the like so that I can re-produce this? thanks in advance.

k-okada, 2016-09-05:
I think I can reproduce the problem, 
```
In [34]:  robot.setTargetPose('head', [0,0,0.5695], [0,0,0.1], 1)
('head', None, [0, 0, 0.5695], [0, 0, 0.1], 1)
Out[34]: True

In [35]:  robot.setTargetPose('head', [0,0,0.5695], [0,0,0.5], 1)
('head', None, [0, 0, 0.5695], [0, 0, 0.5], 1)
Out[35]: True

In [36]:  robot.setTargetPose('head', [0,0,0.5695], [0,0,0.1], 1)
('head', None, [0, 0, 0.5695], [0, 0, 0.1], 1)
Out[36]: True

In [37]:  robot.setTargetPose('head', [0,0,0.5695], [0,0.5,0.1], 1)
('head', None, [0, 0, 0.5695], [0, 0.5, 0.1], 1)
Out[37]: True

In [38]:  robot.setTargetPose('head', [0,0,0.5695], [0,-0.5,-0.1], 1)
('head', None, [0, 0, 0.5695], [0, -0.5, -0.1], 1)
setTargetPose failed. Maybe SequencePlayer failed to solve IK.
Currently, returning IK result error
(like the one in https://github.com/start-jsk/rtmros_hironx/issues/103) is not implemented. Patch is welcomed.
Out[38]: False

In [39]:  robot.setTargetPose('head', [0,0,0.5695], [0,-0.5,0], 1)
('head', None, [0, 0, 0.5695], [0, -0.5, 0], 1)
setTargetPose failed. Maybe SequencePlayer failed to solve IK.
Currently, returning IK result error
(like the one in https://github.com/start-jsk/rtmros_hironx/issues/103) is not implemented. Patch is welcomed.
Out[39]: False

In [40]:  robot.setTargetPose('head', [0,0,0.5695], [0,0,0], 1)
('head', None, [0, 0, 0.5695], [0, 0, 0], 1)
setTargetPose failed. Maybe SequencePlayer failed to solve IK.
Currently, returning IK result error
(like the one in https://github.com/start-jsk/rtmros_hironx/issues/103) is not implemented. Patch is welcomed.
Out[40]: False

In [41]:  robot.setTargetPose('head', [0,0,0.5695], [0,0,-1], 1)
('head', None, [0, 0, 0.5695], [0, 0, -1], 1)
setTargetPose failed. Maybe SequencePlayer failed to solve IK.
Currently, returning IK result error
(like the one in https://github.com/start-jsk/rtmros_hironx/issues/103) is not implemented. Patch is welcomed.
Out[41]: False
```
```
In [14]:  robot.setTargetPoseRelative('head', 'HEAD_JOINT1', dw=1)
(array([ 0.    ,  0.    ,  0.5695]), (0.0, -0.0, 0.0998499033367421))
('head', None, [0.0, 0.0, 0.5694999999999999], [0.0, -0.0, 0.0998499033367421], 10)
Out[14]: True

In [15]:  robot.setTargetPoseRelative('head', 'HEAD_JOINT1', dp=0.5)
(array([ 0.    ,  0.    ,  0.5695]), (0.0, 0.49999999999999994, 0.09968831468369613))
('head', None, [0.0, 0.0, 0.5694999999999999], [0.0, 0.49999999999999994, 0.09968831468369613], 10)
Out[15]: True

In [16]:  robot.setTargetPoseRelative('head', 'HEAD_JOINT1', dp=-0.5)
(array([ 0.    ,  0.    ,  0.5695]), (0.0, -0.0006031484088019347, 0.09968831468369614))
('head', None, [0.0, 0.0, 0.5694999999999999], [0.0, -0.0006031484088019347, 0.09968831468369614], 10)

Out[16]: True

In [17]: 

In [17]:  robot.setTargetPoseRelative('head', 'HEAD_JOINT1', dp=0.5, dw=1)
(array([ 0.    ,  0.    ,  0.5695]), (1.505795270667706e-05, 0.5000084849079418, 1.0996955339068453))
('head', None, [0.0, 0.0, 0.5694999999999999], [1.505795270667706e-05, 0.5000084849079418, 1.0996955339068453], 10)
setTargetPose failed. Maybe SequencePlayer failed to solve IK.
Currently, returning IK result error
(like the one in https://github.com/start-jsk/rtmros_hironx/issues/103) is not implemented. Patch is welcomed.
Out[17]: False

In [18]:  robot.setTargetPoseRelative('head', 'HEAD_JOINT1', dp=0.5, dw=0.5)
(array([ 0.    ,  0.    ,  0.5695]), (8.57924765896957e-06, 0.5000137816194049, 0.5996924278460067))
('head', None, [0.0, 0.0, 0.5694999999999999], [8.57924765896957e-06, 0.5000137816194049, 0.5996924278460067], 10)
setTargetPose failed. Maybe SequencePlayer failed to solve IK.
Currently, returning IK result error
(like the one in https://github.com/start-jsk/rtmros_hironx/issues/103) is not implemented. Patch is welcomed.
Out[18]: False

In [19]:  robot.setTargetPoseRelative('head', 'HEAD_JOINT1', dp=0.1, dw=0.5)
(array([ 0.    ,  0.    ,  0.5695]), (7.566754118113806e-06, 0.10001378163204483, 0.5996890701504933))
('head', None, [0.0, 0.0, 0.5694999999999999], [7.566754118113806e-06, 0.10001378163204483, 0.5996890701504933], 10)
setTargetPose failed. Maybe SequencePlayer failed to solve IK.
Currently, returning IK result error
(like the one in https://github.com/start-jsk/rtmros_hironx/issues/103) is not implemented. Patch is welcomed.
Out[19]: False

In [20]:  robot.setTargetPoseRelative('head', 'HEAD_JOINT1', dp=0.1, dw=0.1)
(array([ 0.    ,  0.    ,  0.5695]), (1.5756671171830663e-06, 0.10001562563410654, 0.1996884720001768))
('head', None, [0.0, 0.0, 0.5694999999999999], [1.5756671171830663e-06, 0.10001562563410654, 0.1996884720001768], 10)
setTargetPose failed. Maybe SequencePlayer failed to solve IK.
Currently, returning IK result error
(like the one in https://github.com/start-jsk/rtmros_hironx/issues/103) is not implemented. Patch is welcomed.
Out[20]: False

In [21]:  robot.setTargetPoseRelative('head', 'HEAD_JOINT1', dw=0.1, dp=0.1)
(array([ 0.    ,  0.    ,  0.5695]), (1.5756671171830663e-06, 0.10001562563410654, 0.1996884720001768))
('head', None, [0.0, 0.0, 0.5694999999999999], [1.5756671171830663e-06, 0.10001562563410654, 0.1996884720001768], 10)
setTargetPose failed. Maybe SequencePlayer failed to solve IK.
Currently, returning IK result error
(like the one in https://github.com/start-jsk/rtmros_hironx/issues/103) is not implemented. Patch is welcomed.
Out[21]: False
```

k-okada, 2016-09-05:
Work around for this is to set wider ik error, as follows.

```
robot.seq_svc.setMaxIKError(0.001, 0.001)
```

```

In [10]:  robot.setTargetPose('head', [0,0,0.5695], [0,0,0.1], 1)
('head', None, [0, 0, 0.5695], [0, 0, 0.1], 1)
Out[10]: True

In [11]:  robot.setTargetPose('head', [0,0,0.5695], [0,0,0.5], 1)
('head', None, [0, 0, 0.5695], [0, 0, 0.5], 1)
Out[11]: True

In [12]:  robot.setTargetPose('head', [0,0,0.5695], [0,0.5,0.5], 1)
('head', None, [0, 0, 0.5695], [0, 0.5, 0.5], 1)
Out[12]: True

In [13]:  robot.setTargetPose('head', [0,0,0.5695], [0,0.5,-0.5], 1)
('head', None, [0, 0, 0.5695], [0, 0.5, -0.5], 1)
Out[13]: True

In [14]:  robot.setTargetPose('head', [0,0,0.5695], [0,-0.5,-0.5], 1)
('head', None, [0, 0, 0.5695], [0, -0.5, -0.5], 1)
setTargetPose failed. Maybe SequencePlayer failed to solve IK.
Currently, returning IK result error
(like the one in https://github.com/start-jsk/rtmros_hironx/issues/103) is not implemented. Patch is welcomed.
Out[14]: False

In [15]:  robot.setTargetPose('head', [0,0,0.5695], [0,0,-0.5], 1)
('head', None, [0, 0, 0.5695], [0, 0, -0.5], 1)
Out[15]: True

In [16]:  robot.setTargetPose('head', [0,0,0.5695], [0,0,-0.5], 1)

```