tork-a/rtmros_nextage #232: Inconsistency between RVIZ/Moveit by adding linear axis to Nextage

hectorherrero, 2016-02-10:
Hi, 

We are trying to put the Nextage robot on to a linear axis.

But we are detecting strange issues in RVIZ/MoveIt simulations. Our first idea was to create virtual transform frames and change TF tree as follows:
- world
     - robot_base
        - odom
             - robot_footprint
                - WAIST

With a simple publisher we have robot TF tree (WAIST, CHEST_JOINT0, RARM_JOINT0, etc.) moving in RVIZ. The problem comes when Moveit is used, we have detected an incoherence between RobotModel and Planning Scene. TF tree moves with RobotModel but another model is visualised in the RVIZ (see attached pictures). We think that moveit planning scene is not updating correctly.

[image]
[image]

Our first idea is to modify the URDF adding a prismatic joint and update nextage_moveit_config, but this probably requires to remapping /joint_states with aditional joint value of the linear axis, and this is very close to hrpsys (is it really necessary?).

As can be seen, we are trying different things but without too much criteria. Any indication would be grateful.

Thanks, 

Héctor

130s, 2016-02-18:
I may be misunderstanding something but I haven't been able to replicate your issue. I've done:

```
term-1$ rtmlaunch nextage_ros_bridge nextage_ros_bridge_simulation.launch

term-2$ 
 (With `tf` plugin on on RViz)

term-3$ rosrun tf static_transform_publisher 0 1 0 0 0 0 /world /WAIST 1
```

I see:
[image]

Switching the `fixed frame` from WAIST to `world`, then I see:
[image]

(For now forget about the interactive marker that dis/appears. It's separately discussed in #233)

Then adding a little more y value makes the robot move.

```
term-3$ rosrun tf static_transform_publisher 0 1.5 0 0 0 0 /world /WAIST 1
```

[image]

I might also have to disable/enable MotionPlanning plugin on RViz to make the visual properly.

Not entirely sure if this is what you wanted but how does this looks for you?

If this isn't the issue you're talking about, then please provide the complete steps to replicate it.

And regardless, we support your idea to add another axes, since `robot_state_publisher` reliably handles `tf` no matter how many nodes that publish joint states. No need to modify hrpsys for customized purpose.

hectorherrero, 2016-02-18:
Hi, 

yes this is the same issue that we are talking about. In this moment if you add RobotModel plugging to RVIZ you will see that RobotModel appears on the `tf` skeleton.

> I might also have to disable/enable MotionPlanning plugging on RViz to make the visual properly.

I have realized that this process of disable/enable is what we are looking for. If I disable and enable again from RVIZ plugging the robot model of the Planning Scene appears in the same (in the right position for us) position than RobotModel and `tf` skeleton. But...is possible to do that from code? I have looked MoveIt services but I haven't found any.

Thanks, 

Héctor

hectorherrero, 2016-05-05:
Hi, 

I will close this issue since I have solved the problem correctly setting the following:

**In hrpsys_ros_bridge/launch/hrpsys_ros_bridge.launch:**
Change:
`<arg name="USE_ROBOT_POSE_EKF" default="true" />`
To:
`<arg name="USE_ROBOT_POSE_EKF" default="false" />`
Add remap to:
```
<node name = "HrpsysSeqStateROSBridge"
        pkg  = "hrpsys_ros_bridge"
        type = "HrpsysSeqStateROSBridge"
        args = '$(arg openrtm_args) -o model:$(arg MODEL_FILE) -o example.HrpsysSeqStateROSBridge.config_file:$(arg CONF_FILE)'
        output = "$(arg OUTPUT)">
    <param name="publish_sensor_transforms" value="$(arg PUBLISH_SENSOR_TF)" />
  </node>
```

So:
```
<node name = "HrpsysSeqStateROSBridge"
        pkg  = "hrpsys_ros_bridge"
        type = "HrpsysSeqStateROSBridge"
        args = '$(arg openrtm_args) -o model:$(arg MODEL_FILE) -o example.HrpsysSeqStateROSBridge.config_file:$(arg CONF_FILE)'
        output = "$(arg OUTPUT)">
    <remap from="joint_states" to="robot_joint_states"/>
    <param name="publish_sensor_transforms" value="$(arg PUBLISH_SENSOR_TF)" />
  </node>
```

**A node which publishes robot_base frame.** This frame represents the position of the linear axis in the world frame.

**In our application launch file is necessary to add the following**:

```
  <node name="joint_state_publisher" pkg="joint_state_publisher" type="joint_state_publisher">
    <rosparam param="source_list">["robot_base", "joint_states_robot"]</rosparam>
    <param name="use_gui" value="false"/>
    <param name="robot_description" textfile="$(find nextage_description)/urdf/NextageOpen.urdf"/>
  </node>
```

With this we are publishing previously remapped robot joint_states and created robot_base frame  into /joint_states topic

I hope this can be useful for someone.

Héctor

closed 2016-05-05 by hectorherrero

130s, 2016-05-09:
Glad you found a solution!

>     <remap from="joint_states" to="robot_joint_states"/>

Isn't that `joint_states_robot`, according to your application launch file?

hectorherrero, 2016-05-09:
You are right, thank you!

I have edited the answer.