tork-a/rtmros_nextage #234: Robot state out of sync with Moveit!

JKBehrens, 2016-03-03:
Hello,

I try to control a real KaWaDa Nextage Open Robot via Moveit!. Therefore, I start the rosbridge with

`roslaunch nextage_ros_bridge nextage_ros_bridge_real.launch nameserver:=nextage`

which gives me the following output:

> nxouser@nxconsole:/opt/ros/hydro/share/nextage_ros_bridge$ roslaunch nextage_ros_bridge nextage_ros_bridge_real.launch nameserver:=nextage
> ... logging to /home/nxouser/.ros/log/af4b6fb4-e140-11e5-ae15-000bab77d794/roslaunch-nxconsole-13395.log
> Checking log directory for disk usage. This may take awhile.
> Press Ctrl-C to interrupt
> Done checking log file disk usage. Usage is <1GB.
> 
> started roslaunch server http://nxconsole:58263/
> 
> SUMMARY
> ========
> 
> PARAMETERS
>  * /collision_state/comp_name
>  * /controller_configuration
>  * /diagnostic_aggregator/analyzers/computers/contains
>  * /diagnostic_aggregator/analyzers/computers/path
>  * /diagnostic_aggregator/analyzers/computers/type
>  * /diagnostic_aggregator/analyzers/hrpsys/contains
>  * /diagnostic_aggregator/analyzers/hrpsys/path
>  * /diagnostic_aggregator/analyzers/hrpsys/type
>  * /diagnostic_aggregator/analyzers/mode/contains
>  * /diagnostic_aggregator/analyzers/mode/path
>  * /diagnostic_aggregator/analyzers/mode/type
>  * /diagnostic_aggregator/analyzers/motor/contains
>  * /diagnostic_aggregator/analyzers/motor/path
>  * /diagnostic_aggregator/analyzers/motor/type
>  * /diagnostic_aggregator/base_path
>  * /diagnostic_aggregator/pub_rate
>  * /robot_description
>  * /robot_pose_ekf/debug
>  * /robot_pose_ekf/freq
>  * /robot_pose_ekf/imu_used
>  * /robot_pose_ekf/odom_used
>  * /robot_pose_ekf/output_frame
>  * /robot_pose_ekf/self_diagnose
>  * /robot_pose_ekf/sensor_timeout
>  * /robot_pose_ekf/vo_used
>  * /rosdistro
>  * /rosversion
> 
> NODES
>   /
>     CollisionDetectorComp (hrpsys/CollisionDetectorComp)
>     DataLoggerServiceROSBridge (hrpsys_ros_bridge/DataLoggerServiceROSBridgeComp)
>     ForwardKinematicsServiceROSBridge (hrpsys_ros_bridge/ForwardKinematicsServiceROSBridgeComp)
>     HrpsysJointTrajectoryBridge (hrpsys_ros_bridge/HrpsysJointTrajectoryBridge)
>     HrpsysSeqStateROSBridge (hrpsys_ros_bridge/HrpsysSeqStateROSBridge)
>     RobotHardwareServiceROSBridge (hrpsys_ros_bridge/RobotHardwareServiceROSBridgeComp)
>     SequencePlayerServiceROSBridge (hrpsys_ros_bridge/SequencePlayerServiceROSBridgeComp)
>     StateHolderServiceROSBridge (hrpsys_ros_bridge/StateHolderServiceROSBridgeComp)
>     base_footprint_rootlink (tf/static_transform_publisher)
>     collision_state (hrpsys_ros_bridge/collision_state.py)
>     diagnostic_aggregator (diagnostic_aggregator/aggregator_node)
>     hrpsys_profile (hrpsys_ros_bridge/hrpsys_profile.py)
>     hrpsys_ros_diagnostics (hrpsys_ros_bridge/diagnostics.py)
>     hrpsys_state_publisher (robot_state_publisher/state_publisher)
>     robot_pose_ekf (robot_pose_ekf/robot_pose_ekf)
>     rtmlaunch_collision_detector (openrtm_tools/rtmlaunch.py)
>     rtmlaunch_hrpsys_ros_bridge (openrtm_tools/rtmlaunch.py)
>     sensor_ros_bridge_connect (hrpsys_ros_bridge/sensor_ros_bridge_connect.py)
> 
> WARNING: unrecognized tag rtconnect
[... 427 lines cut ...]
> [ WARN] [1457010794.751307824]: The root link WAIST has an inertia specified in the URDF, but KDL does not support a root link with an inertia.  As a workaround, you can add an extra dummy link to your URDF.
> [ INFO] [1457010794.756440088]: Constructing new MoveGroup connection for group 'right_arm'
> [ INFO] [1457010795.547280522]: Ready to take MoveGroup commands for group right_arm.
> [ INFO] [1457010795.547486131]: Looking around: no
> [ INFO] [1457010795.547612602]: Replanning: no

In the RViz user interface it is stated that
`Global Status: ERROR`
and that the 
`Fixed Frame [Waist] does not exist`

Routing down the problem:
- Checking rqt gives the insight that the tf-tree is noch build up properly 
- Checking the topic /joint_states shows that the topic is advertised but there are no send messages - Pinging the rosnode /HrpsysSeqStateROSBridge shows that it is alive

BUT: Checking the Rosbridge reveals the following:

`[rtmlaunch] Wait for  /nextage:15005/sh.rtc:StateHolderService   29 /30`
and
`[ WARN] [1457012235.568578397]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is not executed last 5[sec]`

I suspect that the StateHolderService inside the QNX-PC is somehow not connected or running properly. Since I am able to read the joint states from the python terminal (which seems to directly use a hrpsys interface), the problem has to be inside the ros bridge. Somehow the mapping onto the the right topics fails when working with the real robot.

Thank you for your help!
JK

k-okada, 2016-03-04:
> [rtmlaunch] Issue related to this start-jsk/rtmros_hironx#33
> CORBA.COMPLETED_NO) in > ignored
> start-jsk/rtmros_common#247
> [sensor_ros_bridge_connect.py] wait for RobotHardware0 : (timeout 0 < 10)
> [sensor_ros_bridge_connect.py] findComps -> RobotHardware : isActive? =
> CORBA.COMPLETED_NO) in > ignored [rtmlaunch] collision_detector.launch did
> CORBA.COMPLETED_NO) in > ignored Exception AttributeError: "'RTCTree'

can you ping nextage?

JKBehrens, 2016-03-04:
Hello k-okada,

yes, I am and I was able to ping the robot using either 
`ping nextage`
or 
`ping 192.168.128.10`

Actually, we made some nice progress yesterday. Perhaps, someone can add more information to the rtmros_nextage tutorials to save time for future users.

What made a difference was to use eth0 network interface to connect to the robot. This was set up differently by the KaWaDa support. Also the configuration in /etc/network/interfaces was accordingly. We changed and rewired the setup here. My new `/etc/network/interfaces` is this:

`auto lo
iface lo inet loopback

auto eth0
iface eth0 inet static
network 192.168.128.0
address 192.168.128.50
netmask 255.255.255.0
broadcast 192.168.128.255`

This change solved some of the problems with the ROS Bridge startup. An Update of the nextage debian packages then solved the rest. 

The much cleaner startup log looks like the following:

`nxouser@nxconsole:~$ roslaunch nextage_ros_bridge nextage_ros_bridge_real.launch 
... logging to /home/nxouser/.ros/log/d7b2ef4e-e1ff-11e5-9073-000bab77d794/roslaunch-nxconsole-5649.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://nxconsole:41219/

SUMMARY
========

PARAMETERS
 * /HrpsysSeqStateROSBridge/publish_sensor_transforms
 * /collision_state/comp_name
 * /controller_configuration
 * /diagnostic_aggregator/analyzers/computers/contains
 * /diagnostic_aggregator/analyzers/computers/path
 * /diagnostic_aggregator/analyzers/computers/type
 * /diagnostic_aggregator/analyzers/hrpsys/contains
 * /diagnostic_aggregator/analyzers/hrpsys/path
 * /diagnostic_aggregator/analyzers/hrpsys/type
 * /diagnostic_aggregator/analyzers/mode/contains
 * /diagnostic_aggregator/analyzers/mode/path
 * /diagnostic_aggregator/analyzers/mode/type
 * /diagnostic_aggregator/analyzers/motor/contains
 * /diagnostic_aggregator/analyzers/motor/path
 * /diagnostic_aggregator/analyzers/motor/type
 * /diagnostic_aggregator/base_path
 * /diagnostic_aggregator/pub_rate
 * /robot_description
 * /robot_pose_ekf/debug
 * /robot_pose_ekf/freq
 * /robot_pose_ekf/imu_used
 * /robot_pose_ekf/odom_used
 * /robot_pose_ekf/output_frame
 * /robot_pose_ekf/self_diagnose
 * /robot_pose_ekf/sensor_timeout
 * /robot_pose_ekf/vo_used
 * /rosdistro
 * /rosversion

NODES
  /
    CollisionDetectorComp (hrpsys/CollisionDetectorComp)
    DataLoggerServiceROSBridge (hrpsys_ros_bridge/DataLoggerServiceROSBridgeComp)
[... 273 lines cut ...]
[hrpsys.py]   setupLogger : q arleady exists in DataLogger
[rtm.py]    Connect ic.q - log.ic_q
[hrpsys.py]   setupLogger : emergencySignal arleady exists in DataLogger
[rtm.py]    Connect RobotHardware0.emergencySignal - log.emergencySignal
[hrpsys.py]   setupLogger : servoState arleady exists in DataLogger
[rtm.py]    Connect RobotHardware0.servoState - log.RobotHardware0_servoState
[hrpsys.py] setup joint groups
[hrpsys.py] initialized successfully
[ INFO] [1457092470.820488413]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 201[Hz] (exec 1775 Hz, dropped 201)
[INFO] [WallTime: 1457092471.029166] Joint names; Torso: ['CHEST_JOINT0']
	Head: ['HEAD_JOINT0', 'HEAD_JOINT1']
	L-Arm: ['LARM_JOINT0', 'LARM_JOINT1', 'LARM_JOINT2', 'LARM_JOINT3', 'LARM_JOINT4', 'LARM_JOINT5']
	R-Arm: ['RARM_JOINT0', 'RARM_JOINT1', 'RARM_JOINT2', 'RARM_JOINT3', 'RARM_JOINT4', 'RARM_JOINT5']
[WARN] [WallTime: 1457092471.031876] Moveit is not started yet, if you want to use MoveIt!
Make sure you've launched MoveGroup (e.g. by launching moveit_planning_execution.launch)
[hrpsys_py-2] process has finished cleanly
log file: /home/nxouser/.ros/log/d7b2ef4e-e1ff-11e5-9073-000bab77d794/hrpsys_py-2*.log
[ INFO] [1457092471.821263439]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 200[Hz] (exec 1743 Hz, dropped 200)
^C[collision_state-20] killing on exit
`

I next try to do the same setup under Ubuntu 14.04 and with ROS Indigo. If I have some more insights or questions, I will post them here or open a new issue.

Thank you so far.
Jan

130s, 2016-03-07:
@JKBehrens Glad to know you've made it working!

Though very much hidden, `eth0` requirement [has been documented](http://wiki.ros.org/rtmros_nextage/Tutorials/Known%20Troubleshoots#No_or_less_tf_published). This time I also [added a note at a more general instruction page](http://wiki.ros.org/rtmros_nextage/Tutorials/Install%20NEXTAGE%20OPEN%20software%20on%20your%20machine#Working_with_a_real_robot).

closed 2016-03-07 by 130s