tork-a/rtmros_nextage #235: Interactive marker may or may not dissappear

130s, 2016-03-11:
Following the issue that is originally addressed in the PR https://github.com/tork-a/rtmros_nextage/pull/233, I confirmed:
- that the issue still occurs with the latest public-synced debs, on my machine.
- that this PR solves the issue for my machine.
- that on the new machine where ROS was just freshly installed, this issue does NOT occur.

So it's probably my machine. But the question remains; what's wrong?

```
more /etc/apt/sources.list.d/ros-latest.list
deb http://packages.ros.org/ros/ubuntu trusty main
sudo apt-get purge ros-indigo-* && sudo apt-get autoremove
sudo apt-get update && sudo apt-get install ros-indigo-rtmros-nextage 
dpkg -p ros-indigo-moveit-ros ros-indigo-moveit-ros-planning ros-indigo-nextage-ros-bridge ros-indigo-hrpsys-ros-bridge |grep Ver
Version: 0.7.0-0trusty-20160201-120956-0800
Version: 0.7.0-0trusty-20160201-072110-0800
Version: 0.7.5-0trusty-20160208-142556-0800
Version: 1.3.1-0trusty-20160208-125213-0800
$ env|grep ROS
ROS_ROOT=/opt/ros/indigo/share/ros
ROS_PACKAGE_PATH=/opt/ros/indigo/share:/opt/ros/indigo/stacks
ROS_MASTER_URI=http://localhost:11311
LD_LIBRARY_PATH=/opt/ros/indigo/share/euslisp/jskeus/eus//Linux64/lib:/opt/ros/indigo/lib/x86_64-linux-gnu:/opt/ros/indigo/share/euslisp/jskeus/eus//Linux64/lib:/opt/ros/indigo/share/euslisp/jskeus/eus//Linux64/lib:/opt/ros/indigo/share/euslisp/jskeus/eus//Linux64/lib:/home/rospasta/cws_hironxo/install/lib:/home/rospasta/cws_hironxo/devel/lib:/opt/ros/indigo/share/euslisp/jskeus/eus//Linux64/lib:/opt/ros/indigo/lib
ROS_DISTRO=indigo
PKG_CONFIG_PATH=/opt/ros/indigo/lib/x86_64-linux-gnu/pkgconfig:/home/rospasta/cws_hironxo/install/lib/pkgconfig:/home/rospasta/cws_hironxo/devel/lib/pkgconfig:/opt/ros/indigo/lib/pkgconfig
ROS_ETC_DIR=/opt/ros/indigo/etc/ros

term-1$ rtmlaunch nextage_ros_bridge nextage_ros_bridge_simulation.launch 
term-2$ roslaunch nextage_moveit_config moveit_planning_execution.launch 
```

130s, 2016-03-11:
I have to cross out this line, because I found that they are using older version `0.6.2`. So, I don't know if this issue happens to only me or not.

><s>that on the new machine where ROS was just freshly installed, this issue does NOT occur.</s>

k-okada, 2016-03-12:
how about removing cached config, i.e. 'rm -fr ~/.ros/'

also double chekc .bashrc, you still have local path for LD_LIBRARY_PATH 
and PKG_CONFIG_PATH
LD_LIBRARY_PATH=/home/rospasta/cws_hironxo/install/lib:/home/rospasta/cws_hironxo/devel/lib:
PKG_CONFIG_PATH=/home/rospasta/cws_hironxo/install/lib/pkgconfig

130s, 2016-03-13:
No good...
```
$ env|grep ROS
ROS_ROOT=/opt/ros/indigo/share/ros
ROS_PACKAGE_PATH=/opt/ros/indigo/share:/opt/ros/indigo/stacks
ROS_MASTER_URI=http://localhost:11311
ROS_DISTRO=indigo
ROS_ETC_DIR=/opt/ros/indigo/etc/ros
$ ll ~/.ros && du -h -d 1 ~/.ros
total 28K
drwxr-xr-x  4 rospasta rospasta 4.0K Mar 12 18:25 log/
drwxr-xr-x  3 rospasta rospasta 4.0K Mar 12 18:25 ./
-rw-------  1 rospasta rospasta 6.6K Mar 12 18:25 rospack_cache_11486458967921186402
-rw-------  1 rospasta rospasta 7.9K Mar 12 18:23 rospack_cache_16324185631471125104
drwxr-xr-x 67 rospasta rospasta 4.0K Mar 12 18:23 ../
1.1M	/home/rospasta/.ros/log
1.1M	/home/rospasta/.ros
```

harumo11, 2016-04-02:
Hello.
On my PC. (Linux mint17.3)Resulted in following.

$uname -a
Linux harumo-ThinkPad-X250 3.19.0-32-generic #37~14.04.1-Ubuntu SMP Thu Oct 22 09:41:40 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux

$dpkg -p ros-indigo-hironx-ros-bridge ros-indigo-nextage-ros-bridge ros-indigo-hrpsys ros-indigo-moveit-ros ros-indigo-rviz |grep Ver
Version: 1.1.7-0trusty-20160208-134735-0800
Version: 0.7.5-0trusty-20160208-142556-0800
Version: 315.8.0-0trusty-20160201-040843-0800
Version: 0.7.0-0trusty-20160201-120956-0800
Version: 1.11.10-0trusty-20160201-033846-0800

130s, 2016-05-27:
I re-installed Ubuntu on the same machine I used for [my initial report](https://github.com/tork-a/rtmros_nextage/issues/235#issue-140227290) and am still seeing this issue.

```
$ dpkg -p ros-indigo-hironx-ros-bridge ros-indigo-nextage-ros-bridge ros-indigo-hrpsys ros-indigo-moveit-ros ros-indigo-rviz |grep Ver
Version: 1.1.14-0trusty-20160520-194441-0700
Version: 0.7.7-0trusty-20160520-204522-0700
Version: 315.9.0-0trusty-20160422-141424-0700
Version: 0.7.1-0trusty-20160420-101125-0700
Version: 1.11.14-0trusty-20160420-065247-0700
```

k-okada, 2016-05-28:
how about output of glxinfo

k-okada, 2016-09-03:
I had also run into same problem and found following commit does something wrong, this also have trouble when you add "Robot Model" in rviz, it said it can not solve 'world' to 'WAIST' that make sense.
```
--- a/nextage_description/urdf/NextageOpen.urdf
+++ b/nextage_description/urdf/NextageOpen.urdf
@@ -35,11 +35,11 @@
   </xacro:macro>
 
   <!-- Used for fixing robot to Gazebo 'base_link' -->
-  <link name="world"/>
-  <joint name="fixed" type="fixed">
-    <parent link="world"/>
-    <child link="WAIST"/>
-  </joint>
+  <!-- <link name="world"/> -->
+  <!-- <joint name="fixed" type="fixed"> -->
+  <!--   <parent link="world"/> -->
+  <!--   <child link
```
https://github.com/tork-a/rtmros_nextage/commit/afc801b4be578bbcf7d664a991248707ff000164

k-okada, 2016-09-04:
should be fixed via https://github.com/tork-a/rtmros_nextage/pull/255/commits/ec511564771bb7068d75f2859f58f1f18b85fae6

closed 2016-09-04 by k-okada

pazeshun, 2020-04-26:
@k-okada 

> this also have trouble when you add "Robot Model" in rviz, it said it can not solve 'world' to 'WAIST' that make sense.

Sorry, but I couldn't understand this, because TF is published from robot_state_publisher which reads `/robot_description` including a fixed joint from 'world' to 'WAIST'.
Could you describe your problem?

Also, I couldn't fully understand your commit message of https://github.com/tork-a/rtmros_nextage/pull/255/commits/ec511564771bb7068d75f2859f58f1f18b85fae6.
Did you want to say static_transform_publisher should be used instead of writing to robot model for environment (gazebo) specific transformation, because that transformation is not in real robot?
But I think [gazebo specific robot model including camera information](https://github.com/tork-a/rtmros_nextage/blob/0.8.5/nextage_description/urdf/nxo_gz.xacro) already exists, so moving that transformation to that robot model seems enough.

As a trial, I created the following diff and tried `roslaunch nextage_gazebo nextage_world.launch` and `roslaunch nextage_moveit_config moveit_planning_execution.launch`.
```
$ git diff
diff --git a/nextage_description/urdf/NextageOpen.urdf b/nextage_description/urdf/NextageOpen.urdf
index b4e08b5..6e95bdc 100644
--- a/nextage_description/urdf/NextageOpen.urdf
+++ b/nextage_description/urdf/NextageOpen.urdf
@@ -34,6 +34,13 @@
     </transmission>
   </xacro:macro>
 
+  <!-- Used for fixing robot to Gazebo 'base_link' -->
+  <link name="world"/>
+  <joint name="fixed" type="fixed">
+    <parent link="world"/>
+    <child link="WAIST"/>
+  </joint>
+
   <xacro:gazebo_nx_link name="WAIST" color="Black" />
   <link name="WAIST">
     <visual>
```
I could perform moveit planning & execution, so I felt all pipeline works well.
I got the following warning in `roslaunch nextage_moveit_config moveit_planning_execution.launch`. Is your problem about this?
```
[ WARN] [1587885856.782292685, 15.323000000]: Skipping virtual joint 'virtual_joint' because its child frame 'WAIST' does not match the URDF frame 'world'
```

k-okada, 2020-04-26:
> Also, I couldn't fully understand your commit message of ec51156.

> As a trial, I created the following diff and tried roslaunch nextage_gazebo nextage_world.launch and roslaunch nextage_moveit_config moveit_planning_execution.launch.
> I could perform moveit planning & execution, so I felt all pipeline works well.

what was the problem without the diff. I suspect it does not whot
interactive maker. Is is correct?

if so, I remember this problem was related to the move/rviz problem
that requires "world" frame if you want to visualize interactive
marker to control moveit goal pose. (But I think this is "bug" and
should be fixed. I remember I was fixed and send PR but could not
found that..)

And since the real robot does not have "world" frame, I think we
should use static_transform_publisher to set world->WAIST.
 not simply it does not have "world" in real robot, but also
considering when we put our robot in 73B2 scene, which we already have
world frame.
--
◉ Kei Okada

pazeshun, 2020-04-26:
> what was the problem without the diff. I suspect it does not whot interactive maker. Is is correct?

I'm for your opinion -> https://github.com/tork-a/rtmros_nextage/pull/233#issuecomment-195438084

> if so, I remember this problem was related to the move/rviz problem that requires "world" frame if you want to visualize interactive marker to control moveit goal pose. (But I think this is "bug" and should be fixed. I remember I was fixed and send PR but could not found that..)

Perhaps this is fixed on kinetic.
I'm testing on kinetic, so I might face no problem when I added the diff.

> considering when we put our robot in 73B2 scene, which we already have world frame.

I understand your opinion and that makes sense.
Thank you very much.