tork-a/rtmros_nextage #236: nextage.py depends on python-termcolor

longjie, 2016-03-16:
I've got an error to run nextage.py wit ipython:
```
[hrpsys.py] initialized successfully
---------------------------------------------------------------------------
ImportError                               Traceback (most recent call last)
/usr/lib/python2.7/dist-packages/IPython/utils/py3compat.pyc in execfile(fname, 
*where)
    202             else:
    203                 filename = fname
--> 204             __builtin__.execfile(filename, *where)

/opt/ros/indigo/share/nextage_ros_bridge/script/nextage.py in <module>()
     75 
     76     # ROS Client.
---> 77     ros = ROS_Client()
     78 
     79 # for simulated robot

/opt/ros/indigo/lib/python2.7/dist-packages/hironx_ros_bridge/ros_client.pyc in 
__init__(self, jointgroups)
     76             rospy.get_master().getSystemState()
     77         except Exception:
---> 78             from termcolor import colored
     79             print(colored('[ERROR][ros_client] ros master is not running
, so do not create ros client...', 'red'))
     80             return

ImportError: No module named termcolor
```
and it is solved by installing python-termcolor.
The dependency of the package may be added.

k-okada, 2016-03-16:
thanks, this has been solved by https://github.com/start-jsk/rtmros_hironx/pull/442 and waiting for release.

closed 2016-03-16 by k-okada

k-okada, 2016-03-16:
FYI: you can check the PR before they actually release; 
```
wget https://raw.githubusercontent.com/tork-a/88_tools/master/hotfixer/hot_fix.sh -O hot_fix.sh
bash  ./hot_fix.sh start-jsk/rtmros_hironx 442
```
and you can revert this change by
```
sudo apt-get install --reinstall ros-indigo-hironx-ros-bridge
```