tork-a/rtmros_nextage #240: URDF/wrl model of nextage with CKD FH116-D hands

JKBehrens, 2016-03-23:
Hello,

I am working on a nextage robot with different grippers supplied by KaWaDa. Here you can see a 3D-Scan of the gripper.
[image]

I just wanted to know if anyone has already adjusted the URDF and wrl model for using those grippers. 

I would appreciate if anyone can provide me good model files. Thank you in advance!

Best regards,
Jan

130s, 2016-03-23:
I'm sure [this tutorial](http://wiki.ros.org/rtmros_nextage/Tutorials/Changing%20Nextage%20Grippers%20in%20Robot%20Model) helps. Let me know on this ticket if you still need more info.

One thing that tutorial may not be clear about; if you write your code mainly in ROS, editing URDF might be enough. But if you want to utilize feature from `hrpsys` (controller running in lower-level on NEXTAGE Open), you may want to update the VRML (.wrl) file on QNX too. I'll have to clarify this in the referenced tutorial.

130s, 2016-04-15:
@JKBehrens I'm closing due to inactivity but feel free to reopen with more info.

closed 2016-04-15 by 130s