tork-a/rtmros_nextage #243: Topic /head_controller/state is not functional on robot and difference to simulation in gazebo

JKBehrens, 2016-04-29:
Hello,

I was developing a functionality for moving the head using the topics 

`/head_controller/state`

and 

`/head_controller/command`

which was working just fine with the simulation in gazebo. When I tried to transfer to the robot I encountered some problems. First, the topic is of type 

`pr2_controllers_msgs/JointTrajectoryControllerState`

instead of

`control_msgs/JointTrajectoryControllerState`

which it was in the simulation. So there is a difference (which could be avoided because the topics are equivalent). And secondly, the topic is advertised, but nothing is being published on it.

Can someone tell me:
What is the preferred way to control the head (or other joints)? 
Should the topics be silent or is something not working in my setup? 
   - I'm running the rosbridges for nextage and hironx from source and also pulled the rtmros_common from github. 

Thank you!
Jan

130s, 2016-05-03:
How about:

```
/head_controller/follow_joint_trajectory_action/goal
/head_controller/joint_trajectory_action/goal
```

You can also refer to an existing pull request https://github.com/start-jsk/rtmros_hironx/pull/427 for Hironx robot (same hardware with NEXTAGE Open. It's just naming with minor software difference).

JKBehrens, 2016-05-03:
Yes, I will try to use these interfaces next. 

Actually, I adjusted the PR2-Code for pointing the head at a pose which is mentioned there for the nextage robot:
https://github.com/start-jsk/rtmros_hironx/pull/427

I had it working nicely in gazebo simulation. When I finished and cleaned up my code, I will contribute to the mentioned pull request.

I opened this ticket, because the simulation is behaving differently than the real robot. If this is something you don't want to change, you can close this issue.

k-okada, 2016-09-05:
Sorry for long silence, i have updated pointing the head code, https://github.com/start-jsk/rtmros_hironx/pull/468
you need to update rtmros bridge system with https://github.com/start-jsk/rtmros_common/pull/966 patch