tork-a/rtmros_nextage #244: URDF of Nextage Robot wrong

JKBehrens, 2016-04-29:
Hello,

a few weeks ago, I was figuring out, that RARM_JOINT5_Link (probably also in the left arm) seem to be 7 mm to short in the URDF model. I was navigating the robot to a pose via rviz and moveit to save the pose using a query to tf. When I then asked the robot to move to the saved pose, it was not going completely there. A few mm left. As a workaround, I used different transformations in my code to read and command poses to the end effector.

Can someone please check the URDF model and confirm and maybe correct the robot description? Maybe this issue originates from an inconsistency between the different models used in the different middlewares???

Sorry for being so unspecific. I just wanted to leave a note and hope its obvious to someone. I can give more details, when I get back to the robot.

Thanks!
Jan

130s, 2016-05-03:
Ok, I'll check the hardware this week. With rulers/tape measure.
I'd be great too if you could measure the length of your robot.

As a clarification, what component you're exactly referring to? The white shaft right above the second black "box" in the image below?
[image]

130s, 2016-05-03:
Also, can you provide a code and steps that we can observe the issue you mentioned?

130s, 2016-05-03:
FYI @emijah

JKBehrens, 2016-05-03:
Thank you for your fast response! I am working this week at a different location. I can give you exact information on next Monday. 
For now: I moved the gripper to a pose using the interactive markers in RVIZ and read the Position of the EEF via
`rosrun tf tf_echo /WAIST /RARM_JOINT5_Link`
Then I asked moveit to move the arm to the saved pose, which resulted in a slightly different pose.

Might it be that the robot state publisher uses [the HIRONX URDF](https://github.com/start-jsk/rtmros_hironx/blob/indigo-devel/hironx_moveit_config/models/HiroNX.urdf) while moveit calculates the joint angles with [this model](https://github.com/tork-a/rtmros_nextage/blob/indigo-devel/nextage_description/urdf/NextageOpen.urdf)? I saw that in HIRONX urdf, RHAND_JOINT0_Link is defined. This is for example missing in NEXTAGE urdf.

Does this already help? I check on Monday with the robot.

130s, 2016-05-06:
Ok, I'll write a sample code for it.
Meanwhile, can you provide the tf value AND the actual pose value you use? @JKBehrens 

Also, if necessary, @emijah can help measuring the actual robot.

emijah, 2016-05-06:
I've checked the Nextage Open model against the HiroNX collada file and found that the translation for RARM_JOINT5 and LARM_JOINT5 differs between the two files. The Nextage Open files both vrml and collada define the translation for the final joint as (-0.047, 0.0, -0.090). The HiroNX collada file has the translation down as, (-0.04, 0, -0.09). So there is a difference of 7[mm] between the two. Is this what you are seeing?

hectorherrero, 2016-05-06:
Hi Jan, 

At Tecnalia (spanish research center) we have a Nextage robot for a few years, we have worked in several projects not only in terms of research conditions but also for industrial environments. 

We can reproduce this same issue with also 7mm of difference between the collada file of QNX controller and the URDF model that MoveIt is using for path planning and execution. But no only this, we have created new `TFs` directly publishing the value of the controller (`nextage.getCurrentPosition('RARM_JOINT5','WAIST')`) and with this input we have performed both TCP calibration and hand-eye camera calibration obtaining not accurate results, so we strongly believe that the collada file inside of the QNX controller is not accurate enough for working in a normal way. The workaround that we are following is fixing orientations of the TCP in order to alleviate these issues with a set of offsets. 

At previous discussions and tickets we concluded that a geometric calibration of the robot could improve these issues, we have not yet begun this process but it is an alternative that does not discarded in short-term.

Best regards,

Héctor

emijah, 2016-05-09:
It could be a case of one model being used for one thing, and another model being used for another...

emijah, 2016-05-09:
I re-checked the CAD model today. The correct value is 47mm

JKBehrens, 2016-05-10:
@emijah Yes, that is what I experience.

I think, it is really inconvenient to have inconsistent models, because this blocks for example teaching poses or as hector says. 

I post the output of the rosbridge terminal here just to be sure.

> nxouser@nxconsole:~$ roslaunch nextage_ros_bridge nextage_ros_bridge_real.launch MODEL_FILE:=/opt/jsk/etc/HIRONX/model/main.wrl
> ... logging to /home/nxouser/.ros/log/06426eec-16ae-11e6-8d85-000bab77d794/roslaunch-nxconsole-27523.log
> Checking log directory for disk usage. This may take awhile.
> Press Ctrl-C to interrupt
> Done checking log file disk usage. Usage is <1GB.
> 
> started roslaunch server http://nxconsole:37976/
> 
> SUMMARY
> ========
> 
> PARAMETERS
>  * /HrpsysSeqStateROSBridge/publish_sensor_transforms: True
>  * /collision_state/comp_name: CollisionDetector0
>  * /controller_configuration: [{'group_name': '...
>  * /diagnostic_aggregator/analyzers/computers/contains: ['HD Temp', 'CPU ...
>  * /diagnostic_aggregator/analyzers/computers/path: Computers
>  * /diagnostic_aggregator/analyzers/computers/type: diagnostic_aggreg...
>  * /diagnostic_aggregator/analyzers/hrpsys/contains: ['hrpEC', 'Emerge...
>  * /diagnostic_aggregator/analyzers/hrpsys/path: Hrpsys
>  * /diagnostic_aggregator/analyzers/hrpsys/type: diagnostic_aggreg...
>  * /diagnostic_aggregator/analyzers/mode/contains: ['Operating Mode'...
>  * /diagnostic_aggregator/analyzers/mode/path: Mode
>  * /diagnostic_aggregator/analyzers/mode/type: diagnostic_aggreg...
>  * /diagnostic_aggregator/analyzers/motor/contains: ['Motor']
>  * /diagnostic_aggregator/analyzers/motor/path: Motor
>  * /diagnostic_aggregator/analyzers/motor/type: diagnostic_aggreg...
>  * /diagnostic_aggregator/base_path: 
>  * /diagnostic_aggregator/pub_rate: 1.0
>  * /robot_description: <?xml version="1....
>  * /rosdistro: indigo
>  * /rosversion: 1.11.19
>  * /rtmnameserver_port: 15005
>  * /rtmnameserver_robotname: RobotHardware0
> 
> NODES
>   /
>     CollisionDetectorComp (hrpsys/CollisionDetectorComp)
>     DataLoggerServiceROSBridge (hrpsys_ros_bridge/DataLoggerServiceROSBridgeComp)
>     ForwardKinematicsServiceROSBridge (hrpsys_ros_bridge/ForwardKinematicsServiceROSBridgeComp)
>     HrpsysJointTrajectoryBridge (hrpsys_ros_bridge/HrpsysJointTrajectoryBridge)
>     HrpsysSeqStateROSBridge (hrpsys_ros_bridge/HrpsysSeqStateROSBridge)
>     RobotHardwareServiceROSBridge (hrpsys_ros_bridge/RobotHardwareServiceROSBridgeComp)
>     SequencePlayerServiceROSBridge (hrpsys_ros_bridge/SequencePlayerServiceROSBridgeComp)
>     StateHolderServiceROSBridge (hrpsys_ros_bridge/StateHolderServiceROSBridgeComp)
>     collision_state (hrpsys_ros_bridge/collision_state.py)
>     diagnostic_aggregator (diagnostic_aggregator/aggregator_node)
>     hrpsys_profile (hrpsys_ros_bridge/hrpsys_profile.py)
>     hrpsys_py (hironx_ros_bridge/hironx.py)
>     hrpsys_ros_diagnostics (hrpsys_ros_bridge/diagnostics.py)
>     hrpsys_state_publisher (robot_state_publisher/state_publisher)
>     rqt_hironxo (rqt_gui/rqt_gui)
>     rtmlaunch_collision_detector (openrtm_tools/rtmlaunch.py)
>     rtmlaunch_hrpsys_ros_bridge (openrtm_tools/rtmlaunch.py)
>     sensor_ros_bridge_connect (hrpsys_ros_bridge/sensor_ros_bridge_connect.py)
> 
> WARNING: unrecognized tag rtconnect
> WARNING: unrecognized tag rtconnect
> WARNING: unrecognized tag rtconnect
> WARNING: unrecognized tag rtconnect
> WARNING: unrecognized tag rtconnect
> WARNING: unrecognized tag rtconnect
> WARNING: unrecognized tag rtconnect
> WARNING: unrecognized tag rtconnect
[... 280 lines cut ...]
> [INFO] [WallTime: 1462884753.748378] check 4 collision status, 0.796818 Hz

I am especially wondering about these Errors:

> [rqt_hironxo-18] process has died [pid 27777, exit code 1, cmd /opt/ros/indigo/lib/rqt_gui/rqt_gui --perspective-file /home/nxouser/rtmros_nextage_ws/src/rtmros_hironx/hironx_ros_bridge/resource/hironx_rqt_dashboard.perspective __name:=rqt_hironxo __log:=/home/nxouser/.ros/log/06426eec-16ae-11e6-8d85-000bab77d794/rqt_hironxo-18.log].
> log file: /home/nxouser/.ros/log/06426eec-16ae-11e6-8d85-000bab77d794/rqt_hironxo-18*.log

and

> [hrpsys.py] setup joint groups
> [hrpsys.py] initialized successfully
> [ERROR] [1462884747.118157439]: Robot semantic description not found. Did you forget to define or remap '/robot_description_semantic'?
> [ INFO] [1462884747.118223487]: Loading robot model 'HiroNX'...
> [ INFO] [1462884747.118246886]: No root joint specified. Assuming fixed joint
> [ INFO] [1462884747.572375078]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 200[Hz] (exec 1782 Hz, dropped 200)
> [INFO] [WallTime: 1462884747.638430] Joint names; Torso: ['CHEST_JOINT0']
> 	Head: ['HEAD_JOINT0', 'HEAD_JOINT1']
> 	L-Arm: ['LARM_JOINT0', 'LARM_JOINT1', 'LARM_JOINT2', 'LARM_JOINT3', 'LARM_JOINT4', 'LARM_JOINT5']
> 	R-Arm: ['RARM_JOINT0', 'RARM_JOINT1', 'RARM_JOINT2', 'RARM_JOINT3', 'RARM_JOINT4', 'RARM_JOINT5']
> [WARN] [WallTime: 1462884747.639464] Moveit is not started yet, if you want to use MoveIt!
> Make sure you've launched MoveGroup (e.g. by launching moveit_planning_execution.launch)
> [hrpsys_py-1] process has died [pid 27541, exit code -11, cmd /home/nxouser/rtmros_nextage_ws/src/rtmros_hironx/hironx_ros_bridge/scripts/hironx.py --host nextage --port 15005 --modelfile /opt/jsk/etc/HIRONX/model/main.wrl --robot RobotHardware __name:=hrpsys_py __log:=/home/nxouser/.ros/log/06426eec-16ae-11e6-8d85-000bab77d794/hrpsys_py-1.log].
> log file: /home/nxouser/.ros/log/06426eec-16ae-11e6-8d85-000bab77d794/hrpsys_py-1*.log

As side note: [here](https://github.com/tork-a/rtmros_nextage/issues/243), I told that I work on adjusting the PR2 point head action for the Nextage robot. As the rosbridge loads the HIRONX URDF into the /robot_description, I get errors that the root link is not fixed. This was not happening when I used the gazebo simulation. I suppose, this is also an issue with the model of HIRONX.

130s, 2016-05-11:
@JKBehrens To address this issue I made pull requests that I can't test without real robot (I do not have access to it immediately). If you could test that'll be great, but if you're not sure then don't hesitate to let me know.

To apply the changes, you need build `rtmros_nextage` from source like this.
```
mkdir -p ~/cws_nxo_244/src && cd ~/cws_nxo_244/src && catkin_init_workspace
git clone https://github.com/130s/rtmros_hironx.git && cd rtmros_hironx && git checkout impr/collada_realrobot
cd ~/cws_nxo_244/src
git clone https://github.com/130s/rtmros_nextage.git && cd rtmros_nextage && git checkout fix/real/wrong_wrl
cd ~/cws_nxo_244 && catkin_make && source devel/setup.bash
```

JKBehrens, 2016-05-11:
@130s I built it from source and started the rosbridge, but it is not coming up. CollisionDetector is not found and startup of moveit fails. Is it still correct to start the rosbridge by

`roslaunch nextage_ros_bridge nextage_ros_bridge_real.launch MODEL_FILE:=/opt/jsk/etc/HIRONX/model/main.wrl`

??

Rosbridge output:
```
nxouser@nxconsole:~/cws_nxo_244$ roslaunch nextage_ros_bridge nextage_ros_bridge_real.launch MODEL_FILE:=/opt/jsk/etc/HIRONX/model/main.wrl
... logging to /home/nxouser/.ros/log/58b4004a-1759-11e6-a8dd-000bab77d794/roslaunch-nxconsole-9295.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://nxconsole:33069/

SUMMARY
========

PARAMETERS
 * /HrpsysSeqStateROSBridge/publish_sensor_transforms: True
 * /collision_state/comp_name: CollisionDetector0
 * /controller_configuration: [{'group_name': '...
 * /diagnostic_aggregator/analyzers/computers/contains: ['HD Temp', 'CPU ...
 * /diagnostic_aggregator/analyzers/computers/path: Computers
 * /diagnostic_aggregator/analyzers/computers/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/hrpsys/contains: ['hrpEC', 'Emerge...
 * /diagnostic_aggregator/analyzers/hrpsys/path: Hrpsys
 * /diagnostic_aggregator/analyzers/hrpsys/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/mode/contains: ['Operating Mode'...
 * /diagnostic_aggregator/analyzers/mode/path: Mode
 * /diagnostic_aggregator/analyzers/mode/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/motor/contains: ['Motor']
 * /diagnostic_aggregator/analyzers/motor/path: Motor
 * /diagnostic_aggregator/analyzers/motor/type: diagnostic_aggreg...
 * /diagnostic_aggregator/base_path: 
 * /diagnostic_aggregator/pub_rate: 1.0
 * /robot_description: <?xml version="1....
 * /rosdistro: indigo
 * /rosversion: 1.11.19
 * /rtmnameserver_port: 15005
 * /rtmnameserver_robotname: RobotHardware0

NODES
  /
    CollisionDetectorComp (hrpsys/CollisionDetectorComp)
    DataLoggerServiceROSBridge (hrpsys_ros_bridge/DataLoggerServiceROSBridgeComp)
    ForwardKinematicsServiceROSBridge (hrpsys_ros_bridge/ForwardKinematicsServiceROSBridgeComp)
    HrpsysJointTrajectoryBridge (hrpsys_ros_bridge/HrpsysJointTrajectoryBridge)
    HrpsysSeqStateROSBridge (hrpsys_ros_bridge/HrpsysSeqStateROSBridge)
    RobotHardwareServiceROSBridge (hrpsys_ros_bridge/RobotHardwareServiceROSBridgeComp)
    SequencePlayerServiceROSBridge (hrpsys_ros_bridge/SequencePlayerServiceROSBridgeComp)
    StateHolderServiceROSBridge (hrpsys_ros_bridge/StateHolderServiceROSBridgeComp)
    collision_state (hrpsys_ros_bridge/collision_state.py)
    diagnostic_aggregator (diagnostic_aggregator/aggregator_node)
    hrpsys_profile (hrpsys_ros_bridge/hrpsys_profile.py)
    hrpsys_py (hironx_ros_bridge/hironx.py)
    hrpsys_ros_diagnostics (hrpsys_ros_bridge/diagnostics.py)
    hrpsys_state_publisher (robot_state_publisher/state_publisher)
    rqt_hironxo (rqt_gui/rqt_gui)
    rtmlaunch_collision_detector (openrtm_tools/rtmlaunch.py)
    rtmlaunch_hrpsys_ros_bridge (openrtm_tools/rtmlaunch.py)
    sensor_ros_bridge_connect (hrpsys_ros_bridge/sensor_ros_bridge_connect.py)

WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
[... 561 lines cut ...]
Loading 'move_group/MoveGroupQueryPlannersService'...
Loading 'move_group/MoveGroupStateValidationService'...
[ INFO] [1462958737.664808821]: 

********************************************************
* MoveGroup using: 
*     - CartesianPathService
*     - ExecutePathService
*     - GetPlanningSceneService
*     - KinematicsService
*     - MoveAction
*     - PickPlaceAction
*     - MotionPlanService
*     - QueryPlannersService
*     - StateValidationService
********************************************************

[ INFO] [1462958737.664890150]: MoveGroup context using planning plugin ompl_interface/OMPLPlanner
[ INFO] [1462958737.664909417]: MoveGroup context initialization complete

All is well! Everyone is happy! You can start planning now!

[rviz_nxconsole_12858_6887182820923041230-2] process has died [pid 12877, exit code -11, cmd /opt/ros/indigo/lib/rviz/rviz --fixed-frame WAIST -d /home/nxouser/cws_nxo_244/src/rtmros_hironx/hironx_moveit_config/launch/moveit_viewnatto.rviz __name:=rviz_nxconsole_12858_6887182820923041230 __log:=/home/nxouser/.ros/log/1d54b4d0-175a-11e6-954d-000bab77d794/rviz_nxconsole_12858_6887182820923041230-2.log].
log file: /home/nxouser/.ros/log/1d54b4d0-175a-11e6-954d-000bab77d794/rviz_nxconsole_12858_6887182820923041230-2*.log
```

emijah, 2016-05-12:
I issued a pull request for the HiroNX collada model. However, I need someone to generate the other models from the collada file.
The pull request is, start-jsk/rtmros_hironx#461

130s, 2016-05-13:
@JKBehrens My bad, with the change I propose, correct command for your robot will be:

```
roslaunch nextage_ros_bridge nextage_ros_bridge_real.launch MODEL_FILE:=/opt/jsk/etc/HIRONX/model/main.wrl CONF_FILE_COLLISIONDETECT:=`rospack find hironx_ros_bridge`/conf/hironx_realrobot_fixedpath.conf
```

(I know the command is getting longer...we will need to come up with a solution for that but for now please bear with it ;)

----

NOTE for future reader: the aforementioned command is only for the NEXTAGE Open robot that does not contain `/opt/jsk/etc/NEXTAGE` folder internal to the ControlBox/QNX. If your robot does contain the folder then you don't need `CONF_FILE_COLLISIONDETECT` arg.

JKBehrens, 2016-05-13:
@130s Now, the Rosbridge is coming up. Furthermore, from
`rosrun tf tf_echo /RARM_JOINT4_Link /RARM_JOINT5_Link`
I get
```
- Translation: [-0.047, 0.000, -0.090]
- Rotation: in Quaternion [-0.000, 0.000, 0.000, 1.000]
            in RPY (radian) [-0.000, 0.000, 0.000]
            in RPY (degree) [-0.008, 0.000, 0.000]
```
, which is now consistent with the URDF file. Thank you all!!!

Can I now rely on the equivalence of the models or did you just fix the pointed differences? Would it be useful to update the software and files on the QNX?

130s, 2016-05-17:
Glad that the changes worked for you @JKBehrens !

> Can I now rely on the equivalence of the models or did you just fix the pointed differences? Would it be useful to update the software and files on the QNX? 

So for `NEXTAGE Open` robot (which you are using), there's no need to worry for incorrect model files as far as we know -- the root cause of this ticketed issue is that the software of NEXTAGE Open was referring to the model of Hironx robot (former version of NEXTAGE) where there seems to be a slight geometric difference (patch https://github.com/start-jsk/rtmros_hironx/pull/461).

I don't think there's a need for you (and all NEXTAGE Open users) need to update the files on QNX, in terms of this issue.

If there's still anythiing unclear etc. feel free to reopen.

130s, 2016-05-17:
BTW, the changes made for this ticket will soon be available via DEB binary with the following versions (or greater):

- rtmros_hironx 1.1.13
- rtmros_nextage 0.7.7

closed 2016-05-17 by 130s