tork-a/rtmros_nextage #248: CORBA.COMM_FAILURE - Communication with the robot is lost

hectorherrero, 2016-06-22:
Hi, 

we are suffering this CORBA.COMM_FAILURE frecuently, and we have realized that when we are working intensively with the robot testing and debugging applications (which implies frecuent close and relaunch ros applications nodes) this problems always arise. 

The usual scenario is as follows:
Terminal 1: `roslaunch nextage_ros_bridge nextage_ros_bridge_real.launch nameserver:=hiro019`
Terminal 2: `roslaunch nextage_moveit_config moveit_planning_execution.launch`
Terminal 3: ROS application -> Some nodes create a NextageClient and fails at `init(robotname='RobotHardware0', url='/opt/jsk/etc/HIRONX/model/main.wrl')`

Terminal 3 (application) output:
```
File "/home/oscuser/ros/hydro/src/flexbotics/flexbotics_external_libraries/nextage_tork_git/nextage_ros_bridge/src/nextage_ros_bridge/nextage_client.py", line 89, in init
    HIRONX.init(self, robotname=robotname, url=url)
  File "/opt/ros/hydro/lib/python2.7/dist-packages/hironx_ros_bridge/hironx_client.py", line 124, in init
    HrpsysConfigurator.init(self, robotname=robotname, url=url)
  File "/opt/ros/hydro/lib/python2.7/dist-packages/hrpsys/hrpsys_config.py", line 1679, in init
    self.activateComps()
  File "/opt/ros/hydro/lib/python2.7/dist-packages/hrpsys/hrpsys_config.py", line 426, in activateComps
    rtm.serializeComponents(rtcList)
  File "/opt/ros/hydro/lib/python2.7/dist-packages/hrpsys/rtm.py", line 424, in serializeComponents
    if not ec._is_equivalent(rtc.ec):
AttributeError: 'NoneType' object has no attribute 'ec'
```
Terminal 1 (nextage_ros_bridge) output:
```
[rtmlaunch] Wait for  /hiro019:15005/co.rtc   10 /30
[ INFO] [1466577072.715245695]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 201[Hz]
[rtmlaunch] Wait for  /hiro019:15005/co.rtc   11 /30
[ INFO] [1466577073.721279144]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 201[Hz]
[ INFO] [1466577074.727507309]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 201[Hz]
[rtmlaunch] [ERROR] Could not Connect ( /hiro019:15005/sh.rtc:qOut , /hiro019:15005/HrpsysSeqStateROSBridge0.rtc:mcangle ):  CORBA.COMM_FAILURE(omniORB.COMM_FAILURE_WaitingForReply, CORBA.COMPLETED_MAYBE) 
[rtmlaunch] Could not Activate ( hiro019:15005/$(arg COLLISIONDETECTOR_INSTANCE_NAME).rtc ) :  CORBA.COMM_FAILURE(omniORB.COMM_FAILURE_WaitingForReply, CORBA.COMPLETED_MAYBE) 
[ WARN] [1466577079.727932223]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is not executed last 5[sec]
[ WARN] [1466577084.728469162]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is not executed last 5[sec]
```
The consecuence of this error is the lost of the communication with the robot. The only thing that we can do is restarting the robot (which implies several minutes for starting all again, checkEncoders...) . We have reproduced the error in Nextage and Hironx robots, and also happens in simulation (relaunching nextage_ros_bridge_simulation is enough in this case). 

Let me know if more information is required to delimit the possible causes. 

Regards, 
Héctor.

130s, 2016-06-22:
@hectorherrero while some error messages seem familliar, I do appreciate more info indeed.

> when we are working intensively with the robot testing and debugging applications (which implies frecuent close and relaunch ros applications nodes) this problems always arise.

It'd be helpful if you could count how many times you run and teminate nodes with how long of intervals, so that I can write test codes (that could hopefull reproduce your issue).

> Terminal 3 (application) output:
> ```
> File "/home/oscuser/ros/hydro/src/flexbotics/flexbotics_external_libraries/nextage_tork_git/nextage_ros_bridge/src/nextage_ros_bridge/nextage_client.py", line 89, in init
> :
> AttributeError: 'NoneType' object has no attribute 'ec'
> ```
>
> Terminal 1 (nextage_ros_bridge) output:
> ```
> :
> [rtmlaunch] [ERROR] Could not Connect ( /hiro019:15005/sh.rtc:qOut , /hiro019:15005/HrpsysSeqStateROSBridge0.rtc:mcangle ):  CORBA.COMM_FAILURE(omniORB.COMM_FAILURE_WaitingForReply, CORBA.COMPLETED_MAYBE) 
> [rtmlaunch] Could not Activate ( hiro019:15005/$(arg COLLISIONDETECTOR_INSTANCE_NAME).rtc ) :  CORBA.COMM_FAILURE(omniORB.COMM_FAILURE_WaitingForReply, CORBA.COMPLETED_MAYBE) 
> ```

Just a clarification; do you always see the same error message like above whenever the issue you metioned happens?

And, to share the whole logs and environment infos, we've made it a little bit easier. Please run the following, and send back the tarball (github now allows attachments).

    curl -sL https://raw.githubusercontent.com/start-jsk/rtmros_common/master/hrpsys_tools/scripts/rtmroswtf.sh | bash -

  (This fetches `~/.ros/log/latest` folder so you need to run this right after you see the issue.)

Thanks.

hectorherrero, 2016-06-27:
> It'd be helpful if you could count how many times you run and teminate nodes with how long of intervals, so that I can write test codes (that could hopefull reproduce your issue).

This is difficult to determinate, as the intervals change depending the errors or the results of the tests that we are testing. But the perception is that the faster the application is relaunched (small changes and adjustments) most probable is the error appears.
I was able to reproduce the error launching and killing by hand several (al least 6 terminal in pararell) `ipython nextage `clients and the launch file of one of our applications, but was not a real situation and the obtained tarball does not provide relevant information ([rtmros_diagnosisinfo_all_20160622-084307.tar.gz](https://github.com/tork-a/rtmros_nextage/files/334259/rtmros_diagnosisinfo_all_20160622-084307.tar.gz)). I have compressed the last `.ros/log` folder which provides more information ([d6a2f13e-3840-11e6-a912-00187d19c997.tar.gz](https://github.com/tork-a/rtmros_nextage/files/334255/d6a2f13e-3840-11e6-a912-00187d19c997.tar.gz)). 
Besides, also you can find attached the logs of a real situation when the error appears ([rtmros_diagnosisinfo_all_20160624-141650.tar.gz](https://github.com/tork-a/rtmros_nextage/files/334262/rtmros_diagnosisinfo_all_20160624-141650.tar.gz)). 

> Just a clarification; do you always see the same error message like above whenever the issue you metioned happens?

Yes, always appears at the same time in both terminals. 

As you can see our applications involves lot of nodes, I hope this does not mask the problem, or in any case, may be able to understand what is the problem of interacting.

Thanks

hectorherrero, 2016-07-08:
Since we continue suffering this issue, we have started to perform different stress tests, leaving the robot working for long time. But as commented above the issue appears when we are relaunching applications intensively. 
Do you know another test that we can perform in order to get more information? How can we obtain information or logfiles from the QNX system?

Thanks

k-okada, 2016-07-08:
HI, not sure if I understand your problem correctly, theoretically  you do not need to run `nextage_clinet.py` for your application, that command is basically designed to "wake up" the robots, which means it  loads and activates hrpsys plugins (components), and once they have started, you do not need run thease procedure again. So if you want to use ROS client,  just to run python and import rospy, if you need python interface for hrpsys system, try, `hrpsyspy --host hiro019 --port 15005`, this will not call ` File "/opt/ros/hydro/lib/python2.7/dist-packages/hrpsys/hrpsys_config.py", line 426, in activateComps` but still have access to the robot.

If these approach works for you, we'd change nextage_client.py, not to load and activate plugins, if they are already activated.

hectorherrero, 2016-07-08:
Hi, currently our applications are using nextage_client because we need an instance for controlling the robot via python. I have tried `hrpsyspy`  for one of our application and works (apparently) without problems. The only issue that I have found is that hrpsys does not have `goInitial()` method. 

So, if nextage_client.py could be modified for not loading and activating plugins will be great, because this does not require any change on our applications. For nextage_client modification, could you sent us a set of instructions to update it (if is not too complicated)? In order to test in the real robot as soon as possible and see if the issue disappears.

Thanks

534o, 2016-07-10:
OK, please make sure that you have `hironx_ros_bridge` package under `/opt/ros/...`
```
$  rospack find hironx_ros_bridge
/opt/ros/indigo/share/hironx_ros_bridge
```
and then run following command for apply patch file;
```
wget https://raw.githubusercontent.com/tork-a/88_tools/master/hotfixer/hot_fix.sh -O /tmp/hot_fix.sh
chmod u+x /tmp/hot_fix.sh 
/tmp/hot_fix.sh  start-jsk/rtmros_hironx 466
```

TORK support team

130s, 2016-07-11:
Now [the proposed fix](https://github.com/tork-a/rtmros_nextage/issues/248#issuecomment-231581126) is merged to the master (a.k.a `indigo-devel`) branch. If you're using `nextage_ros_bridge` package from source, you can simply clone the repo into your catkin workspace. Let us know if it works. Thanks.

hectorherrero, 2016-07-12:
For the moment I have tested on my PC and works without side effects (simulation). The Ubuntu PC of the robot has ROS Hydro, so I cannot test directly. I will connect my laptop and perform some tests as soon as possible. 

Thanks!

130s, 2016-10-18:
Closing with a lack of activity. Feel free to reopen if the issue persists.

TecnaliaRobotics, 2016-10-18:
The issue persists! Even in your latest patch  in Indigo.
Hector was going to contact you for a real and stable solution…

Enviado el: martes, 18 de octubre de 2016 11:14
Para: tork-a/rtmros_nextage

hectorherrero, 2016-10-18:
Sorry for the inactivity, last week we had been in a trade fair executing continuously another application. In this case we have updated the Ubuntu PC to 14.04 and Indigo with standard hironx/nextage packages. We have detected (probably not too often than with the Hydro release) CORBA connection issue. 

I think that the deb that we used did not have the proposed fix, so as starting point, how can I check if the fix is applied to standard deb packages?

In the following weeks we will be working in order to solve this problem, so I will provide more information when I have.

Thanks.

534o, 2016-10-18:
please let me know which version of hrpsys in PC side (rosversion hrpsys) and also try something like `hrpsyspy --host hiro019 --port 15005` to connect ros client, my first thought is that `nextage_clinet.py` is calling `self.activateComps()` which is not requried if that is already activated.

hectorherrero, 2016-10-21:
```
$ rosversion hrpsys
315.10.0
```
The output of `hrpsyspy --host hiro019 --port 15005`:
```
[hrpsys.py] wait for RTCmanager : hiro019
[hrpsys.py] wait for RobotHardware0 : <hrpsys.rtm.RTcomponent instance at 0x7f0cd7e35b48> ( timeout 0 < 10)
[hrpsys.py] findComps -> RobotHardware : <hrpsys.rtm.RTcomponent instance at 0x7f0cd7e35b48> isActive? = True 
[hrpsys.py] simulation_mode : False
Starting hrpsys python interface for RobotHardware0
 Installed rtc are ...  seq   sh   fk   el   log  
 Use 'hcf' instance to access to the hrpsys controller
Python 2.7.6 (default, Jun 22 2015, 17:58:13) 
Type "copyright", "credits" or "license" for more information.
```
If I run `ipython -i `rospack find nextage_ros_bridge`/script/nextage.py -- --host hiro019 --robot RobotHardware0 --port 15005 --modelfile /opt/jsk/etc/HIRONX/model/main.wrl`:
```
 Python 2.7.6 (default, Jun 22 2015, 17:58:13) 
Type "copyright", "credits" or "license" for more information.

IPython 1.2.1 -- An enhanced Interactive Python.
?         -> Introduction and overview of IPython's features.
%quickref -> Quick reference.
help      -> Python's own help system.
object?   -> Details about 'object', use 'object??' for extra details.
configuration ORB with hiro019:15005
[hrpsys.py] waiting ModelLoader
[hrpsys.py] start hrpsys
[hrpsys.py] finding RTCManager and RobotHardware
[hrpsys.py]  waitForRTCManagerAndRoboHardware has renamed to waitForRTCManagerAndRoboHardware: Please update your code
[hrpsys.py] wait for RTCmanager : None
[hrpsys.py] wait for RobotHardware0 : <hrpsys.rtm.RTcomponent instance at 0x7f9573d53518> ( timeout 0 < 10)
[hrpsys.py] findComps -> RobotHardware : <hrpsys.rtm.RTcomponent instance at 0x7f9573d53518> isActive? = True 
[hrpsys.py] simulation_mode : False
[hrpsys.py]  Hrpsys controller version info: 
[hrpsys.py]    ms =  <hrpsys.rtm.RTCmanager instance at 0x7f9573d53320>
[hrpsys.py]    ref =  <RTM._objref_Manager instance at 0x7f9573d532d8>
[hrpsys.py]    version  =  315.2.8
[hrpsys.py]   bodyinfo URL = file:///opt/jsk/etc/HIRONX/model/main.wrl
[hrpsys.py] hrpsys components are already created and running
[hrpsys.py] [self.seq, self.seq_svc, self.seq_version] = self.findComp("SequencePlayer","seq",0,True)
[hrpsys.py]  find Comp    : seq = <hrpsys.rtm.RTcomponent instance at 0x7f9573d53998> (315.2.8)
[hrpsys.py]  find CompSvc : seq_svc = <OpenHRP._objref_SequencePlayerService instance at 0x7f9574fa5cb0>
[hrpsys.py] [self.sh, self.sh_svc, self.sh_version] = self.findComp("StateHolder","sh",0,True)
[hrpsys.py]  find Comp    : sh = <hrpsys.rtm.RTcomponent instance at 0x7f9574f5bd88> (315.2.8)
[hrpsys.py]  find CompSvc : sh_svc = <OpenHRP._objref_StateHolderService instance at 0x7f9574ed0e60>
[hrpsys.py] [self.fk, self.fk_svc, self.fk_version] = self.findComp("ForwardKinematics","fk",0,True)
[hrpsys.py]  find Comp    : fk = <hrpsys.rtm.RTcomponent instance at 0x7f9574f72440> (315.2.8)
[hrpsys.py]  find CompSvc : fk_svc = <OpenHRP._objref_ForwardKinematicsService instance at 0x7f9574ed9440>
[hrpsys.py] [self.el, self.el_svc, self.el_version] = self.findComp("SoftErrorLimiter","el",0,True)
[hrpsys.py]  find Comp    : el = <hrpsys.rtm.RTcomponent instance at 0x7f9574bb5c20> (315.2.8)
[hrpsys.py]  find CompSvc : el_svc = <OpenHRP._objref_SoftErrorLimiterService instance at 0x7f9574ec0b48>
[hrpsys.py] [self.log, self.log_svc, self.log_version] = self.findComp("DataLogger","log",0,True)
[hrpsys.py]  find Comp    : log = <hrpsys.rtm.RTcomponent instance at 0x7f9574f42680> (315.2.8)
[hrpsys.py]  find CompSvc : log_svc = <OpenHRP._objref_DataLoggerService instance at 0x7f9574fb1c68>
[hrpsys.py] setup joint groups for hrpsys controller
[hrpsys.py] initialized successfully
[ INFO] [1477054029.297479135]: Loading robot model 'HiroNX'...
[ INFO] [1477054029.297527904]: No root joint specified. Assuming fixed joint
[INFO] [WallTime: 1477054029.924474] Joint names; Torso: ['CHEST_JOINT0']
	Head: ['HEAD_JOINT0', 'HEAD_JOINT1']
	L-Arm: ['LARM_JOINT0', 'LARM_JOINT1', 'LARM_JOINT2', 'LARM_JOINT3', 'LARM_JOINT4', 'LARM_JOINT5']
	R-Arm: ['RARM_JOINT0', 'RARM_JOINT1', 'RARM_JOINT2', 'RARM_JOINT3', 'RARM_JOINT4', 'RARM_JOINT5']
[ INFO] [1477054030.716386437]: Ready to take MoveGroup commands for group left_arm.
[ INFO] [1477054031.021930089]: Ready to take MoveGroup commands for group right_arm.
[ INFO] [1477054031.427055098]: Ready to take MoveGroup commands for group botharms.
[ INFO] [1477054031.822948260]: Ready to take MoveGroup commands for group head.
[ INFO] [1477054032.237062620]: Ready to take MoveGroup commands for group torso.
[ INFO] [1477054032.831710023]: Ready to take MoveGroup commands for group upperbody.
```

130s, 2016-10-24:
@hectorherrero In your https://github.com/tork-a/rtmros_nextage/issues/248#issuecomment-255370200, I don't see your issue any more. Can using Indigo version be a fix to you?

hectorherrero, 2016-10-24:
With my last comment I want to assure that the proposed fix is actually applied (from my understanding it is). But as I commented: 
> We have detected (probably not too often than with the Hydro release) CORBA connection issue.
We need to continue making tests and obtaining logs in order to be sure that the problem is fixed.

534o, 2016-10-25:
@hectorherrero
the difference between `hrpsyspy` and `nextage_clinet.py` is that later one calls `activateComp()` which is already called before, 
- Please run `hrpsyspy` many times and  see if the problem occurs, If not we can add `--client` mode for `nextage_client.py` that will fix this problem
-  if you run `hrpsyspy --host hiro019 --port 15005 `, then you can use the `hcf` object instance as you used `robot` or `nxo` in `nextage_clinet.py`. I'm not sure how you problem your application, but most of cases, you just add 'rospy.init_node()` and continue writing your program upon `hprsyspy`

534o, 2016-10-25:
@hectorherrero

please run `hrpsyspy --host hiro019 --port 15005` may times and that did
not raise problem ,
if that works you can use hcf instance as robot/nxo instance in nextage.py,
if you prefer to continue using nextage.py, we can add client mode for
nextage.py, which did not run `self.activateComps()`

hectorherrero, 2016-11-07:
I have prepared a simple set of scripts (find attached) for launching and killing nextage.py client and one of our applications (which have some instances of the client). The error is reproducible for both cases, the only different is the number of required executions. I will try to describe the test environment, which is similar than the real use case:

Terminal 1: hironx/nextage_ros_bridge. It is launched once and kept running throughout the test or work session. 
`roslaunch hironx_ros_bridge hironx_ros_bridge_real.launch`
Terminal 2: hironx/nextgage_moveit: Optional if required
`roslaunch hironx_moveit_config moveit_planning_execution.launch`
Terminal 3: TECNALIA application. Composed by several nodes, some of them create instances of hironx/nextage.py client. Also tested only executing hironx.py. This terminal is killed and relaunched again after some seconds continuously. 
`ipython -i `rospack find hironx_ros_bridge`/scripts/hironx.py -- --host hiro019 --robot RobotHardware0 --port 15005 --modelfile /opt/jsk/etc/HIRONX/model/main.wrl` 

After a different number of executions connection error are obtained both in Terminal 1 and Terminal 3:
Terminal 1:
```
Execution: 39
[rtmlaunch] check connection/activation
Traceback (most recent call last):
  File "/opt/ros/indigo/share/openrtm_tools/scripts/rtmlaunch.py", line 13, in <module>
    rtmlaunch.main()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/openrtm_tools/rtmlaunch.py", line 253, in main
    tree.add_name_server(nameserver, [])
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rtctree/tree.py", line 131, in add_name_server
    self._parse_name_server(server, filter, dynamic=dynamic)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rtctree/tree.py", line 285, in _parse_name_server
    trim_filter(deepcopy(filter), 2), dynamic=dynamic)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rtctree/nameserver.py", line 54, in __init__
    self._parse_server(address, orb, filter)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rtctree/nameserver.py", line 79, in _parse_server
    self._parse_context(root_context, orb, filter)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rtctree/directory.py", line 111, in _parse_context
    self._process_binding(binding, orb, filter)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rtctree/directory.py", line 190, in _process_binding
    filter=trimmed_filter)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rtctree/directory.py", line 111, in _parse_context
    self._process_binding(binding, orb, filter)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rtctree/directory.py", line 142, in _process_binding
    leaf = Manager(name, self, obj, dynamic=self.dynamic)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rtctree/manager.py", line 55, in __init__
    self._parse()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rtctree/manager.py", line 318, in _parse
    self._parse_children()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rtctree/manager.py", line 323, in _parse_children
    self._parse_component_children()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rtctree/manager.py", line 340, in _parse_component_children
    leaf = Component(instance_name + '.rtc', self, c)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rtctree/component.py", line 94, in __init__
    self._parse_profile()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rtctree/component.py", line 1088, in _parse_profile
    profile = self._obj.get_component_profile()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rtctree/rtmidl/RTC_idl.py", line 1344, in get_component_profile
    return _omnipy.invoke(self, "get_component_profile", _0_RTC.RTObject._d_get_component_profile, args)
omniORB.CORBA.COMM_FAILURE: CORBA.COMM_FAILURE(omniORB.COMM_FAILURE_WaitingForReply, CORBA.COMPLETED_MAYBE)
[31m[rtmlaunch_collision_detector-18] process has died [pid 24904, exit code 1, cmd /opt/ros/indigo/share/openrtm_tools/scripts/rtmlaunch.py /opt/ros/indigo/share/hrpsys_ros_bridge/launch/collision_detector.launch __name:=rtmlaunch_collision_detector __log:=/home/oscuser/.ros/log/d555b9b6-a1a8-11e6-a6e4-4ccc6a62b1ae/rtmlaunch_collision_detector-18.log].
log file: /home/oscuser/.ros/log/d555b9b6-a1a8-11e6-a6e4-4ccc6a62b1ae/rtmlaunch_collision_detector-18*.log[0m
Traceback (most recent call last):
  File "/opt/ros/indigo/share/openrtm_tools/scripts/rtmlaunch.py", line 13, in <module>
    rtmlaunch.main()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/openrtm_tools/rtmlaunch.py", line 253, in main
    tree.add_name_server(nameserver, [])
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rtctree/tree.py", line 131, in add_name_server
    self._parse_name_server(server, filter, dynamic=dynamic)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rtctree/tree.py", line 285, in _parse_name_server
    trim_filter(deepcopy(filter), 2), dynamic=dynamic)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rtctree/nameserver.py", line 54, in __init__
    self._parse_server(address, orb, filter)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rtctree/nameserver.py", line 79, in _parse_server
    self._parse_context(root_context, orb, filter)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rtctree/directory.py", line 111, in _parse_context
    self._process_binding(binding, orb, filter)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rtctree/directory.py", line 190, in _process_binding
[... cut ...]
Attached you can find an example of the set of scripts for start and killing nextage client and some logs that we have obtained. If you can reproduce this situation, may be the obtained logs could be interesting for determining more exactly what is happening. 

[corba_connection_error_logs.tar.gz](https://github.com/tork-a/rtmros_nextage/files/575275/corba_connection_error_logs.tar.gz)

[start_kill_test_script.tar.gz](https://github.com/tork-a/rtmros_nextage/files/575277/start_kill_test_script.tar.gz)

hectorherrero, 2016-11-07:
I have to make a correction regarding the error that appears on simulation:
`omniORB.CORBA.MARSHAL: CORBA.MARSHAL(omniORB.MARSHAL_MessageSizeExceedLimit, CORBA.COMPLETED_YES))`

**This error does appear in real robot** after executing lot of tests. Probably I have executed 4 or 5 times this start_kill_test_scripts with around 400 or 500 executions of the hironx_client.py. After that is not possible to connect to the robot, always the same MessageSizeExceedLimit error appears. I have tried with: 
ipython -irospack find hironx_ros_bridge/scripts/hironx.py -- --host hiro019 --robot RobotHardware0 --port 15005 --modelfile /opt/jsk/etc/HIRONX/model/main.wrl
And the following error was returned:
```
 robot.servoOn()
('[hrpsys.py] ', 'Exception during servoOn while removing JoingGroup. Your report to https://github.com/start-jsk/rtmros_hironx/issues about the issue you are seeing is appreciated.')
('[hrpsys.py] ', 'Exception during servoOn while removing setSelfGroups. Your report to https://github.com/start-jsk/rtmros_hironx/issues about the issue you are seeing is appreciated.')
[hrpsys.py]  exception occured
Move to Actual State, Just a minute.
[hrpsys.py]  post servo on motion trouble
Turn on Hand Servo
hrpsys ServoController not found. You may want to restart QNX/ControllerBox afterward and run the command again.
Out[3]: -1
```
I am not sure if CORBA.COMM_FAILURE and CORBA.MARSHAL error are related but with the last one is not possible to continue working and a restart of the QNX is necessary.

130s, 2016-11-07:
> ```
> ipython -irospack find hironx_ros_bridge/scripts/hironx.py -- --host hiro019 --robot RobotHardware0 --port 15005 --modelfile /opt/jsk/etc/HIRONX/model/main.wrl
> ```

You must have typo-ed for "-irospack" -- that must have been "-i rospack".

> And the following error was returned:
> ```
> robot.servoOn()
('[hrpsys.py] ', 'Exception during servoOn while removing JoingGroup. Your report to https://github.com/start-jsk/rtmros_hironx/issues about the issue you are seeing is appreciated.')
('[hrpsys.py] ', 'Exception during servoOn while removing setSelfGroups. Your report to https://github.com/start-jsk/rtmros_hironx/issues about the issue you are seeing is appreciated.')
[hrpsys.py]  exception occured
Move to Actual State, Just a minute.
[hrpsys.py]  post servo on motion trouble
Turn on Hand Servo
hrpsys ServoController not found. You may want to restart QNX/ControllerBox afterward and run the command again.
Out[3]: -1
> ```

This error is a bug recently introduced and a fix is already merged in source. https://github.com/start-jsk/rtmros_hironx/pull/475

For other error output I'll have to take a deeper look.

hectorherrero, 2016-12-12:
Any news about this topic?

534o, 2016-12-13:
did you check with hrpsyspy ?

hectorherrero, 2016-12-13:
Yes, with the same result, only varies the number of iterations until the error appears

534o, 2016-12-13:
that should be different error, since it never calls

```
  File "/opt/ros/hydro/lib/python2.7/dist-packages/hrpsys/hrpsys_config.py", line 1679, in init
    self.activateComps()
  File "/opt/ros/hydro/lib/python2.7/dist-packages/hrpsys/hrpsys_config.py", line 426, in activateComps
    rtm.serializeComponents(rtcList)
  File "/opt/ros/hydro/lib/python2.7/dist-packages/hrpsys/rtm.py", line 424, in serializeComponents
    if not ec._is_equivalent(rtc.ec)
```

534o, 2016-12-13:
sorry that I have confused since this thread includes several issues

1.  `   if not ec._is_equivalent(rtc.ec): AttributeError: 'NoneType' object has no attribute 'ec'`

This has been solved. If you're using hironx_ros_bridge newer than 1.1.16, activaeComps() will not called, so this error will not occur.
https://github.com/start-jsk/rtmros_hironx/commit/0ac478567d598fb57d63ff3fd00b087bc69a0c33

2. 
```
  File "/opt/ros/indigo/lib/python2.7/dist-packages/hrpsys/hrpsys_config.py", line 943, in checkSimulationMode
    self.rh_svc = narrow(self.rh.service("service0"), "RobotHardwareService")
  File "/opt/ros/indigo/lib/python2.7/dist-packages/hrpsys/rtm.py", line 62, in service
    return findService(self, port_name, type_name, instance_name)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/hrpsys/rtm.py", line 728, in findService
    ret, con_prof = port.connect(con_prof)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/OpenRTM_aist/RTM_IDL/RTC_idl.py", line 1214, in connect
    return _omnipy.invoke(self, "connect", _0_RTC.PortService._d_connect, args)
omniORB.CORBA.COMM_FAILURE: CORBA.COMM_FAILURE(omniORB.COMM_FAILURE_WaitingForReply, CORBA.COMPLETED_MAYBE)
terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::lock_error> >'
  what():  boost: mutex lock failed in pthread_mutex_lock: Invalid argument
```
I think this is very rare case, and only happens in start-up process, so if you run the node with `respawn` attribute, we can avoid the problem for now. if this happens frequently, we need to look into flexbotics_core_gui_exec_controller`

3. `hrpsys ServoController not found. You may want to restart QNX/ControllerBox afterward and run the command again.`

 this should be fixed in 1.1.18 of hironx_ros_bridge, https://github.com/start-jsk/rtmros_hironx/blob/indigo-devel/hironx_ros_bridge/CHANGELOG.rst

hectorherrero, 2016-12-20:
Hello, 

2. 

> I think this is very rare case, and only happens in start-up process, so if you run the node with respawn attribute, we can avoid the problem for now. if this happens frequently, we need to look into flexbotics_core_gui_exec_controller`

Ok, I agree, we can probably mitigate it respawning the node when this error appears.

3. 

Regarding this error I can confirm that 1.1.18 of hironx_ros_bridge does not fix it:
```
dpkg -s ros-indigo-hironx-ros-bridge | grep '^Version:' 
Version: 1.1.18-0trusty-20161115-213601-0800
```
After a certain number of executions it seems that a buffer is exceeded and a restart of the QNX is necessary.

```
$ ipython -i `rospack find nextage_ros_bridge`/script/nextage.py -- --host hiro019 --robot RobotHardware0 --port 15005 --modelfile /opt/jsk/etc/HIRONX/model/main.wrl
Python 2.7.6 (default, Oct 26 2016, 20:30:19) 
Type "copyright", "credits" or "license" for more information.

IPython 1.2.1 -- An enhanced Interactive Python.
?         -> Introduction and overview of IPython's features.
%quickref -> Quick reference.
help      -> Python's own help system.
object?   -> Details about 'object', use 'object??' for extra details.
ROS Master not found running. If you intended to use it (either for rosbridge or for Gazebo purposes, make sure you run necessary nodes.
configuration ORB with hiro019:15005
[hrpsys.py] waiting ModelLoader
[hrpsys.py] start hrpsys
[hrpsys.py] finding RTCManager and RobotHardware
[hrpsys.py]  waitForRTCManagerAndRoboHardware has renamed to waitForRTCManagerAndRoboHardware: Please update your code
[hrpsys.py] wait for RTCmanager : None
[hrpsys.py] wait for RobotHardware0 : <hrpsys.rtm.RTcomponent instance at 0x7f114445bb48> ( timeout 0 < 10)
[hrpsys.py] findComps -> RobotHardware : <hrpsys.rtm.RTcomponent instance at 0x7f114445bb48> isActive? = True 
[hrpsys.py] simulation_mode : False
[hrpsys.py]  Hrpsys controller version info: 
[hrpsys.py]    ms =  <hrpsys.rtm.RTCmanager instance at 0x7f114445be60>
[hrpsys.py]    ref =  <RTM._objref_Manager instance at 0x7f114445bea8>
---------------------------------------------------------------------------
MARSHAL                                   Traceback (most recent call last)
/usr/lib/python2.7/dist-packages/IPython/utils/py3compat.pyc in execfile(fname, *where)
    202             else:
    203                 filename = fname
--> 204             __builtin__.execfile(filename, *where)

/opt/ros/indigo/share/nextage_ros_bridge/script/nextage.py in <module>()
     72     # This is backward compatible so that users can still keep using `nxc`.
     73     # See http://code.google.com/p/rtm-ros-robotics/source/detail?r=6926
---> 74     robot.init(robotname=args.robot, url=args.modelfile)
     75 
     76     # ROS Client.

/opt/ros/indigo/lib/python2.7/dist-packages/nextage_ros_bridge/nextage_client.pyc in init(self, robotname, url)
    107         '''
    108         if not self.use_gazebo:
--> 109             HIRONX.init(self, robotname=robotname, url=url)
    110 
    111     def get_hand_version(self):

/opt/ros/indigo/lib/python2.7/dist-packages/hironx_ros_bridge/hironx_client.pyc in init(self, robotname, url)
    302         if self.ms and self.ms.ref :
    303             print self.configurator_name, "  ref = ", self.ms.ref
--> 304         if self.ms and self.ms.ref and len(self.ms.ref.get_component_profiles()) > 0:
    305             print self.configurator_name, "  version  = ", self.ms.ref.get_component_profiles()[0].version
    306         if self.ms and self.ms.ref and len(self.ms.ref.get_component_profiles()) > 0 and StrictVersion(self.ms.ref.get_component_profiles()[0].version) < StrictVersion('315.2.0'):

/opt/ros/indigo/lib/python2.7/dist-packages/OpenRTM_aist/RTM_IDL/Manager_idl.pyc in get_component_profiles(self, *args)
    166 
    167     def get_component_profiles(self, *args):
--> 168         return _omnipy.invoke(self, "get_component_profiles", _0_RTM.Manager._d_get_component_profiles, args)
    169 
    170     def get_profile(self, *args):

MARSHAL: CORBA.MARSHAL(omniORB.MARSHAL_MessageSizeExceedLimitOnServer, CORBA.COMPLETED_YES)
```

130s, 2017-04-05:
>> 3. `hrpsys ServoController not found. You may want to restart QNX/ControllerBox afterward and run the command again.`
>> 
>>  this should be fixed in 1.1.18 of hironx_ros_bridge, https://github.com/start-jsk/rtmros_hironx/blob/indigo-devel/hironx_ros_bridge/CHANGELOG.rst
> 
> Regarding this error I can confirm that 1.1.18 of hironx_ros_bridge does not fix it:

Looking at the error log you posted, it seems to have nothing to do with the "`ServoController not found`" issue mentioned in https://github.com/tork-a/rtmros_nextage/issues/248#issuecomment-266707027, doesn't it?

hectorherrero, 2017-04-05:
I think that does not have relation with the bug that was solved in 1.1.18 of hironx_ros_bridge. The problem seems to appear when a buffer is filled and until QNX is restarted this buffer is not cleaned.

534o, 2017-04-07:
sorry that you still have this problem.
Do you have this problem only with QNX? can you reproduce this with
simulation running on Linux?
Do you have any idea on what buffer is filled?

hectorherrero, 2017-04-25:
Sorry for the late response, I have repeat the tests and as I have commented above the error is reproducible both in the real robot and in simulation (Ubuntu PC with Indigo 14.04). 
I have no idea about the buffer, this was one of my suggestion (intuition) after obtaining this error message:
`MARSHAL: CORBA.MARSHAL(omniORB.MARSHAL_MessageSizeExceedLimitOnServer, CORBA.COMPLETED_YES)`