tork-a/rtmros_nextage #249: Write access to this repo is granted to another user @ros-pull-request-builder

130s, 2016-07-11:
Once my question [here](http://answers.ros.org/question/239060/usage-of-test_commit-and-test_pull_requests-in-rosdistro-file/) is resolved we may revoke the write access (I have a feeling that the ROS pull request test doesn't require write access to each repo, so hopefully I might be misunderstanding something).

k-okada, 2016-09-03:
did you fix this problem? if not, can we remove ros-pull-request-builder? I do not think I can take care of both travis and ros-pull-request-builder when we have problem...

130s, 2016-09-04:
Turned out that the check by the ROS buildfarm is the best suitable for libraries in lower level (e.g. roscpp or such). Since NXO packages are very high-level, along with ambiguous build errors (#253 etc.), I think we can remove it from our PR check, which I already did.

closed 2016-09-04 by 130s