tork-a/rtmros_nextage #260: getCurrentPosition() returns bad values

hectorherrero, 2016-10-05:
Hi,

after recent updates (I have detected last week) we are getting not coherent results of getCurrentPosition() function from nextage.py script. 

The difference between `getCurrentPosition()` and `TF` values is around 0.97m. the same value that we have detected long time ago:
```
In [7]: nxc.getCurrentPosition('LARM_JOINT5')
Out[7]: [0.3255751238415409, 0.18236314012331808, 1.037624522677471]

In [8]: nxc.getCurrentPosition('LARM_JOINT5', 'WAIST')
frame_name (LARM_JOINT5) is not supported
Out[8]: [0.3255751238415409, 0.18236314012331808, 1.037624522677471]

In [9]: nxc.getCurrentPosition('RARM_JOINT5')
Out[9]: [0.3255751238415409, -0.18236314012331808, 1.037624522677471]

In [10]: nxc.getCurrentPosition('RARM_JOINT5', 'WAIST')
frame_name (RARM_JOINT5) is not supported
Out[10]: [0.3255751238415409, -0.18236314012331808, 1.037624522677471]

--------------------------------------------

rosrun tf tf_echo 'WAIST' 'LARM_JOINT5_Link'
At time 441.695
- Translation: [0.326, 0.182, 0.068]
- Rotation: in Quaternion [-0.000, -0.708, -0.000, 0.706]
            in RPY (radian) [-3.073, -1.569, 3.073]
            in RPY (degree) [-176.095, -89.898, 176.084]

rosrun tf tf_echo 'WAIST' 'RARM_JOINT5_Link'
At time 475.975
- Translation: [0.326, -0.182, 0.068]
- Rotation: in Quaternion [0.000, -0.708, 0.000, 0.706]
            in RPY (radian) [3.073, -1.569, -3.073]
            in RPY (degree) [176.095, -89.898, -176.084]
```

However, hironx (I have tested with hironx.py script) is working well, giving expected values:
```
In [4]: robot.getCurrentPosition('LARM_JOINT5', 'WAIST')
frame_name (LARM_JOINT5) is not supported
Out[4]: [0.3255627368715471, 0.1823638733778268, 0.07462449717662004]

In [5]: robot.getCurrentPosition('LARM_JOINT5')
Out[5]: [0.3255627368715471, 0.1823638733778268, 0.07462449717662004]

------------------------------------------------

rosrun tf tf_echo 'WAIST' 'LARM_JOINT5_Link'
At time 258.695
- Translation: [0.326, 0.182, 0.075]
- Rotation: in Quaternion [-0.000, -0.708, -0.000, 0.706]
            in RPY (radian) [-3.073, -1.569, 3.073]
            in RPY (degree) [-176.082, -89.898, 176.072]
```

I have tried reverting to a previous version of rtmros_nextage (0.7.8 tag) but this do not solve the problem.
Have you got any idea what could have happened?

Thanks, 

Héctor

k-okada, 2016-10-07:
Sorry for inconvenience, The problem is that the version of hrpsys bumped to 315.10.x and version and our checking script  did not working well. We have fixed this issue at 
https://github.com/start-jsk/rtmros_hironx/pull/469
and if you'd like to apply hotfix to your deb  installed scripts, please try
```
wget https://raw.githubusercontent.com/tork-a/88_tools/master/hotfixer/hot_fix.sh
bash ./hot_fix.sh start-jsk/rtmros_hironx 469
```

130s, 2016-10-13:
I just made a new release 1.1.17 of rtmros_hironx https://github.com/ros/rosdistro/pull/12940
Its binary will be available in the next public sync, and if you're keen on it then you'll be able to grab them from [shadow-fixed](http://wiki.ros.org/ShadowRepository) repo prior to pub sync.

hectorherrero, 2016-10-17:
Thanks, after applying the patch all is working as expected. 

Do you know when will be available on normal ros repositories?

130s, 2016-10-17:
> Do you know when will be available on normal ros repositories?

It's OSRF that decides the schedule, but I just [posted a request](http://discourse.ros.org/t/preparing-for-indigo-sync-2016-09-21/506/3) so if you're keen on the timing please watch that thread.

closed 2016-10-17 by 130s