tork-a/rtmros_nextage #265: roslaunch nextage_ros_bridge nextage_ros_bridge_real.launch nameserver:=nextage doesn't work

shzargar, 2016-10-17:
Hi, 
I am a beginner, and start working with a nextage open robot. I had difficulty to connect to the robot (the same issues as in https://github.com/start-jsk/rtmros_hironx/issues/33) even with only eth0 connected.

I tried to make an update $sudo apt-get update && sudo apt-get dist-upgrade (as it is suggested in http://wiki.ros.org/rtmros_nextage/Tutorials/Quick%20Start%20for%20Robot%20Users)

After a difficult restart, I did the same thing and this time, I have this error in a loop:

 CORBA::SystemException raised by ModelLoader: IDL:omg.org/CORBA/INV_OBJREF:1.0
omniORB: ERROR -- the application attempted to invoke an operation
 on a nil reference.

I appreciate any help.
Thanks

130s, 2016-10-18:
A few things,

Without taking a deeper look, I assume you're using a NEXTAGE Open with `hrpsys` controller of version 315.1.x. For that version, you need to run `rtm_ros bridge` launch with `MODEL_FILE` argument.

```
roslaunch nextage_ros_bridge nextage_ros_bridge_real.launch MODEL_FILE:=/opt/jsk/etc/HIRONX/model/main.wrl
```

For the detail of this change, see the announcement https://github.com/tork-a/rtmros_nextage/issues/153

BTW, knowing it's very cumbersome to add such a convoluted argument, I proposed a new launch file that allows launch command without an argument https://github.com/tork-a/rtmros_nextage/pull/266.

----

Above should solve your issue, but I respond to your other comments:

> I had difficulty to connect to the robot (the same issues as in start-jsk/rtmros_hironx#33) even with only eth0 connected.

If you're still experiencing an issue regarding that, run the following commands on Ubuntu and post:

- `ifconfig`
- `netstat -r`

> I tried to make an update $sudo apt-get update && sudo apt-get dist-upgrade (as it is suggested in http://wiki.ros.org/rtmros_nextage/Tutorials/Quick%20Start%20for%20Robot%20Users)

I [updated the wiki to clarify](http://wiki.ros.org/action/info/rtmros_nextage/Tutorials/Quick%20Start%20for%20Robot%20Users?action=diff&rev2=52&rev1=51) that those commands update ONLY software on Ubuntu, which means that the lower level controller software doesn't get updated (hope [this tutorial section](http://wiki.ros.org/rtmros_nextage/Tutorials/Install%20NEXTAGE%20OPEN%20software%20on%20your%20machine#Overview_of_system) helps) if that's what you expected.

shzargar, 2016-10-18:
Thanks for your reply. First, when you are talking about hrpsys version of controller, do you mean QNX? Then, how can I make sure that is the version you mentioned?

I added the model to roslunch, I disconnected from internet to make sure that eth0 is the only connection (when I am connected to internet, the error is [rtmlaunch] hrpsys_ros_bridge.launch did not work on both OpenRTM 1.0.0 and 1.1.0 ), and I got the below ([rtm.py]    Failed to connect RobotHardware0.servoState to None([None])) and no hrpsys viewer on screen:

... logging to /home/nxouser/.ros/log/b0c2719a-9576-11e6-a674-000bab77d7ba/roslaunch-tork-desktop-11346.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://tork-desktop:45848/

SUMMARY
========

PARAMETERS
 * /HrpsysSeqStateROSBridge/publish_sensor_transforms
 * /collision_state/comp_name
 * /controller_configuration
 * /diagnostic_aggregator/analyzers/computers/contains
 * /diagnostic_aggregator/analyzers/computers/path
 * /diagnostic_aggregator/analyzers/computers/type
 * /diagnostic_aggregator/analyzers/hrpsys/contains
 * /diagnostic_aggregator/analyzers/hrpsys/path
 * /diagnostic_aggregator/analyzers/hrpsys/type
 * /diagnostic_aggregator/analyzers/mode/contains
 * /diagnostic_aggregator/analyzers/mode/path
 * /diagnostic_aggregator/analyzers/mode/type
 * /diagnostic_aggregator/analyzers/motor/contains
 * /diagnostic_aggregator/analyzers/motor/path
 * /diagnostic_aggregator/analyzers/motor/type
 * /diagnostic_aggregator/base_path
 * /diagnostic_aggregator/pub_rate
 * /robot_description
 * /robot_pose_ekf/debug
 * /robot_pose_ekf/freq
 * /robot_pose_ekf/imu_used
 * /robot_pose_ekf/odom_used
 * /robot_pose_ekf/output_frame
 * /robot_pose_ekf/self_diagnose
 * /robot_pose_ekf/sensor_timeout
 * /robot_pose_ekf/vo_used
 * /rosdistro
 * /rosversion

NODES
  /
    CollisionDetectorComp (hrpsys/CollisionDetectorComp)
    DataLoggerServiceROSBridge (hrpsys_ros_bridge/DataLoggerServiceROSBridgeComp)
    ForwardKinematicsServiceROSBridge (hrpsys_ros_bridge/ForwardKinematicsServiceROSBridgeComp)
    HrpsysJointTrajectoryBridge (hrpsys_ros_bridge/HrpsysJointTrajectoryBridge)
    HrpsysSeqStateROSBridge (hrpsys_ros_bridge/HrpsysSeqStateROSBridge)
    RobotHardwareServiceROSBridge (hrpsys_ros_bridge/RobotHardwareServiceROSBridgeComp)
    SequencePlayerServiceROSBridge (hrpsys_ros_bridge/SequencePlayerServiceROSBridgeComp)
    StateHolderServiceROSBridge (hrpsys_ros_bridge/StateHolderServiceROSBridgeComp)
    base_footprint_rootlink (tf/static_transform_publisher)
    collision_state (hrpsys_ros_bridge/collision_state.py)
    diagnostic_aggregator (diagnostic_aggregator/aggregator_node)
    hrpsys_profile (hrpsys_ros_bridge/hrpsys_profile.py)
    hrpsys_py (hironx_ros_bridge/hironx.py)
    hrpsys_ros_diagnostics (hrpsys_ros_bridge/diagnostics.py)
    hrpsys_state_publisher (robot_state_publisher/state_publisher)
    robot_pose_ekf (robot_pose_ekf/robot_pose_ekf)
    rtmlaunch_collision_detector (openrtm_tools/rtmlaunch.py)
    rtmlaunch_hrpsys_ros_bridge (openrtm_tools/rtmlaunch.py)
    sensor_ros_bridge_connect (hrpsys_ros_bridge/sensor_ros_bridge_connect.py)

WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
[... 449 lines cut ...]
[rtmlaunch] Activate <- Inactive /nextage:15005/ForwardKinematicsServiceROSBridge.rtc
[rtmlaunch] Activate <- Inactive /nextage:15005/StateHolderServiceROSBridge.rtc
[rtmlaunch] Activate <- Inactive /nextage:15005/RobotHardwareServiceROSBridge.rtc
[ INFO] [1476824865.394873208]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 201[Hz] (exec 1739 Hz, dropped 201)
[ INFO] [1476824866.395589278]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 200[Hz] (exec 1735 Hz, dropped 200)
[INFO] [WallTime: 1476824866.468630] check 4 collision status, 0.799995 Hz
[ INFO] [1476824867.400687349]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 201[Hz] (exec 1743 Hz, dropped 201)
[ INFO] [1476824868.405497670]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 201[Hz] (exec 1740 Hz, dropped 201)
[ INFO] [1476824869.410499093]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 201[Hz] (exec 1742 Hz, dropped 201)
[rtmlaunch] check connection/activation
[rtmlaunch] Activate <- Inactive /nextage:15005/CollisionDetector0.rtc
[ INFO] [1476824870.415007496]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 201[Hz] (exec 1740 Hz, dropped 201)
[ INFO] [1476824871.420138538]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 201[Hz] (exec 1744 Hz, dropped 201)
[INFO] [WallTime: 1476824871.488599] check 4 collision status, 0.796813 Hz
[ INFO] [1476824872.420773352]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 200[Hz] (exec 1736 Hz, dropped 200)
[ INFO] [1476824873.426102326]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 201[Hz] (exec 1743 Hz, dropped 201)
[ INFO] [1476824874.430518020]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 201[Hz] (exec 1743 Hz, dropped 201)
[rtmlaunch] check connection/activation
[rtmlaunch] Activate <- Inactive /nextage:15005/HrpsysSeqStateROSBridge0.rtc
[rtmlaunch] Activate <- Inactive /nextage:15005/HrpsysJointTrajectoryBridge0.rtc
[rtmlaunch] Activate <- Inactive /nextage:15005/DataLoggerServiceROSBridge.rtc
[rtmlaunch] Activate <- Inactive /nextage:15005/SequencePlayerServiceROSBridge.rtc
[rtmlaunch] Activate <- Inactive /nextage:15005/ForwardKinematicsServiceROSBridge.rtc
[rtmlaunch] Activate <- Inactive /nextage:15005/StateHolderServiceROSBridge.rtc
[rtmlaunch] Activate <- Inactive /nextage:15005/RobotHardwareServiceROSBridge.rtc

shzargar, 2016-10-18:
with internet connection:
nxouser@tork-desktop:~$ ifconfig
eth0      Link encap:Ethernet  HWaddr 00:0b:ab:77:d7:ba  
          inet addr:192.168.128.50  Bcast:192.168.128.255  Mask:255.255.255.0
          inet6 addr: fe80::20b:abff:fe77:d7ba/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:2304559 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1917187 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:1504775577 (1.5 GB)  TX bytes:204202742 (204.2 MB)
          Interrupt:20 Memory:df200000-df220000 

eth1      Link encap:Ethernet  HWaddr 00:0b:ab:77:d7:bb  
          inet addr:10.180.129.15  Bcast:10.180.129.127  Mask:255.255.255.128
          inet6 addr: fe80::20b:abff:fe77:d7bb/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:55529 errors:0 dropped:0 overruns:0 frame:0
          TX packets:19946 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:67111001 (67.1 MB)  TX bytes:2272562 (2.2 MB)
          Memory:df100000-df120000 

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:3766531 errors:0 dropped:0 overruns:0 frame:0
          TX packets:3766531 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:1037333117 (1.0 GB)  TX bytes:1037333117 (1.0 GB)

nxouser@tork-desktop:~$ netstat -r
Kernel IP routing table
Destination     Gateway         Genmask         Flags   MSS Window  irtt Iface
default         10.180.129.1    0.0.0.0         UG        0 0          0 eth1
10.180.129.0    *               255.255.255.128 U         0 0          0 eth1
link-local      *               255.255.0.0     U         0 0          0 eth0
192.168.128.0   *               255.255.255.0   U         0 0          0 eth0

130s, 2016-10-19:
In your https://github.com/tork-a/rtmros_nextage/issues/265#issuecomment-254657955 where you disconnect the cable to the internet, I don't see the issue you originally raised, which is good.

In https://github.com/tork-a/rtmros_nextage/issues/265#issuecomment-254658308 where you probably connect internet cable as well, what did you get when you ran `roslaunch`? Network config seems fine to me but not entirely sure. Make sure you can `ping ros.org` -- if it can't, you need to resolve a basic network issue first.

shzargar, 2016-11-02:
roslaunch nextage_ros_bridge nextage_ros_bridge_real.launch MODEL_FILE:=/opt/jsk/etc/HIRONX/model/main.wrl

This command is working, even though is giving some error messages.

I bug, since in Hydro before update, when I run this command (without model) hrpsys viewer was launching. With this command, even with errors and the fact that hrpsys viewer doesn't show up, we can go ahead and run moveit, then python to move the robot with python commands.

130s, 2016-11-05:
Glad that things got moving forward for you!

I've been under impression that the tutorials might have confused you about how to launch ROS process by `nextage_ros_bridge*.launch` files.

- One starts with `rtmlaunch` is always used for simulation ONLY. You'll never use this command with the real robot.
- For the real robot, always use `nextage_ros_bridge_real*.launch`, with **`roslaunch`** command, not rtmlaunch.

At this moment some error messages are expected to appear.

closed 2016-11-05 by 130s

shzargar, 2016-11-05:
Isaac,

Thanks for your comment, actually, I think that I have always used roslaunch for real robot. The fact is that using roslaunch for the real robot, I see the rtmlaunch error, which I learned to ignore it and go forward to launch moveit.

Thanks,
Amir

130s, 2016-11-06:
I see.

If possible, could you share the full output of the **roslaunch** command where you see `rtmlaunch` included (I don't have an access to real robot now)? I saw you posted https://github.com/tork-a/rtmros_nextage/issues/265#issuecomment-254657955 that I can't tell clearly if it comes from real or simulated robot. If you can confirm that is from real, that's fine.

shzargar, 2016-11-08:
As you can see that is roslaunch, but if you search "[rtmlaunch] [ERROR]" and "[rtmlaunch] Could not Activate" you will see the rtmlaunch errors

shzargar@Nxtag:~$  roslaunch nextage_ros_bridge nextage_ros_bridge_real.launch MODEL_FILE:=/opt/jsk/etc/HIRONX/model/main.wrl
... logging to /home/shzargar/.ros/log/da402964-a5f6-11e6-874e-98e7f45b297c/roslaunch-Nxtag-7083.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://localhost:33178/

SUMMARY
========

PARAMETERS
 * /HrpsysSeqStateROSBridge/publish_sensor_transforms: True
 * /collision_state/comp_name: CollisionDetector0
 * /controller_configuration: [{'group_name': '...
 * /diagnostic_aggregator/analyzers/computers/contains: ['HD Temp', 'CPU ...
 * /diagnostic_aggregator/analyzers/computers/path: Computers
 * /diagnostic_aggregator/analyzers/computers/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/hrpsys/contains: ['hrpEC', 'Emerge...
 * /diagnostic_aggregator/analyzers/hrpsys/path: Hrpsys
 * /diagnostic_aggregator/analyzers/hrpsys/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/mode/contains: ['Operating Mode'...
 * /diagnostic_aggregator/analyzers/mode/path: Mode
 * /diagnostic_aggregator/analyzers/mode/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/motor/contains: ['Motor']
 * /diagnostic_aggregator/analyzers/motor/path: Motor
 * /diagnostic_aggregator/analyzers/motor/type: diagnostic_aggreg...
 * /diagnostic_aggregator/base_path: 
 * /diagnostic_aggregator/pub_rate: 1.0
 * /robot_description: <?xml version="1....
 * /rosdistro: indigo
 * /rosversion: 1.11.20
 * /rtmnameserver_port: 15005
 * /rtmnameserver_robotname: RobotHardware0

NODES
  /
    CollisionDetectorComp (hrpsys/CollisionDetectorComp)
    DataLoggerServiceROSBridge (hrpsys_ros_bridge/DataLoggerServiceROSBridgeComp)
    ForwardKinematicsServiceROSBridge (hrpsys_ros_bridge/ForwardKinematicsServiceROSBridgeComp)
    HrpsysJointTrajectoryBridge (hrpsys_ros_bridge/HrpsysJointTrajectoryBridge)
    HrpsysSeqStateROSBridge (hrpsys_ros_bridge/HrpsysSeqStateROSBridge)
    RobotHardwareServiceROSBridge (hrpsys_ros_bridge/RobotHardwareServiceROSBridgeComp)
    SequencePlayerServiceROSBridge (hrpsys_ros_bridge/SequencePlayerServiceROSBridgeComp)
    StateHolderServiceROSBridge (hrpsys_ros_bridge/StateHolderServiceROSBridgeComp)
    collision_state (hrpsys_ros_bridge/collision_state.py)
    diagnostic_aggregator (diagnostic_aggregator/aggregator_node)
    hrpsys_profile (hrpsys_ros_bridge/hrpsys_profile.py)
    hrpsys_py (hironx_ros_bridge/hironx.py)
    hrpsys_ros_diagnostics (hrpsys_ros_bridge/diagnostics.py)
    hrpsys_state_publisher (robot_state_publisher/state_publisher)
    rqt_hironxo (rqt_gui/rqt_gui)
    rtmlaunch_collision_detector (openrtm_tools/rtmlaunch.py)
    rtmlaunch_hrpsys_ros_bridge (openrtm_tools/rtmlaunch.py)
    sensor_ros_bridge_connect (hrpsys_ros_bridge/sensor_ros_bridge_connect.py)

WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtactivate
[... 268 lines cut ...]
[rtmlaunch] Wait for  /nextage:15005/RobotHardware0.rtc:q   13 /30
[rtmlaunch] Wait for  /nextage:15005/CollisionDetector0.rtc   23 /30
[rtmlaunch] Wait for  /nextage:15005/RobotHardware0.rtc:q   14 /30
[rtmlaunch] Wait for  /nextage:15005/CollisionDetector0.rtc   24 /30
[ WARN] [1478639090.062876340]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is not executed last 5[sec]
[rtmlaunch] Wait for  /nextage:15005/RobotHardware0.rtc:q   15 /30
[rtmlaunch] Wait for  /nextage:15005/CollisionDetector0.rtc   25 /30
[rtmlaunch] Wait for  /nextage:15005/RobotHardware0.rtc:q   16 /30
[rtmlaunch] Wait for  /nextage:15005/CollisionDetector0.rtc   26 /30
[rtmlaunch] Wait for  /nextage:15005/RobotHardware0.rtc:q   17 /30
[rtmlaunch] Wait for  /nextage:15005/CollisionDetector0.rtc   27 /30
[rtmlaunch] Wait for  /nextage:15005/RobotHardware0.rtc:q   18 /30
[rtmlaunch] Wait for  /nextage:15005/CollisionDetector0.rtc   28 /30
[rtmlaunch] Wait for  /nextage:15005/RobotHardware0.rtc:q   19 /30
[ WARN] [1478639095.063385768]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is not executed last 5[sec]
[rtmlaunch] Wait for  /nextage:15005/CollisionDetector0.rtc   29 /30
[rtmlaunch] Wait for  /nextage:15005/RobotHardware0.rtc:q   20 /30
[rtmlaunch] Could not Activate ( nextage:15005/$(arg COLLISIONDETECTOR_INSTANCE_NAME).rtc ) :  No such object: /nextage:15005/CollisionDetector0.rtc 
[rtmlaunch] Wait for  /nextage:15005/RobotHardware0.rtc:q   21 /30
[rtmlaunch] Wait for  /nextage:15005/RobotHardware0.rtc:q   22 /30
[rtmlaunch] Wait for  /nextage:15005/RobotHardware0.rtc:q   23 /30
[rtmlaunch] Wait for  /nextage:15005/RobotHardware0.rtc:q   24 /30
[ WARN] [1478639100.063868904]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is not executed last 5[sec]
[rtmlaunch] Wait for  /nextage:15005/RobotHardware0.rtc:q   25 /30
[rtmlaunch] Wait for  /nextage:15005/RobotHardware0.rtc:q   26 /30

130s, 2016-11-16:
Thanks for the detail. Seems like I was wrong about `rtmlaunch`. I've updated the comment https://github.com/tork-a/rtmros_nextage/issues/265#issuecomment-258599949 to tell that the string rtmlaunch can be printed even when you're working with real robots (although I think users will never have to explicitly run the command rtmlaunch).

k-okada, 2016-11-17:
The problem is wen can not find model files in QNX
```
ModelLoaderException : /opt/jsk/etc/NEXTAGE/model/main.wrl cannot be found.
```
and seems 
```
roslaunch nextage_ros_bridge nextage_ros_bridge_real.launch MODEL_FILE:=/opt/jsk/etc/HIRONX/model/main.wrl
```
is not working correctly, please check if we have same launch file
```
$ roscat nextage_ros_bridge nextage_ros_bridge_real.launch 
<launch>
  <arg name="COLLADA_FILE" default="$(find nextage_description)/models/main.dae" /> 
  <arg name="CONF_FILE_COLLISIONDETECT" default="$(find nextage_ros_bridge)/conf/realrobot_fixedpath.conf" />
  <arg name="corbaport" default="15005" />
  <arg name="MODEL_FILE" default="/opt/jsk/etc/NEXTAGE/model/main.wrl" /> <!-- This shouldn't be changed unless you know what you're doing; this is the location of vrml file internal to the robot (on QNX OS). -->
  <arg name="nameserver" default="nextage" /> <!-- Host name of the QNX. -->
  <arg name="USE_SERVOCONTROLLER" default="false" />

  <env name="RTCTREE_NAMESERVERS" value="$(arg nameserver):$(arg corbaport)" />

  <include file="$(find hironx_ros_bridge)/launch/hironx_ros_bridge_real.launch" >
    <arg name="COLLADA_FILE" value="$(arg COLLADA_FILE)" /> 
    <arg name="CONF_FILE_COLLISIONDETECT" value="$(arg CONF_FILE_COLLISIONDETECT)" />
    <arg name="corbaport" value="$(arg corbaport)" />
    <arg name="MODEL_FILE" value="$(arg MODEL_FILE)" />
    <arg name="nameserver" value="$(arg nameserver)" />
    <arg name="USE_SERVOCONTROLLER" value="$(arg USE_SERVOCONTROLLER)" />
  </include>
</launch>
```