tork-a/rtmros_nextage #268: Is it possible to connect to QNX with a laptop and how?

shzargar, 2016-10-21:
Hi, 

Is it possible to connect to QNX with a laptop and how?
I want to check if Ubuntu 14.04 and Indigo can solve my problem to launch hrpsys, before upgrading UI computer.

Thanks

130s, 2016-10-21:
Go through quick start http://wiki.ros.org/rtmros_nextage/Tutorials/Quick%20Start%20for%20Robot%20Users

You may want to add to `/etc/hosts` the following):

```
# NXO. Comment in only when I connect to NEXTAGE.     
192.168.128.10 nextage
```

(IP address may vary per your environment. This should be done by the manufacturer for the UI Controller's Ubuntu. If this is not documented I'll add to a tutorial, sorry)

shzargar, 2016-10-22:
I have read the quick start many times, but it is not talking how to connect another computer (not UI computer) to QNX. There are lots of connections between QNX and UI computer, what should be done if another computer or a laptop is used instead of UI computer?
Or I should only connect the second computer via lan connection which is beside, the green and blue button on robot's back? and don't touch anything else.
Tnx

130s, 2016-10-24:
Quick Start tutorial only talks about the very basic situation (primarilly for those who just purchased the robot).
For adding another Ubuntu machine, you can see [this section of a tutorial](http://wiki.ros.org/rtmros_nextage/Tutorials/Install%20NEXTAGE%20OPEN%20software%20on%20your%20machine#Working_with_a_real_robot).

> Or I should only connect the second computer via lan connection which is beside, the green and blue button on robot's back? and don't touch anything else.

Unfortunately I'm not entirely sure about this hardware-level setting. Your best bet is consult the manufacturer (I'm from TORK, software provider) for the network configuration.
My guess is the connector you mentioned is reserved for connecting ethernet cable to the LAN. So yes you might connect your own Ubuntu computer from LAN via that connector. But you can also connect it to the internal hub (if you open the back lid of the base you'll see a hub with a few ports).

shzargar, 2016-10-25:
Thanks, I will contact Kawada directly.

130s, 2016-12-06:
Closing since this is beyond the scope of NEXTAGE's opensource software. Feel free to add comments if anything remains unclear regarding software.

closed 2016-12-06 by 130s