tork-a/rtmros_nextage #272: Simulation- python problem

shzargar, 2016-11-06:
In this sequence to simulate the robot motion:

on the first terminal;
rtmlaunch nextage_ros_bridge nextage_ros_bridge_simulation.launch

on the second one;
roslaunch nextage_moveit_config moveit_planning_execution.launch

and on the third one;
ipython -i `rospack find nextage_ros_bridge`/script/nextage.py

I am getting this error.

Python 2.7.6 (default, Jun 22 2015, 17:58:13) 
Type "copyright", "credits" or "license" for more information.

IPython 1.2.1 -- An enhanced Interactive Python.
?         -> Introduction and overview of IPython's features.
%quickref -> Quick reference.
help      -> Python's own help system.
object?   -> Details about 'object', use 'object??' for extra details.
configuration ORB with Nxtag:15005
An exception has occurred, use %tb to see the full traceback.

SystemExit: [ERROR] Connection Failed with the Nameserver (hostname=Nxtag port=15005).
Make sure the hostname is correct and the Nameserver is running.
CORBA.TRANSIENT(omniORB.TRANSIENT_ConnectFailed, CORBA.COMPLETED_NO)

In [1]: 

and then those commands for robot. is not available.

shzargar, 2016-11-06:
The above sequence is based on http://wiki.ros.org/rtmros_nextage/Tutorials/Operating%20Hiro%2C%20NEXTAGE%20OPEN.

In another tutorial (http://wiki.ros.org/rtmros_nextage/Tutorials/Run%20simulation), it is not mentioned to launch moveit (Rviz). I confirm that with or without moveit, python is not working for me.

130s, 2016-11-07:
From your report only I can't tell what's going on. Could you share:

1. Tarball generated by the following command (github now allows attachments).
```
curl -sL https://raw.githubusercontent.com/start-jsk/rtmros_common/master/hrpsys_tools/scripts/rtmroswtf.sh | bash -
```
(This fetches `~/.ros/log/latest` folder so you need to run this **right after** you see the issue.)

2. If possible, `~/.bashrc` file content (if it's not desirable to share the entire file, post only what seems to be related to ROS and NEXTAGE).

Particularly I'm concerned why the host `Nxtag` is related here (simulation by default tries to use `localhost`).

shzargar, 2016-11-08:
Something which seems to be related to this issue, before all of above (specially the first two, rtmlaunch for hrpsys and roslaunch for moveit) I have to run export ROS_HOSTNAME=localhost, otherwise I get error message.

This is my entire ~/.bashrc which is located in usr/share/base-files.

[dot.bashrc.tar.gz](https://github.com/tork-a/rtmros_nextage/files/579461/dot.bashrc.tar.gz)

534o, 2016-11-16:
sorry for late, may you have wrong ROS environment variable or network settings,
please provide output of rtmlaunch / roslaunch program, env | grep ROS, and cat /etc/hosts
 
beset

shzargar, 2016-11-16:
Output of roslaunch

shzargar@Nxtag:~$ roslaunch nextage_ros_bridge nextage_ros_bridge_real.launch MODEL_FILE:=/opt/jsk/etc/HIRONX/model/main.wrl
... logging to /home/shzargar/.ros/log/fc7a9f7e-ac44-11e6-832b-98e7f45b297c/roslaunch-Nxtag-31025.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://localhost:37700/

SUMMARY
========

PARAMETERS
 * /HrpsysSeqStateROSBridge/publish_sensor_transforms: True
 * /collision_state/comp_name: CollisionDetector0
 * /controller_configuration: [{'group_name': '...
 * /diagnostic_aggregator/analyzers/computers/contains: ['HD Temp', 'CPU ...
 * /diagnostic_aggregator/analyzers/computers/path: Computers
 * /diagnostic_aggregator/analyzers/computers/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/hrpsys/contains: ['hrpEC', 'Emerge...
 * /diagnostic_aggregator/analyzers/hrpsys/path: Hrpsys
 * /diagnostic_aggregator/analyzers/hrpsys/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/mode/contains: ['Operating Mode'...
 * /diagnostic_aggregator/analyzers/mode/path: Mode
 * /diagnostic_aggregator/analyzers/mode/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/motor/contains: ['Motor']
 * /diagnostic_aggregator/analyzers/motor/path: Motor
 * /diagnostic_aggregator/analyzers/motor/type: diagnostic_aggreg...
 * /diagnostic_aggregator/base_path: 
 * /diagnostic_aggregator/pub_rate: 1.0
 * /robot_description: <?xml version="1....
 * /rosdistro: indigo
 * /rosversion: 1.11.20
 * /rtmnameserver_port: 15005
 * /rtmnameserver_robotname: RobotHardware0

NODES
  /
    CollisionDetectorComp (hrpsys/CollisionDetectorComp)
    DataLoggerServiceROSBridge (hrpsys_ros_bridge/DataLoggerServiceROSBridgeComp)
    ForwardKinematicsServiceROSBridge (hrpsys_ros_bridge/ForwardKinematicsServiceROSBridgeComp)
    HrpsysJointTrajectoryBridge (hrpsys_ros_bridge/HrpsysJointTrajectoryBridge)
    HrpsysSeqStateROSBridge (hrpsys_ros_bridge/HrpsysSeqStateROSBridge)
    RobotHardwareServiceROSBridge (hrpsys_ros_bridge/RobotHardwareServiceROSBridgeComp)
    SequencePlayerServiceROSBridge (hrpsys_ros_bridge/SequencePlayerServiceROSBridgeComp)
    StateHolderServiceROSBridge (hrpsys_ros_bridge/StateHolderServiceROSBridgeComp)
    collision_state (hrpsys_ros_bridge/collision_state.py)
    diagnostic_aggregator (diagnostic_aggregator/aggregator_node)
    hrpsys_profile (hrpsys_ros_bridge/hrpsys_profile.py)
    hrpsys_py (hironx_ros_bridge/hironx.py)
    hrpsys_ros_diagnostics (hrpsys_ros_bridge/diagnostics.py)
    hrpsys_state_publisher (robot_state_publisher/state_publisher)
    rqt_hironxo (rqt_gui/rqt_gui)
    rtmlaunch_collision_detector (openrtm_tools/rtmlaunch.py)
    rtmlaunch_hrpsys_ros_bridge (openrtm_tools/rtmlaunch.py)
    sensor_ros_bridge_connect (hrpsys_ros_bridge/sensor_ros_bridge_connect.py)

WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtactivate
[... 264 lines cut ...]
[rtm.py]      sh.zmpOut and log.sh_zmpOut are already connected
[hrpsys.py]   setupLogger : q arleady exists in DataLogger
[rtm.py]      ic.q and log.ic_q are already connected
[hrpsys_py-2] killing on exit
omniORB: Caught an unexpected Python exception during up-call.
Traceback (most recent call last):
  File "/opt/ros/indigo/lib/python2.7/dist-packages/OpenRTM_aist/RTM_IDL/RTC_idl.py", line 1201, in __init__
    def __init__(self):
KeyboardInterrupt
Exception AttributeError: "_objref_PortService instance has no attribute '_Object__release'" in <bound method _objref_PortService.__del__ of <RTC._objref_PortService instance at 0x7f5975d10cb0>> ignored
Traceback (most recent call last):
  File "/opt/ros/indigo/share/hironx_ros_bridge/scripts/hironx.py", line 76, in <module>
    robot.init(robotname=args.robot, url=args.modelfile)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/hironx_ros_bridge/hironx_client.py", line 330, in init
    self.setupLogger()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/hrpsys/hrpsys_config.py", line 866, in setupLogger
    self.connectLoggerPort(self.el, 'q')
  File "/opt/ros/indigo/lib/python2.7/dist-packages/hrpsys/hrpsys_config.py", line 803, in connectLoggerPort
    log_name = log_name if log_name else artc.name() + "_" + sen_name
  File "/opt/ros/indigo/lib/python2.7/dist-packages/hrpsys/rtm.py", line 185, in name
    cprof = self.ref.get_component_profile()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/OpenRTM_aist/RTM_IDL/RTC_idl.py", line 1344, in get_component_profile
.
.
.

shzargar, 2016-11-16:
It is hard to take the output of rtmlaunch, since it fills the whole terminal ans some parts will be missing. If there is a command that send the output of a command to a file or to Github, please let me know. By the way that i what I was able to get after rtmlaunch.
.
.
.

The model was successfully loaded ! 
[rtmlaunch] starting...  /opt/ros/indigo/share/hrpsys_ros_bridge/launch/collision_detector.launch
[rtmlaunch] RTCTREE_NAMESERVERS localhost:15005 localhost:15005 
[rtmlaunch] SIMULATOR_NAME HiroNX(Robot)0 
[rtmlaunch] starting...  /opt/ros/indigo/share/hrpsys_ros_bridge/launch/hrpsys_ros_bridge.launch
[rtmlaunch] RTCTREE_NAMESERVERS localhost:15005 localhost:15005 
[rtmlaunch] SIMULATOR_NAME HiroNX(Robot)0 
Collada Warning: fill asset info
Collada Warning: Target Node 'visual1/node_joint0_axis0' not found
Collada Warning: could not find binding for axis: kmodel1/jointsid1000/axis0, axis0
Collada Warning: could not find binding for axis: kmodel1/jointsid1000/axis0, axis0
Collada Warning: joint HEAD_JOINT1 has axis: 0.000000 1.000000 0.000000
Collada Warning: joint HEAD_JOINT0 has axis: 0.000000 0.000000 1.000000
Collada Warning: joint RARM_JOINT5 has axis: 1.000000 0.000000 0.000000
Collada Warning: joint RARM_JOINT4 has axis: 0.000000 1.000000 0.000000
Collada Warning: joint RARM_JOINT3 has axis: 0.000000 0.000000 1.000000
Collada Warning: joint RARM_JOINT2 has axis: 0.000000 1.000000 0.000000
Collada Warning: joint RARM_JOINT1 has axis: 0.000000 1.000000 0.000000
Collada Warning: joint RARM_JOINT0 has axis: 0.000000 0.000000 1.000000
Collada Warning: joint LARM_JOINT5 has axis: 1.000000 0.000000 0.000000
Collada Warning: joint LARM_JOINT4 has axis: 0.000000 1.000000 0.000000
Collada Warning: joint LARM_JOINT3 has axis: 0.000000 0.000000 1.000000
Collada Warning: joint LARM_JOINT2 has axis: 0.000000 1.000000 0.000000
Collada Warning: joint LARM_JOINT1 has axis: 0.000000 1.000000 0.000000
Collada Warning: joint LARM_JOINT0 has axis: 0.000000 0.000000 1.000000
Collada Warning: joint CHEST_JOINT0 has axis: 0.000000 0.000000 1.000000
Collada Warning: cannot find instance_actuator for attached sensor HiroNX:WAIST
The model was successfully loaded ! 
Loading body customizer "/opt/ros/indigo/share/OpenHRP-3.1/customizer/libSampleCustomizer.so" for springJoint
[rtmlaunch] Could not start rtmlaunch.py, Caught exception ( CORBA.COMM_FAILURE(omniORB.COMM_FAILURE_WaitingForReply, CORBA.COMPLETED_MAYBE) )
[rtmlaunch] .. Could not connect to NameServer at  localhost:15005 , Caught exception ( CORBA.COMM_FAILURE(omniORB.COMM_FAILURE_WaitingForReply, CORBA.COMPLETED_MAYBE) )
[rtmlaunch] .. Please make sure that you have NameServer running at localhost:15005/`
[rtmlaunch] .. You can check with `rtls localhost:15005/`
[rtmlaunch] Could not start rtmlaunch.py, Caught exception ( CORBA.COMM_FAILURE(omniORB.COMM_FAILURE_WaitingForReply, CORBA.COMPLETED_MAYBE) )
[rtmlaunch] .. Could not connect to NameServer at  localhost:15005 , Caught exception ( CORBA.COMM_FAILURE(omniORB.COMM_FAILURE_WaitingForReply, CORBA.COMPLETED_MAYBE) )
[rtmlaunch] .. Please make sure that you have NameServer running at localhost:15005/`
[rtmlaunch] .. You can check with `rtls localhost:15005/`
[hrpsys-3] process has died [pid 29923, exit code -11, cmd /opt/ros/indigo/lib/hrpsys/hrpsys-simulator /opt/ros/indigo/share/nextage_ros_bridge/conf/nextage_nosim.xml -o manager.is_master:YES -o corba.nameservers:localhost:15005 -o naming.formats:%n.rtc -o logger.file_name:/tmp/rtc%p.log -endless -realtime -o exec_cxt.periodic.rate:1000000 -o manager.shutdown_onrtcs:NO -o manager.modules.load_path:/opt/ros/indigo/share/hrpsys/lib -o manager.modules.preload:HGcontroller.so -o manager.components.precreate:HGcontroller -o exec_cxt.periodic.type:SynchExtTriggerEC -o example.SequencePlayer.config_file:/opt/ros/indigo/share/nextage_ros_bridge/conf/nextage.conf -o example.ForwardKinematics.config_file:/opt/ros/indigo/share/nextage_ros_bridge/conf/nextage.conf -o example.ImpedanceController.config_file:/opt/ros/indigo/share/nextage_ros_bridge/conf/nextage.conf -o example.AutoBalancer.config_file:/opt/ros/indigo/share/nextage_ros_bridge/conf/nextage.conf -o example.StateHolder.config_file:/opt/ros/indigo/share/nextage_ros_bridge/conf/nextage.conf -o example.TorqueFilter.config_file:/opt/ros/indigo/share/nextage_ros_bridge/conf/nextage.conf -o example.TorqueController.config_file:/opt/ros/indigo/share/nextage_ros_bridge/conf/nextage.conf -o example.ThermoEstimator.config_file:/opt/ros/indigo/share/nextage_ros_bridge/conf/nextage.conf -o example.ThermoLimiter.config_file:/opt/ros/indigo/share/nextage_ros_bridge/conf/nextage.conf -o example.VirtualForceSensor.config_file:/opt/ros/indigo/share/nextage_ros_bridge/conf/nextage.conf -o example.AbsoluteForceSensor.config_file:/opt/ros/indigo/share/nextage_ros_bridge/conf/nextage.conf -o example.RemoveForceSensorLinkOffset.config_file:/opt/ros/indigo/share/nextage_ros_bridge/conf/nextage.conf -o example.KalmanFilter.config_file:/opt/ros/indigo/share/nextage_ros_bridge/conf/nextage.conf -o example.Stabilizer.config_file:/opt/ros/indigo/share/nextage_ros_bridge/conf/nextage.conf -o example.CollisionDetector.config_file:/opt/ros/indigo/share/nextage_ros_bridge/conf/nextage.conf -o example.SoftErrorLimiter.config_file:/opt/ros/indigo/share/nextage_ros_bridge/conf/nextage.conf -o example.HGcontroller.config_file:/opt/ros/indigo/share/nextage_ros_bridge/conf/nextage.conf -o example.PDcontroller.config_file:/opt/ros/indig [... cut ...]
[ INFO] [1479332583.978231793, 8.209999999]: [HrpsysJointTrajectoryBridge0] @~jointTrajectoryActionObj (rarm
terminate called after throwing an instance of 'CORBA::OBJECT_NOT_EXIST'
terminate called after throwing an instance of 'CORBA::COMM_FAILURE'
[ INFO] [1479332584.003771273, 8.209999999]: [HrpsysJointTrajectoryBridge0] @~jointTrajectoryActionObj (head
[ INFO] [1479332584.023895523, 8.209999999]: [HrpsysJointTrajectoryBridge0] @~jointTrajectoryActionObj (torso
[ INFO] [1479332584.037619931, 8.209999999]: [HrpsysJointTrajectoryBridge0] @~jointTrajectoryActionObj (larm_torso

shzargar, 2016-11-16:
env | grep ROS
ROS_ROOT=/opt/ros/indigo/share/ros
ROS_PACKAGE_PATH=/opt/ros/indigo/share:/opt/ros/indigo/stacks
ROS_MASTER_URI=http://localhost:11311
ROS_HOSTNAME=localhost
ROSLISP_PACKAGE_DIRECTORIES=
ROS_DISTRO=indigo
ROS_ETC_DIR=/opt/ros/indigo/etc/ros

shzargar, 2016-11-16:
cat /etc/hosts
127.0.0.1	localhost
#For accessing QNX for NEXTAGE Open
192.168.128.10	Nextage

# The following lines are desirable for IPv6 capable hosts
::1     ip6-localhost ip6-loopback
fe00::0 ip6-localnet
ff00::0 ip6-mcastprefix
ff02::1 ip6-allnodes
ff02::2 ip6-allrouters

130s, 2016-11-16:
To dump output into a file you can try this (fill in `COMMAND` with your command):
```
unbuffer COMMAND 2>&1 | tee log
```

shzargar, 2016-11-16:
Thanks, here all the logs for the three commands in the first comment
[Logs.zip](https://github.com/tork-a/rtmros_nextage/files/596066/Logs.zip)

534o, 2016-11-17:
thanks, what is `Nxtag` it it the name of your machine? if so , please add that name to /etc/hosts, like

cat /etc/hosts
```
127.0.0.1 localhost
127.0.1.1 Nxtag
#For accessing QNX for NEXTAGE Open
192.168.128.10 Nextage
# The following lines are desirable for IPv6 capable hosts

::1 ip6-localhost ip6-loopback
fe00::0 ip6-localnet
ff00::0 ip6-mcastprefix
ff02::1 ip6-allnodes
ff02::2 ip6-allrouters
```

shzargar, 2016-11-17:
Thanks, Now my simulation is working. A month ago, with your help I set my eth0 connection and changed my hosts to add Nextage.  I am just interested to learn more about the line 127.0.1.1 Nxtag. This is to know for a new installation or new user, how should I change it.

534o, 2016-11-18:
> A month ago, with your help I set my eth0 connection and changed my hosts to add Nextage.

sorry for that, please point out where did we ask you to change that, 127.0.1.1 is very basic information for ubuntu system so I think it is very bad idea to remove that -> http://askubuntu.com/questions/754213/what-is-difference-between-localhost-address-127-0-0-1-and-127-0-1-1

shzargar, 2016-11-18:
Thanks for your explanation, however, I don’t think that 127.0.1.1 was there from the beginning. That is a fresh Ubuntu installation on a laptop. I had only the instruction to add  these two below lines:
#For accessing QNX for NEXTAGE Open
192.168.128.10 Nextage
Except the above, I hadn't changed anything in the original hosts file, up until now that I add 127.0.1.1.

130s, 2017-01-17:
I assume this issue is addressed. Please feel free to re-open with updated info if the issue persists.

closed 2017-01-17 by 130s