tork-a/rtmros_nextage #273: robot.setJointAnglesSequence

shzargar, 2016-11-09:
Using setTargetPose, I move the robot to some positions, for each I saved Joint angles in a variable. Then I try to do the same movement from aa4 to aa1, aa2, aa3 and return back to aa4 by using robot.setJointAnglesSequence([aa1,aa2,aa3,aa4],[5,5,5,5])

I got the error and after this error my aa1 to aa4 values have been changed I think by the above function.

The first run I used the degree values (first part), then radian values (second part), in both cases no movement, as well, the input values were changed. 

In [89]: robot.goInitial()
[hrpsys.py] , JntAnglesOfGr=torso, INITPOSE[i]=[0], tm=7, wait=True
[hrpsys.py] , JntAnglesOfGr=head, INITPOSE[i]=[0, 0], tm=7, wait=True
[hrpsys.py] , JntAnglesOfGr=rarm, INITPOSE[i]=[-0.6, 0, -100, 15.2, 9.4, 3.2], tm=7, wait=True
[hrpsys.py] , JntAnglesOfGr=larm, INITPOSE[i]=[0.6, 0, -100, -15.2, 9.4, -3.2], tm=7, wait=True
Out[89]: True

In [90]: pos = robot.getCurrentPosition('RARM_JOINT5')

In [91]: pos
Out[91]: [0.32557647381110855, -0.1823598289665732, 1.0376258723172136]

In [92]: pos[0]=0.50

In [93]: robot.setTargetPose('rarm', pos, rpy, tm)
('rarm', None, [0.5, -0.1823598289665732, 1.0376258723172136], (3.0734370292070814, -1.569022617731364, -3.0732470442070254), 3)
Out[93]: True

In [94]: aa1=robot.getJointAngles()

In [95]: aa1
Out[95]: 
[0.0,
 0.0,
 0.0,
 -1.2922019064509318,
 -47.944892425156176,
 -32.29115006976289,
 15.21701386396911,
 -9.863880593864172,
 -1.2921947219497114,
 0.6,
 0.0,
 -100.0,
 -15.2,
 9.4,
 -3.2]

In [96]: pos[2]=1.20

In [97]: robot.setTargetPose('rarm', pos, rpy, tm)
('rarm', None, [0.5, -0.1823598289665732, 1.2], (3.0734370292070814, -1.569022617731364, -3.0732470442070254), 3)
Out[97]: True

In [98]: aa2=robot.getJointAngles()

In [99]: aa2
Out[99]: 
[0.0,
 0.0,
 0.0,
 5.523884427323548,
 -47.05571075509552,
 -72.15215182107883,
 17.210173086207686,
 29.443551214244565,
 2.977750528155962,
 0.6,
 0.0,
 -100.0,
 -15.2,
 9.4,
 -3.2]

[... 332 lines cut ...]
 0.002041886269024358,
 0.00018267779504207312,
 0.0,
 -0.030446299173678863,
 -0.0046278374743991876,
 0.0028619521223258127,
 -0.0009742815735577237]

In [136]: aa4
Out[136]: 
[0.0,
 0.0,
 0.0,
 -0.00018267779504206808,
 3.3098282987821374e-19,
 -0.030446299173678863,
 0.004627837474399185,
 0.002861952122325795,
 0.0009742815735577237,
 0.00018267779504207312,
 0.0,
 -0.030446299173678863,
 -0.0046278374743991876,
 0.0028619521223258127,
 -0.0009742815735577237]

shzargar, 2016-11-09:
I used the below command after, I am not sure if this will upload automatically the log file to github. Please let me know if I have to upload it manually.

curl -sL https://raw.githubusercontent.com/start-jsk/rtmros_common/master/hrpsys_tools/scripts/rtmroswtf.sh | bash -

130s, 2016-11-09:
I assume you see this issue while you're working with the real robot, and if you run the same code on the simulation you might not see it.
[`setJointAnglesSequence`, `setJointAnglesSequenceOfGroup` were added since hrpsys 315.5.0](https://github.com/fkanehiro/hrpsys-base/commit/45b00db659b143c18fcd3bdccfd5a4f42047cef5#diff-6c33bc3d5bb515b220c1b7d93821cd9dR971) while the version of your robot's software (the one running in the Controller Box) is less than that.

Feel free for the further comments.

closed 2016-11-09 by 130s

shzargar, 2016-11-09:
Yes, that is on real robot. Is it possible to upgrade hrpsys on the robot and how?
Is there any code for moving all joints together for previous version.

130s, 2016-11-16:
>  Is it possible to upgrade hrpsys on the robot and how?

Please see [here](http://wiki.ros.org/rtmros_nextage/Tutorials/Install%20NEXTAGE%20OPEN%20software%20on%20your%20machine#Installation_and_usage_on_QNX). No info is disclosed. With [paid support service by TORK](http://opensource-robotics.tokyo.jp/?page_id=681&lang=en) you may upgrade per request.

>  Is there any code for moving all joints together for previous version. 

Is calling [setJointAngles](http://fkanehiro.github.io/hrpsys-base/df/d98/classpython_1_1hrpsys__config_1_1HrpsysConfigurator.html#a8f6c980d05cc00f41d9667036ebd436f) multiple times not an option for you?

k-okada, 2016-11-20:
you can use playPattern instead of setJointAnglesSequence , since for ROS users they use playPattern until very recently https://github.com/start-jsk/rtmros_common/pull/765/files
Here is some explanation of how to use this -> https://github.com/tork-a/rtmros_nextage/issues/275#issuecomment-261143558