tork-a/rtmros_nextage #274: nothing happens and no error with robot.HandClose('lhand',0),  robot.HandOpen('lhand',100)

130s, 2016-11-14:
@shzargar Sorry for the inconvenience. Those methods are primarilly for Hironx, a NEXTAGE Open's ancestor robot that comes with servo-based hands.

With a factory setting NEXTAGE Open is not equipped with the hands that are controllable by those commands. Instead you can use DIO-based commands that are available via `NextageClient` class. See [the api](http://docs.ros.org/indigo/api/nextage_ros_bridge/html/classnextage__ros__bridge_1_1nextage__client_1_1NextageClient.html). You can call them from `ipython` interpreter, same as other methods in the tutorial. See [this tutorial](http://wiki.ros.org/rtmros_nextage/Tutorials/Using%20digital%20IO%20%28NXO%20only%29) for more info.

shzargar, 2016-11-15:
My observation:

robot._hands.handtool_r_eject() this ejects the right hand.
robot._hands.handtool_r_attach() this attaches the right hand.
robot._hands.handtool_l_eject() this closes the right gripper when it is supposed to eject the left hand.
robot._hands.handtool_l_attach() this opens the right gripper when it is supposed to attach the left hand.

robot._hands.gripper_l_open()  this ejects the left hand and (no gripper opens).
robot._hands.gripper_l_close() this attaches the left hand (no gripper closes).
robot._hands.airhand_r_drawin() this close the left hand gripper.

and I couldn't yet find a command to open the left gripper (except robot._hands.init_dio() which initialize everything)

Is that a table to redefine the IO?

130s, 2016-11-15:
Ah, then your robot's DIO is probably older than the change https://github.com/tork-a/rtmros_nextage/pull/88 

How about for example `robot.gripper_l_open`? If this works as the method name suggests, then please use the [methods listed directly under `NextageClient` class](http://docs.ros.org/indigo/api/nextage_ros_bridge/html/classnextage__ros__bridge_1_1nextage__client_1_1NextageClient.html), instead of bypassing `_hands` object.

In the meantime I'll update the tutorial to correctly cover the older api.

shzargar, 2016-11-15:
robot.gripper_l_open is not working. Please see below with and without parenthesis.

In [98]: robot.gripper_l_open()
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-98-f7236d382e3f> in <module>()
----> 1 robot.gripper_l_open()

/opt/ros/indigo/lib/python2.7/dist-packages/nextage_ros_bridge/nextage_client.pyc in gripper_l_open(self)
    238                 self._hands.gripper_r_command.GRIPPER_OPEN)
    239         except AttributeError:
--> 240             self._print_msg_deprecated_dio_methods(sys._getframe().f_code.co_name)
    241 
    242     def gripper_r_open(self):

/opt/ros/indigo/lib/python2.7/dist-packages/nextage_ros_bridge/nextage_client.pyc in _print_msg_deprecated_dio_methods(self, name_method)
    131         '''
    132         rospy.logerr("You're likely to have called a deprecated method." +
--> 133                      " Use self._hands.%S instead", name_method)
    134 
    135     def handlight_r(self, is_on=True):

/usr/lib/python2.7/logging/__init__.pyc in error(self, msg, *args, **kwargs)
   1176         """
   1177         if self.isEnabledFor(ERROR):
-> 1178             self._log(ERROR, msg, args, **kwargs)
   1179 
   1180     def exception(self, msg, *args, **kwargs):

/usr/lib/python2.7/logging/__init__.pyc in _log(self, level, msg, args, exc_info, extra)
   1269                 exc_info = sys.exc_info()
   1270         record = self.makeRecord(self.name, level, fn, lno, msg, args, exc_info, func, extra)
-> 1271         self.handle(record)
   1272 
   1273     def handle(self, record):

/usr/lib/python2.7/logging/__init__.pyc in handle(self, record)
   1279         """
   1280         if (not self.disabled) and self.filter(record):
-> 1281             self.callHandlers(record)
   1282 
   1283     def addHandler(self, hdlr):

/usr/lib/python2.7/logging/__init__.pyc in callHandlers(self, record)
   1319                 found = found + 1
   1320                 if record.levelno >= hdlr.level:
-> 1321                     hdlr.handle(record)
   1322             if not c.propagate:
   1323                 c = None    #break out

/usr/lib/python2.7/logging/__init__.pyc in handle(self, record)
    747             self.acquire()
    748             try:
--> 749                 self.emit(record)
    750             finally:
    751                 self.release()

/opt/ros/indigo/lib/python2.7/dist-packages/rosgraph/roslogging.pyc in emit(self, record)
    199             if self._get_time is not None and not self._is_wallclock():
    200                 msg += ' [%f]' % self._get_time()
--> 201             msg += ' %s\n' % record.getMessage()
    202         if record.levelno < logging.WARNING:
    203             self._write(sys.stdout, msg, color)

/usr/lib/python2.7/logging/__init__.pyc in getMessage(self)
    326                     msg = self.msg      #Defer encoding till later
    327         if self.args:
--> 328             msg = msg % self.args
    329         return msg
    330 

ValueError: unsupported format character 'S' (0x53) at index 67

In [99]: robot.gripper_l_open
Out[99]: <bound method NextageClient.gripper_l_open of <nextage_ros_bridge.nextage_client.NextageClient object at 0x7fdc274e6650>>

130s, 2016-11-16:
Hm, then I'm not sure what's going on. 

If possible, 
> robot._hands.handtool_r_eject() this ejects the right hand.
> robot._hands.handtool_r_attach() this attaches the right hand.
> robot._hands.handtool_l_eject() this closes the right gripper when it is [...]

after each of the command (or only a few of them) above, can you also run `robot.readDigitalOutput()` and share the result?

Btw the particular error in your post https://github.com/tork-a/rtmros_nextage/issues/274#issuecomment-260801272 seems to be a bug that I opened a fix PR #276 for. Although this bug has nothing to do with the DIO functionality itself, fixing it will print the DIO's in/output, which will also be helpful to figure out what port is used for the particular commands. 

Also **if possible**, please update your rtmros_nextage software and try `robot.gripper_l_open/close` commands again. To update, 

- if you're using rtmros_nextage software from source, just update it by:

```
roscd rtmros_nextage
git fetch origin && git rebase origin/indigo-devel
```

- if you're using binary (installed by apt-get):
```
wget https://raw.githubusercontent.com/tork-a/88_tools/master/hotfixer/hot_fix.sh bash ./hot_fix.sh tork-a/rtmros_nextage 276
```

shzargar, 2016-11-16:
In [5]: robot._hands.handtool_r_eject()
dout, mask:
[0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0],
[0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0, 0]
[1, 2, 3, 4, 5, 6, 7, 8, 9, 0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 0, 1, 2]
Out[5]: False

In [6]: robot._hands.handtool_r_attach()
dout, mask:
[0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0],
[0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
[1, 2, 3, 4, 5, 6, 7, 8, 9, 0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 0, 1, 2]
Out[6]: False

In [7]: robot._hands.handtool_l_eject()
dout, mask:
[0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0],
[0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0]
[1, 2, 3, 4, 5, 6, 7, 8, 9, 0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 0, 1, 2]
Out[7]: False

In [8]: robot._hands.handtool_l_attach()
dout, mask:
[0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0],
[0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
[1, 2, 3, 4, 5, 6, 7, 8, 9, 0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 0, 1, 2]
Out[8]: False

In [9]: robot.readDigitalOutput()
Out[9]: 
[1,
 0,
 0,
 0,
 0,
 0,
 0,
 0,
 0,
 0,
 0,
 0,
 0,
 0,
 0,
 0,
 0,
 0,
 0,
 0,
 0,
 0,
 0,
 0,
 0,
 0,
 0,
 0,
 0,
 0,
 0,
 0]

In [10]: robot._hands.handtool_l_eject()
dout, mask:
[0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0],
[0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0]
[1, 2, 3, 4, 5, 6, 7, 8, 9, 0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 0, 1, 2]
Out[10]: False

In [11]: robot.readDigitalOutput()
Out[11]: 
[1,
 0,
 0,
 0,
 0,
 0,
 0,
 0,
 0,
 0,
 0,
 0,
 0,
 0,
 0,
 0,
 0,
 0,
 0,
 0,
 0,
 0,
 0,
 0,
 0,
 0,
 0,
 0,
 0,
 0,
 0,
 0]

shzargar, 2016-11-16:
After below (look the output, since I am not sure if the update is done properly), robot.gripper_l_open/close are not working as before.

wget https://raw.githubusercontent.com/tork-a/88_tools/master/hotfixer/hot_fix.sh bash ./hot_fix.sh tork-a/rtmros_nextage 276
--2016-11-16 15:57:08--  https://raw.githubusercontent.com/tork-a/88_tools/master/hotfixer/hot_fix.sh
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 151.101.20.133
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|151.101.20.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1927 (1.9K) [text/plain]
Saving to: ‘hot_fix.sh.1’

100%[======================================>] 1,927       --.-K/s   in 0s      

2016-11-16 15:57:08 (82.9 MB/s) - ‘hot_fix.sh.1’ saved [1927/1927]

--2016-11-16 15:57:08--  http://bash/
Resolving bash (bash)... failed: Name or service not known.
wget: unable to resolve host address ‘bash’
--2016-11-16 15:57:08--  http://./hot_fix.sh
Resolving . (.)... failed: No address associated with hostname.
wget: unable to resolve host address ‘.’
--2016-11-16 15:57:08--  http://tork-a/rtmros_nextage
Resolving tork-a (tork-a)... failed: Name or service not known.
wget: unable to resolve host address ‘tork-a’
--2016-11-16 15:57:08--  http://276/
Resolving 276 (276)... 0.0.1.20
Connecting to 276 (276)|0.0.1.20|:80... failed: Invalid argument.
FINISHED --2016-11-16 15:57:08--
Total wall clock time: 0.4s
Downloaded: 1 files, 1.9K in 0s (82.9 MB/s)

k-okada, 2016-11-17:
> hotfixer/hot_fix.sh bash ./hot_fix.sh tork-a/rtmros_nextage 276
> com/tork-a/88_tools/master/hotfixer/hot_fix.sh

first, download script as
- wget https://raw.githubusercontent.com/tork-a/88_tools/master/
hotfixer/hot_fix.sh
Then run that sciprt
- bash ./hot_fix.sh tork-a/rtmros_nextage 276

Best

130s, 2016-12-06:
Ok, I'm getting to the core of this issue and will open a fix PR hopefully soon.

Just as a memo in case I won't be able to open a PR, here's my thought (includes internal design. Don't bother try understanding it unless you feel like contributing patches):

- There was DIO design change by manufacturer sometime around 2014 July. Robots shipped BEFORE and AFTER then come with different DIO assignment.
- That design change was I thought handled in https://github.com/tork-a/rtmros_nextage/pull/88, which I just figured might have fixed the issue ONLY for "AFTER" robots users.
- "BEFORE" users are supposed to instantiate older version of hand class, but code doesn't do that ([around here](https://github.com/tork-a/rtmros_nextage/blob/740eaecb4cefb8846abd82491ee49b75145a0a4e/nextage_ros_bridge/src/nextage_ros_bridge/nextage_client.py#L125)). Instead, they are using the new hand class that "AFTER" users use.
  - There's a way to use [Iros13Hands](https://github.com/tork-a/rtmros_nextage/blob/740eaecb4cefb8846abd82491ee49b75145a0a4e/nextage_ros_bridge/src/nextage_ros_bridge/nextage_client.py#L123), which is a very specific configuration with air hand and gripper hand so normal users don't want to use that.

TODOs:
- Implement generic hand class for version BEFORE 05.
- Add option to choose that version of hand class.

130s, 2017-01-10:
I took a look into this issue thinking of adding a patch, then I found confident myself that there's no need for the change.
For the robot with older DIO spec like I mentioned in https://github.com/tork-a/rtmros_nextage/issues/274#issuecomment-265272979, you can just call `[set_hand_version](http://docs.ros.org/indigo/api/nextage_ros_bridge/html/classnextage__ros__bridge_1_1nextage__client_1_1NextageClient.html#a670568692e46cfc3f10633ad962b8616)('0.4.2')` method (0.4.2 indicates the older spec).

```
$ ipython -i `rospack find nextage_ros_bridge`/script/nextage.py
:
In [2]: robot.set_hand_version(version='0.4.2')                                                           
```

Calling this every time is a bit cumbersome and could be prone to forgetting, so I'll still make a patch. But the solution above should be already available for all robot users who use rtmros_nextage software newer than 0.4.

130s, 2017-01-10:
took a look into this issue thinking of adding a patch, then I found confident myself that there's no need for the change.
For the robot with older DIO spec like I mentioned in https://github.com/tork-a/rtmros_nextage/issues/274#issuecomment-265272979, you can just call [set_hand_version](http://docs.ros.org/indigo/api/nextage_ros_bridge/html/classnextage__ros__bridge_1_1nextage__client_1_1NextageClient.html#a670568692e46cfc3f10633ad962b8616)('0.4.2') method (0.4.2 indicates the older spec).

```
$ ipython -i `rospack find nextage_ros_bridge`/script/nextage.py
:
In [2]: robot.set_hand_version(version='0.4.2')                                                           
```

Calling this every time is a bit cumbersome and could be prone to forgetting, so I'll still make a patch. But the solution above should be already available for all robot users who use rtmros_nextage software newer than 0.4.

130s, 2017-01-10:
> - [ ] Make a convenient patch https://github.com/tork-a/rtmros_nextage/issues/274#issuecomment-271603322

--> https://github.com/tork-a/rtmros_nextage/pull/280

> - [ ] Document the patch above on ros wiki tutorial

[done on wiki](http://wiki.ros.org/rtmros_nextage/Tutorials/Using%20digital%20IO%20%28NXO%20only%29#HiroNXO_shipped_before_August_2014_need_to_specify_dio_spec_version)

closed 2017-01-10 by 130s