tork-a/rtmros_nextage #275: How robot.playPattern is working?

shzargar, 2016-11-15:
First, is this command is working for QNX hrpsys version 315.1.8 and if yes what is zmp list? if it is not working what is the option for moving all joints (no matter in which group ther are located) together?

robot.playPattern?
Type:       instancemethod
String Form:<bound method NextageClient.playPattern of <nextage_ros_bridge.nextage_client.NextageClient object at 0x7fdc274e6650>>
File:       /opt/ros/indigo/lib/python2.7/dist-packages/hrpsys/hrpsys_config.py
Definition: robot.playPattern(self, jointangles, rpy, zmp, tm)
Docstring:
!@brief
Play motion pattern using a given trajectory that is represented by 
a list of joint angles, rpy, zmp and time.

@param jointangles list of list of float: 
                   The whole list represents a trajectory. Each element
                   of the 1st degree in the list consists of the joint
                   angles.
@param rpy list of float: Orientation in rpy.
@param zmp list of float: TODO: description
@param tm float: Time to complete the task.
@return bool:

130s, 2016-11-16:
Delegated to the upstream https://github.com/fkanehiro/hrpsys-base/issues/1072

closed 2016-11-16 by 130s

k-okada, 2016-11-17:
as answered in https://github.com/fkanehiro/hrpsys-base/issues/1072 you can use `[]` for that parameter, pleaes not that `tm` is not float, that should be list of float (
```https://github.com/fkanehiro/hrpsys-base/blob/master/idl/SequencePlayerService.idl#L110) and angle value should be deg, not rad in setJointAngles.
$ hrpsyspy
configuration ORB with localhost:15005
[hrpsys.py] wait for RTCmanager : p40-yoga
[hrpsys.py] wait for HiroNX(Robot)0 : <hrpsys.rtm.RTcomponent instance at 0x7fdf1222ca70> ( timeout 0 < 10)
[hrpsys.py] findComps -> RobotHardware : <hrpsys.rtm.RTcomponent instance at 0x7fdf1222ca70> isActive? = True 
[hrpsys.py] simulation_mode : True
Starting hrpsys python interface for HiroNX(Robot)0
 Installed rtc are ...  seq   sh   fk   rmfo   ic   el   log  
 Use 'hcf' instance to access to the hrpsys controller
Python 2.7.6 (default, Jun 22 2015, 17:58:13) 
Type "copyright", "credits" or "license" for more information.

IPython 1.2.1 -- An enhanced Interactive Python.
?         -> Introduction and overview of IPython's features.
%quickref -> Quick reference.
help      -> Python's own help system.
object?   -> Details about 'object', use 'object??' for extra details.

In [1]:hcf.playPattern([[0, 1,  0,  0, 0.3, -1.9, 0, 0, 0,  0, 0.3, -1.9, 0, 0]], [], [], [1.0])
```
but in nextage case, you should use playPatternOfGroup, once someone use `ofGroup` seqneucer, that would overwrite fullbody seqeuncer output 
```
In [2]: hcf.playPatternOfGroup('head', [[-1, 0]], [1.0])
```
also note that you can use `/head_controller/follow_joint_trajectory_action/goal` ROS interface, which basically just send trajectory to the robot using playPattern or similar one.