tork-a/rtmros_nextage #277: IKFAST plugin for left arm of nextage robot seems to have an offset

ysolana, 2016-12-02:
Hi all,
I need to compute at lot of IK requests (billion) for the nextage robot in order to generate a reachability map of the workspace for the arms (see attached picture). The computation with IKFAST fast is very fast (~1 day) compared to KDL (more than a week) to generate the data. All works great with the right arm plugin, however for the left arm there is a problem.

By displaying the states of the left arm after several IK request, I can see there is an offset of the end effector between the solution generated and the given pose of the end effector.
This can be also seen in RViz, when the interactive marker (from the motion planning display) is used.

Picture: An example of the data generated for the right arm using IKFAST:
[image]

Video1: Normal behaviour for the left arm using KDL plugin:
[KDL_plugin_with_RViz.mp4.zip](https://github.com/tork-a/rtmros_nextage/files/627375/KDL_plugin_with_RViz.mp4.zip)

Video2: Problem with using IKFAST plugin with the left arm:
[IKFAST_plugin_with_RViz.mp4.zip](https://github.com/tork-a/rtmros_nextage/files/627378/IKFAST_plugin_with_RViz.mp4.zip)

To solve this problem: 
_ I tried to install openrave in order to generate the IKFAST plugin but I was not able to make it. The install on my ubuntu distribution (trusty, 14.04) is not an easy task.
_ I tried to track the error about the offset thinking it could be a minor problem with transformation, but the cpp file for the IKFAST left arm plugin is not easy to read (there is a lot of equations).

So, Is someone can confirm me is he is encounter the same problem?
And, if the problem is validated, is it possible to fix this or generate the IKFAST plugin again?

Thank you in advance,
Regards,
Yoann.

k-okada, 2016-12-03:
nice work, can you show us both arguments and the output of the  ikfast.py program when you created the plugin software?

k-okada, 2016-12-03:
fyi:

You do not have to compile openrave from source, just to install ros-indigo-openrave and trun

```
 PYTHONPATH=$PYTHONPATH:`openrave-config --python-dir` python `openrave-config --python-dir`/openravepy/_openravepy_/ikfast.py
```

ysolana, 2016-12-06:
Hi k-okada,
Thank you for the answer.

This is the command, I am using to generate the IK solver, based on the moveit tutorial 
[http://docs.ros.org/indigo/api/moveit_tutorials/html/doc/ikfast_tutorial.html](url)
$python `openrave-config --python-dir`/openravepy/_openravepy_/ikfast.py --robot=NextageOpen.dae --iktype=transform6d --baselink=3 --eelink=11 --savefile=ikfast61_NextageOpen_left_arm

The indexes for 'baselink' and 'eelink' are chosen wrt the result of the command (see file openrave_cmds.txt):
$ openrave-robot.py NextageOpen.dae --info links
[openrave_cmds.txt](https://github.com/tork-a/rtmros_nextage/files/633358/openrave_cmds.txt)
In addition the command
$ openrave NextageOpen.dae 
shows the robot without the base and the WAIST, but I think it is OK because i only work on the left arm.

And this is the output file for ikfast.py:
[ikfast.py_output.txt](https://github.com/tork-a/rtmros_nextage/files/633329/ikfast.py_output.txt)
In this file, I have seen these warnings:
_ failed to create a name for fcl_ and ode
_ some links have a transparency equal to 1
_ more unknowns than equations
_ equations are not linearly independent
_ equation too complex to simplify for rot norm
_ not all equations zero, so [j5, j6] vars are not collinear
_ cannot handle linear solving for more than 4 equations
_ determinant complexity is too big 1257
However at the end solutions are found, so I am not sure if these warnings are problems.

Then, I continue to follow the moveit tutorial to generate the package and select it in kinematics.yaml file, but there is still this offset when i am testing the new ik plugin in RViz.
Do you an any idea about this offset?

Do you encounter the same problem by using the ik plugin for the left arm from this repository?

Thank you in advance,
Regards,
Yoann.

130s, 2017-01-12:
By using [nextage_ik_plugin](http://wiki.ros.org/nextage_ik_plugin) package I can confirm that on simulation the Interactive Marker at the left arm often not only offsets, but also doesn't react to my mouse operation. 
And unlike @ysolana who seems to have customized including grippers, I'm using everything in the default setting.

Output from moveit launch. I can't find anything peculiar for this issue.

```
$ roslaunch nextage_moveit_config moveit_planning_execution.launch kinematics_conf:=`rospack find nextage_moveit_config`/config/kinematics_ikfast.yaml
... logging to /home/rosnoodle/.ros/log/7bc8045a-d8bf-11e6-b323-80fa5b08d1d8/roslaunch-tork-kudu1-2092.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://tork-kudu1:46697/

SUMMARY
========

PARAMETERS
 * /move_group/allow_trajectory_execution: True
 * /move_group/botharms/default_planner_config: RRTConnectkConfig...
 * /move_group/botharms/longest_valid_segment_fraction: 0.05
 * /move_group/botharms/planner_configs: ['SBLkConfigDefau...
 * /move_group/botharms/projection_evaluator: joints(LARM_JOINT...
 * /move_group/capabilities: move_group/MoveGr...
 * /move_group/controller_list: [{'default': True...
 * /move_group/controller_manager_name: pr2_controller_ma...
 * /move_group/controller_manager_ns: pr2_controller_ma...
 * /move_group/head/default_planner_config: RRTConnectkConfig...
 * /move_group/head/kinematics_solver: kdl_kinematics_pl...
 * /move_group/head/kinematics_solver_attempts: 3
 * /move_group/head/kinematics_solver_search_resolution: 0.005
 * /move_group/head/kinematics_solver_timeout: 0.005
 * /move_group/head/longest_valid_segment_fraction: 0.05
 * /move_group/head/planner_configs: ['SBLkConfigDefau...
 * /move_group/head/projection_evaluator: joints(HEAD_JOINT...
 * /move_group/jiggle_fraction: 0.05
 * /move_group/left_arm/default_planner_config: RRTConnectkConfig...
 * /move_group/left_arm/kinematics_solver: nextage_left_arm_...
 * /move_group/left_arm/kinematics_solver_attempts: 3
 * /move_group/left_arm/kinematics_solver_search_resolution: 0.005
 * /move_group/left_arm/kinematics_solver_timeout: 0.005
 * /move_group/left_arm/longest_valid_segment_fraction: 0.05
 * /move_group/left_arm/planner_configs: ['SBLkConfigDefau...
 * /move_group/left_arm/projection_evaluator: joints(LARM_JOINT...
 * /move_group/max_safe_path_cost: 1
 * /move_group/moveit_controller_manager: pr2_moveit_contro...
 * /move_group/moveit_manage_controllers: True
 * /move_group/planner_configs/BKPIECEkConfigDefault/type: geometric::BKPIECE
 * /move_group/planner_configs/ESTkConfigDefault/type: geometric::EST
 * /move_group/planner_configs/KPIECEkConfigDefault/type: geometric::KPIECE
 * /move_group/planner_configs/LBKPIECEkConfigDefault/type: geometric::LBKPIECE
 * /move_group/planner_configs/PRMkConfigDefault/type: geometric::PRM
 * /move_group/planner_configs/PRMstarkConfigDefault/type: geometric::PRMstar
 * /move_group/planner_configs/RRTConnectkConfigDefault/type: geometric::RRTCon...
 * /move_group/planner_configs/RRTkConfigDefault/type: geometric::RRT
 * /move_group/planner_configs/RRTstarkConfigDefault/type: geometric::RRTstar
 * /move_group/planner_configs/SBLkConfigDefault/type: geometric::SBL
 * /move_group/planner_configs/TRRTkConfigDefault/type: geometric::TRRT
 * /move_group/planning_plugin: ompl_interface/OM...
 * /move_group/planning_scene_monitor/publish_geometry_updates: True
 * /move_group/planning_scene_monitor/publish_planning_scene: True
 * /move_group/planning_scene_monitor/publish_state_updates: True
 * /move_group/planning_scene_monitor/publish_transforms_updates: True
 * /move_group/request_adapters: default_planner_r...
 * /move_group/right_arm/default_planner_config: RRTConnectkConfig...
 * /move_group/right_arm/kinematics_solver: nextage_right_arm...
 * /move_group/right_arm/kinematics_solver_attempts: 3
 * /move_group/right_arm/kinematics_solver_search_resolution: 0.005
 * /move_group/right_arm/kinematics_solver_timeout: 0.005
 * /move_group/right_arm/longest_valid_segment_fraction: 0.05
[... 1314 lines cut ...]
[hrpsys_ros_diagnostics-8] killing on exit
[hrpsys_state_publisher-7] killing on exit
[HrpsysJointTrajectoryBridge-6] killing on exit
[HrpsysSeqStateROSBridge-5] killing on exit
[hrpsys-3] killing on exit
[modelloader-2] killing on exit
[ INFO] [1484222845.461210692, 83.754999999]: [HrpsysJointTrajectoryBridge0] @~jointTrajectoryActionObj (larm
[ INFO] [1484222845.461675528, 83.754999999]: [HrpsysSeqStateROSBridge] @onFinalize : HrpsysSeqStateROSBridge0
terminate called after throwing an instance of 'CORBA::COMM_FAILURE'
terminate called after throwing an instance of 'CORBA::OBJECT_NOT_EXIST'
[ INFO] [1484222845.478412902, 83.754999999]: [HrpsysJointTrajectoryBridge0] @~jointTrajectoryActionObj (rarm
[ INFO] [1484222845.488137149, 83.754999999]: [HrpsysJointTrajectoryBridge0] @~jointTrajectoryActionObj (head
[ INFO] [1484222845.496403285, 83.754999999]: [HrpsysJointTrajectoryBridge0] @~jointTrajectoryActionObj (torso
[ INFO] [1484222845.508302406, 83.754999999]: [HrpsysJointTrajectoryBridge0] @~jointTrajectoryActionObj (larm_torso
[ INFO] [1484222845.518622528, 83.754999999]: [HrpsysJointTrajectoryBridge0] @~jointTrajectoryActionObj (rarm_torso
terminate called after throwing an instance of 'CORBA::COMM_FAILURE'
terminate called after throwing an instance of 'class_loader::LibraryUnloadException'
  what():  Attempt to unload library that class_loader is unaware of.
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
[rtmlaunch] terminate omniNames at port 15005
```

130s, 2017-01-12:
In https://github.com/tork-a/rtmros_nextage/pull/285 I've added a config file for using [trac_ik](http://wiki.ros.org/trac_ik). Haven't tested much at all but at least the issue raised here doesn't occur.

@ysolana it may or may not be fulfilling for your needs. We'd appreciate if you could benchmark it (and even greater if you could repot back here so that we report to trac_ik guy the result like [this table](https://bitbucket.org/traclabs/trac_ik/src/a5008432774126586063a74661b86eb1bcefdc9c/README.md?at=master&fileviewer=file-view-default)).

ysolana, 2017-01-12:
Hi 130s,
Thank you for the answer. 
So it is confirmed there is a problem with the IKFast left arm plugin. For now, I don't have time to investigate on this.

Following the table, trac_ik is faster with an order of magnitude between 2-4 compared to KDL. For me it will not be enough, for the capability map and the dual arm planning it is required to use an analytical solver. If I remember well the order of magnitude in time to solve an IK request between IKFast and KDL for the right arm was closed to 100.

I will take a look next week to use track_ik and make some benchmark. I will keep you informed.

Regards,
Yoann.

k-okada, 2017-01-18:
@ysolana 

Here is some updates, if you change the order of `group` definition of NextageOpens.srdf, then we can have correct behavior of left arm, but have similar offset on right arm, so the problem is not ikfast generated cpp code, but some initialization between moveit and ikfast plugins.

```
$ cat 277.diff 
diff --git a/nextage_moveit_config/config/NextageOpen.srdf b/nextage_moveit_config/config/NextageOpen.srdf
index c2f47ad..0c40e72 100644
--- a/nextage_moveit_config/config/NextageOpen.srdf
+++ b/nextage_moveit_config/config/NextageOpen.srdf
@@ -9,24 +9,6 @@
     <!--JOINTS: When a joint is specified, the child link of that joint (which will always exist) is automatically included-->
     <!--CHAINS: When a chain is specified, all the links along the chain (including endpoints) are included in the group. Additionally, all the joints that are parents to included links are also included. This means that joints along the chain and the parent joint of the base link are included in the group-->
     <!--SUBGROUPS: Groups can also be formed by referencing to already defined group names-->
-    <group name="right_arm">
-        <link name="RARM_JOINT0_Link" />
-        <link name="RARM_JOINT1_Link" />
-        <link name="RARM_JOINT2_Link" />
-        <link name="RARM_JOINT3_Link" />
-        <link name="RARM_JOINT4_Link" />
-        <link name="RARM_JOINT5_Link" />
-        <joint name="RARM_JOINT0" />
-        <joint name="RARM_JOINT1" />
-        <joint name="RARM_JOINT2" />
-        <joint name="RARM_JOINT3" />
-        <joint name="RARM_JOINT4" />
-        <joint name="RARM_JOINT5" />
-    </group>
-    <group name="right_hand">
-        <link name="RARM_JOINT5_Link" />
-        <joint name="RARM_JOINT5" />
-    </group>
     <group name="left_arm">
         <link name="LARM_JOINT0_Link" />
         <link name="LARM_JOINT1_Link" />
@@ -41,10 +23,28 @@
         <joint name="LARM_JOINT4" />
         <joint name="LARM_JOINT5" />
     </group>
+    <group name="right_arm">
+        <link name="RARM_JOINT0_Link" />
+        <link name="RARM_JOINT1_Link" />
+        <link name="RARM_JOINT2_Link" />
+        <link name="RARM_JOINT3_Link" />
+        <link name="RARM_JOINT4_Link" />
+        <link name="RARM_JOINT5_Link" />
+        <joint name="RARM_JOINT0" />
+        <joint name="RARM_JOINT1" />
+        <joint name="RARM_JOINT2" />
+        <joint name="RARM_JOINT3" />
+        <joint name="RARM_JOINT4" />
+        <joint name="RARM_JOINT5" />
+    </group>
     <group name="left_hand">
         <link name="LARM_JOINT5_Link" />
         <joint name="LARM_JOINT5" />
     </group>
+    <group name="right_hand">
+        <link name="RARM_JOINT5_Link" />
+        <joint name="RARM_JOINT5" />
+    </group>
     <group name="botharms">
         <group name="right_arm" />
         <group name="left_arm" />
```

k-okada, 2017-01-18:
https://github.com/tork-a/rtmros_nextage/pull/286 fixed the problem, I only confirmed in rviz interface. please check that

ysolana, 2017-01-19:
Hi,

@k-okada 
Thank you the problem is solved. I have tested #286 in RViz and with the capability map. 

@130s 
I have put in my TODO list the benchmark of the KDL, TRAC-IK, IKFast plugins. I will communicate the results soon.

Thank you again and have a good day,
Yoann.

PR #286 (2017-01-19): split namespace between left and right arm
fix #277, may be shareing global vairable caused the problem
closed 2017-01-19 by 130s