tork-a/rtmros_nextage #289: Test on real robot the conf/nosim.xml files

130s, 2017-01-24:
`nextage_ros_bridge/conf/*.xml` files are used in simulation by OpenHRP3/hrpsys if I'm not mistaken. Currently, NEXTAGE simulation uses these files from the upstream, hironx, although in NEXTAGE package these files are maintained [here](https://github.com/tork-a/rtmros_nextage/tree/indigo-devel/nextage_ros_bridge/conf). This setting introduces an inconvenience as in https://github.com/tork-a/rtmros_nextage/pull/279, but in order to use these config files in NEXTAGE package, testing on the real robot is essential as noted in https://github.com/tork-a/rtmros_nextage/pull/279#issuecomment-274310815

If someone who's interested in testing please let us know on this ticket.

k-okada, 2017-01-28:
> essential as noted in #279 (comment)

That file is only used in simulation, I think.

130s, 2017-02-10:
Thank you for clarification. Invalid this ticket is then.

closed 2017-02-10 by 130s