tork-a/rtmros_nextage #291: Checkerboard for Nextage/HIRO

y-yosuke, 2017-01-28:
Checking on Gazebo Simulation.

Adjusted camera posture with PointCloud
[image]
[image]

y-yosuke, 2017-01-28:
The checkerboard model put on gazebo.
[image]
[image]

y-yosuke, 2017-01-28:
Nextage Xtion Head Camera angle of view on CAD

Neck Angle : 65 deg

Right View
[image]

Front View
[image]

Perspective View
[image]

y-yosuke, 2017-02-06:
[Gazebo Simulation] Images about calculating Kinect TF using the checkerboard
[image]

[image]

[image]

[image]

Result of TF
```
$ rosrun tf tf_echo HEAD_JOINT1_Link camera_base_link
Failure at 48.045000000
Exception thrown:"HEAD_JOINT1_Link" passed to lookupTransform argument target_frame does not exist. 
The current list of frames is:

Failure at 48.045000000
Exception thrown:"HEAD_JOINT1_Link" passed to lookupTransform argument target_frame does not exist. 
The current list of frames is:

Failure at 49.058000000
Exception thrown:The tf tree is invalid because it contains a loop.
Frame CHEST_JOINT0_Link exists with parent WAIST.
Frame WAIST exists with parent object.
Frame HEAD_JOINT0_Link exists with parent CHEST_JOINT0_Link.
Frame HEAD_JOINT1_Link exists with parent HEAD_JOINT0_Link.
Frame LARM_JOINT0_Link exists with parent CHEST_JOINT0_Link.
Frame LARM_JOINT1_Link exists with parent LARM_JOINT0_Link.
Frame LARM_JOINT2_Link exists with parent LARM_JOINT1_Link.
Frame LARM_JOINT3_Link exists with parent LARM_JOINT2_Link.
Frame LARM_JOINT4_Link exists with parent LARM_JOINT3_Link.
Frame LARM_JOINT5_Link exists with parent LARM_JOINT4_Link.
Frame RARM_JOINT0_Link exists with parent CHEST_JOINT0_Link.
Frame RARM_JOINT1_Link exists with parent RARM_JOINT0_Link.
Frame RARM_JOINT2_Link exists with parent RARM_JOINT1_Link.
Frame RARM_JOINT3_Link exists with parent RARM_JOINT2_Link.
Frame RARM_JOINT4_Link exists with parent RARM_JOINT3_Link.
Frame RARM_JOINT5_Link exists with parent RARM_JOINT4_Link.
Frame camera_depth_frame exists with parent camera_rgb_frame.
Frame camera_rgb_frame exists with parent camera_base_link.
Frame camera_depth_optical_frame exists with parent camera_depth_frame.
Frame camera_link exists with parent camera_rgb_frame.
Frame camera_rgb_optical_frame exists with parent camera_rgb_frame.
Frame object exists with parent camera_depth_optical_frame.

The current list of frames is:
Frame CHEST_JOINT0_Link exists with parent WAIST.
Frame WAIST exists with parent object.
Frame HEAD_JOINT0_Link exists with parent CHEST_JOINT0_Link.
Frame HEAD_JOINT1_Link exists with parent HEAD_JOINT0_Link.
Frame LARM_JOINT0_Link exists with parent CHEST_JOINT0_Link.
Frame LARM_JOINT1_Link exists with parent LARM_JOINT0_Link.
Frame LARM_JOINT2_Link exists with parent LARM_JOINT1_Link.
Frame LARM_JOINT3_Link exists with parent LARM_JOINT2_Link.
Frame LARM_JOINT4_Link exists with parent LARM_JOINT3_Link.
Frame LARM_JOINT5_Link exists with parent LARM_JOINT4_Link.
Frame RARM_JOINT0_Link exists with parent CHEST_JOINT0_Link.
Frame RARM_JOINT1_Link exists with parent RARM_JOINT0_Link.
Frame RARM_JOINT2_Link exists with parent RARM_JOINT1_Link.
Frame RARM_JOINT3_Link exists with parent RARM_JOINT2_Link.
Frame RARM_JOINT4_Link exists with parent RARM_JOINT3_Link.
Frame RARM_JOINT5_Link exists with parent RARM_JOINT4_Link.
Frame camera_depth_frame exists with parent camera_rgb_frame.
Frame camera_rgb_frame exists with parent camera_base_link.
Frame camera_depth_optical_frame exists with parent camera_depth_frame.
Frame camera_link exists with parent camera_rgb_frame.
Frame camera_rgb_optical_frame exists with parent camera_rgb_frame.
Frame object exists with parent camera_depth_optical_frame.

Failure at 50.045000000
[... 119 lines cut ...]
- Rotation: in Quaternion [0.005, 0.705, 0.005, 0.709]
            in RPY (radian) [1.217, 1.556, 1.217]
            in RPY (degree) [69.731, 89.143, 69.753]
At time 56.980
- Translation: [-0.168, -0.016, 0.079]
- Rotation: in Quaternion [0.005, 0.705, 0.005, 0.709]
            in RPY (radian) [1.217, 1.556, 1.217]
            in RPY (degree) [69.734, 89.143, 69.755]
At time 57.994
- Translation: [-0.168, -0.016, 0.079]
- Rotation: in Quaternion [0.005, 0.705, 0.005, 0.709]
            in RPY (radian) [1.218, 1.556, 1.218]
            in RPY (degree) [69.775, 89.143, 69.796]
At time 59.021
- Translation: [-0.168, -0.016, 0.079]
- Rotation: in Quaternion [0.005, 0.705, 0.005, 0.709]
            in RPY (radian) [1.217, 1.556, 1.218]
            in RPY (degree) [69.754, 89.143, 69.772]
At time 59.950
- Translation: [-0.168, -0.016, 0.079]
- Rotation: in Quaternion [0.005, 0.705, 0.005, 0.709]
            in RPY (radian) [1.218, 1.556, 1.218]
            in RPY (degree) [69.775, 89.143, 69.792]

```

y-yosuke, 2017-02-14:
Checking on the real Nextage Open robot.

Manual Camera Posision/Angle Calibration (Using jsk_interactive_marker )

```
$ roslaunch nextage_ros_bridge nextage_ros_bridge_real.launch nameserver:=192.168.128.10
```

```
$ rviz -d nxt.rviz
```

```
$ roslaunch openni2_launch openni2.launch
```

```
$ roslaunch sensor_kinect.launch
```

Result of TF by Manual Camera Posision/Angle Calibration

```
$ rosrun tf tf_echo HEAD_JOINT1_Link camera_link
At time 1487054068.491
- Translation: [0.120, 0.041, 0.018]
- Rotation: in Quaternion [-0.029, 0.660, 0.033, 0.750]
            in RPY (radian) [0.000, 1.444, 0.088]
            in RPY (degree) [0.000, 82.720, 5.067]
At time 1487054068.691
- Translation: [0.120, 0.041, 0.018]
- Rotation: in Quaternion [-0.029, 0.660, 0.033, 0.750]
            in RPY (radian) [0.000, 1.444, 0.088]
            in RPY (degree) [0.000, 82.720, 5.067]
```

[image]

y-yosuke, 2017-02-14:
Checkerboard Camera Position/Angle Calibration

```
$ roslaunch nextage_ros_bridge nextage_ros_bridge_real.launch nameserver:=192.168.128.10
```

```
$ rviz -d nxt.rviz 
```

```
$ roslaunch xtion_calibration xtion_checkerboard.launch 
```

xtion_checkerboard.launch 
```
<launch>
  <arg name="launch_openni2" default="true" />

  <include file="$(find openni2_launch)/launch/openni2.launch"
           if="$(arg launch_openni2)">
    <arg name="depth_registration" value="true"/>
  </include>
  <include file="$(find freenect_launch)/launch/freenect.launch"
           unless="$(arg launch_openni2)">
  </include>

  <!-- checkerboard -->
  <group ns="/camera/rgb/" >
    <!-- <param name="single" value="1" /> -->
    <param name="display" type="int" value="1" />
    <param name="rect0_size_x" value="0.020" />
    <param name="rect0_size_y" value="0.020" />
    <param name="grid0_size_x" value="7" />
    <param name="grid0_size_y" value="4" />
    <param name="type0" value="test_object" />

    <node name="checkerboard_detector" 
          pkg="checkerboard_detector"
          type="checkerboard_detector">
      <remap from="image" to="/camera/rgb/image_rect_color" />
      <remap from="camera_info" to="/camera/rgb/camera_info" />
    </node>

    <node name="objectdetection_tf_publisher"
          pkg="xtion_calibration"
          type="objectdetection_tf_publisher.py" >
      <param name="latest_publish" value="true"/>
    </node>
  </group>

  <!--  right or left(0.1) / up or down (0.0)/ fornt or back (0.15) -->
  <node pkg="tf" 
        type="static_transform_publisher" 
        name="waist_to_checkerboard"
        args="0.060 -0.1105 0.335 1.5708 0.0 3.1415 /object /WAIST 100" />
  <node pkg="dynamic_tf_publisher" name="tf_publish" type="tf_publish.py"/>

  <!-- end checkerboard  -->

</launch>
```

Result of Checkerboard Camera Position/Angle Calibration

```
$ rosrun tf tf_echo HEAD_JOINT1_Link camera_link
At time 1487051620.258
- Translation: [0.112, 0.053, -0.001]
- Rotation: in Quaternion [0.032, 0.639, -0.030, 0.768]
            in RPY (radian) [0.061, 1.387, -0.028]
            in RPY (degree) [3.474, 79.482, -1.627]
At time 1487051620.786
- Translation: [0.112, 0.053, -0.001]
- Rotation: in Quaternion [0.032, 0.639, -0.030, 0.768]
            in RPY (radian) [0.061, 1.387, -0.028]
            in RPY (degree) [3.487, 79.478, -1.615]
```

[image]
[image]
[image]

y-yosuke, 2017-02-14:
Using Xtion Camera, point cloud are deviated by 20 - 30 mm from the model.

Is the way of Xtion Camera OpenNI launch different from that of Kinect?

y-yosuke, 2017-02-14:
In Gazebo Simulation, the point cloud seems on the correct position.

[image]

k-okada, 2017-02-14:
2017-02-14 16:20 GMT+09:00 Yosuke Yamamoto <notifications@github.com>:

> Using Xtion Camera, point cloud are deviated by 20 - 30 mm from the model.
>
> Is the way of Xtion Camera OpenNI launch different from that of Kinect?
>

I think that could happen, because xtion camera did not have "factory
carlibration" which kinect have

--
◉ Kei Okada

y-yosuke, 2017-02-14:
I found a difference of the relative position of `camera_rgb_frame` and `camera_frame` between the real robot/camera and Gazebo simulation. Please see the pictures below.

I think It is because of spawning a Kinecto model in the Gazebo simulation instead of Xtion.

Is it necessary to spawn? or read? Xtion model when I calibrate the real Xtion Camera?

RViz View of Real Robot/Camera
[image]
 
RViz View of Gazebo Simulation
[image]

y-yosuke, 2017-02-14:
In the Gazebo Simulation, Kinect model and the checkerboard model are spawned as below.

```
<launch>

  <param name="sensor_description"
         type="str"
         command="$(find xacro)/xacro.py '$(find xtion_calibration)/models/sensor_kinect.xacro'"
         />

  <node pkg="gazebo_ros"
        name="spawn_sensor_kinect"
        type="spawn_model" 
        args="-urdf -param sensor_description -model sensor_kinect
              -reference_frame HEAD_JOINT1_Link -P 1.5708 -x -0.15 -z 0.05"
        />

  <node pkg="robot_state_publisher"
        type="robot_state_publisher"
        name="sensor_kinect_state_publisher" >
    <remap from="robot_description" to="sensor_description" />
  </node>

  <node pkg="joint_state_publisher"
        type="joint_state_publisher"
        name="sensor_kinect_joint_state_publisher" >
    <remap from="robot_description" to="sensor_description" />
  </node>

  <node name="spawn_urdf" pkg="gazebo_ros" type="spawn_model" 
        args="-file $(find xtion_calibration)/models/checkerboard_horizontal.urdf
        -urdf -model checkerboard_horizontal
        -reference_frame WAIST" />

</launch>
```

y-yosuke, 2017-02-14:
@k-okada 
Is it the limit on using the Xtion camera? Or are there any ways to make it better?

y-yosuke, 2017-02-14:
Photos of the real robot/camera.
The position/posture of Xtion camera is about that of Xtion Head for Nextage.

[image]
[image]
[image]
[image]
[image]
[image]

k-okada, 2017-02-14:
i see
1. what happens if you show checker board to the depth camera, the TF of checker board should be equal to that of pint cloud

y-yosuke, 2017-02-14:
The checkerboard is too near to show depth point cloud.
The distance between Xtion camera and the checkerboard is about 220mm.

y-yosuke, 2017-02-14:
Are the depth data used in the caluculation of the relative coodination of the checkerboard and camera?

k-okada, 2017-02-14:
checkerboard detection uses gray image and intrinsic parameter(camera_info)

And what I want to check in 1. is that I intrinsic parameters are correct,
if not we may try
http://wiki.ros.org/camera_calibration/Tutorials/MonocularCalibration

k-okada, 2017-02-14:
> The distance between Xtion camera and the checkerboard is about 220mm.

Yes so we have to move a camera for this test

2017年2月14日(火) 19:30 Kei Okada <k-okada@jsk.t.u-tokyo.ac.jp>:

> checkerboard detection uses gray image and intrinsic parameter(camera_info)
>
> And what I want to check in 1. is that I intrinsic parameters are correct,
> if not we may try
> http://wiki.ros.org/camera_calibration/Tutorials/MonocularCalibration
>
>
> Are the depth data used in the caluculation of the relative coodination of
> the checkerboard and camera?
>
>
>
>
> ◉ Kei Okada
--
◉ Kei Okada

y-yosuke, 2017-02-14:
Tomorrow, I will check these below.

1. The relative position of PointCloud and TF by making the camera farther away from the Checkerboard
2. Comparing the PointCloud errors of the left hand and right hand
3. Calibrating a Monocular Camera

y-yosuke, 2017-02-14:
I got calibration file as shown below.

ost.yaml
```
image_width: 640
image_height: 480
camera_name: narrow_stereo
camera_matrix:
  rows: 3
  cols: 3
  data: [538.402903, 0.000000, 313.026434, 0.000000, 536.256068, 241.123289, 0.000000, 0.000000, 1.000000]
distortion_model: plumb_bob
distortion_coefficients:
  rows: 1
  cols: 5
  data: [0.044520, -0.126029, 0.001214, -0.000555, 0.000000]
rectification_matrix:
  rows: 3
  cols: 3
  data: [1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000]
projection_matrix:
  rows: 3
  cols: 4
  data: [537.498352, 0.000000, 311.984542, 0.000000, 0.000000, 537.211975, 241.021739, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000]
```

k-okada, 2017-02-15:
use this line to publish calibrated data into /camera_info
https://github.com/ros-drivers/openni2_launch/blob/indigo-devel/launch/openni2.launch#L18,
that will used for checker board node to get 3D position of the target. I
think adding calibrated data to launch file will change the location of
target position.

y-yosuke, 2017-02-15:
Before applying the result of the monocular calibration, in the same conditions as yesterday 
I checked two below.

1. Comparison of right and left hand deviations between the model and PointCloud 
2. Relative position of Checkerboard's PointCloud and object frame

---

1. Comparison of left and right hand deviations between the model and PointCloud 

Both PointCloud positions of right and left hand shift to the right.

[image]

2. Relative position of Checkerboard's PointCloud and object frame

It seems almost in a reasonable position.

[image]

k-okada, 2017-02-15:
>    1. Relative position of Checkerboard's PointCloud and object frame

> It seems almost in a reasonable position.

Ok, so let's move hand close to the checkerboard the n if relative pose
between checkerboard and correct, the point cloud shown in correcrt position

y-yosuke, 2017-02-15:
Hands close to the checkerboard

[image]

k-okada, 2017-02-15:
then, how about overrap between the robot model and point cloud

y-yosuke, 2017-02-15:
It still seems shifting right but the deviation is much smaller, 5-10mm.

k-okada, 2017-02-15:
5-10mm might be ok ->
http://www.mas.bg.ac.rs/_media/istrazivanje/fme/vol43/1/8_bkaran.pdf

y-yosuke, 2017-02-15:
I tried the camera at the Xtion Head position again.
The point cloud of the right and left hand shifts right 30-40mm same as before.

I think there is a possibility that the distortion at the periphery of the image may be affected.
I will apply the result of the monocular calibration to the camera_info.

[image]

y-yosuke, 2017-02-15:
I changed a file `xtion_checkerboard.launch` to read the camera infomation yaml file as below.

Then I tried the camera at the Xtion Head position again.
The point cloud of the right and left hand shifts right 30-40mm, the same as before.

xtion_checkerboard.launch
```
<launch>
  <arg name="launch_openni2" default="true" />

  <include file="$(find openni2_launch)/launch/openni2.launch"
           if="$(arg launch_openni2)">
    <arg name="depth_registration" value="true"/>
    <arg name="rgb_camera_info_url" value="file://${ROS_HOME}/camera_info/rgb_PS1080_PrimeSense.yaml" />
<!--    <arg name="rgb_camera_info_url" value="file://$(find xtion_calibration)/info/ost.yaml" /> -->
  </include>
  <include file="$(find freenect_launch)/launch/freenect.launch"
           unless="$(arg launch_openni2)">
  </include>

  <!-- checkerboard -->
  <group ns="/camera/rgb/" >
    <!-- <param name="single" value="1" /> -->
    <param name="display" type="int" value="1" />
    <param name="rect0_size_x" value="0.020" />
    <param name="rect0_size_y" value="0.020" />
    <param name="grid0_size_x" value="7" />
    <param name="grid0_size_y" value="4" />
    <param name="type0" value="test_object" />

    <node name="checkerboard_detector" 
          pkg="checkerboard_detector"
          type="checkerboard_detector">
      <remap from="image" to="/camera/rgb/image_rect_color" />
      <remap from="camera_info" to="/camera/rgb/camera_info" />
    </node>

    <node name="objectdetection_tf_publisher"
          pkg="xtion_calibration"
          type="objectdetection_tf_publisher.py" >
      <param name="latest_publish" value="true"/>
    </node>
  </group>

  <!--  right or left(0.1) / up or down (0.0)/ fornt or back (0.15) -->
  <node pkg="tf" 
        type="static_transform_publisher" 
        name="waist_to_checkerboard"
        args="0.060 -0.1105 0.335 1.5708 0.0 3.1415 /object /WAIST 100" />
  <node pkg="dynamic_tf_publisher" name="tf_publish" type="tf_publish.py"/>

  <!-- end checkerboard  -->

</launch>
```

[image]

k-okada, 2017-02-15:
how about move checkerboard to more lower, for example, on top of computer
box.

y-yosuke, 2017-02-15:
I tried the case of Xtion camera on a table, not on the robot head, as shown in the photo below.

[image]

The checkerboard calibraiotn result is fine, 
the deviation between the robot model and point cloud seems about 5mm in RViz.

[image]

y-yosuke, 2017-02-15:
I will try the case of the checkerboard at the WAIST height and on the comuputer height.

y-yosuke, 2017-02-15:
The case of the checkerboard on the computer.

The deviation between the robot model and the point cloud seems about 10-20mm in the RViz.

[image]

y-yosuke, 2017-02-15:
The case of the checkerboard on the WAIST.

The deviation between the robot model and the point cloud seems about 10-20mm in the RViz.

[image]

y-yosuke, 2017-02-15:
I'm sorry that it is time to quit tests today.

k-okada, 2017-02-16:
Sorry for taking your time..
I think we need to check https://github.com/tork-a/rtmros_nextage/issues/291#issuecomment-279642237 again

1. I should have to check again, but usually, Z-axis is the frontal direction in camera optical frame (X axis in camera frame), and rgb frame and depth frame should be side by side. So

 'RViz View of Real Robot/Camera' might be strange, 1. optical frame should have Z axis as frontal direction. Also position of rgb/link? or depth/link position is same. That's different from fetch simulation shown in below, which launch file did you use?

 'RViz View of Gazebo Simulation' is also strange, rgb/depth frame must be horizontal, that would be my fault.

[image]

k-okada, 2017-02-16:
may be my comment on https://github.com/tork-a/rtmros_nextage/issues/291#issuecomment-280188638 was not correct.
From https://github.com/tork-a/rtmros_nextage/issues/291#issuecomment-279974543, 
1. checker board (ond related nodes) publishes tf of board -> camera link, which should be correct of you uses correct intrinsic parameter
2. the position of camera to the robot body is determined by manual, using rviz plugin, so if you move the camera position, the checkerboard position (both tf and point cloud) in rviz also moves, and the relative position between tf and point cloud always fixed.
3. So we can move camera position so that point cloud of arm is as same position as rviz model position

 - TF tree: base_link -> waist ->(can be changed through rviz plugin)-> camera ->(chckerboard node output) -> checker board

4. If you move arm position the point cloud may differ from rviz model position. This happens when the robot has incorrect encoder value, or intrinsic parameter of the camera is not correct. Or your manual calibration in 3. was not correct.
5. If you are shoure about the relative position between waist and checker board, then you can build tf tree something like follows, then you can specify the location of camera without using manual calibration using rviz plugins

- TF tree:  base_link -> waist -> checker board ->(checkerboard onde output)-> camera

y-yosuke, 2017-02-16:
I found my code was wrong!!

The checkerbord `object` frame must be based on the frame `CHEST_JOINT0_Link`, 
but in my code that `object` frame is based on `WAIST`.

And I tested those without servo control of the joints, so sometimes the waist-chest joint angle was around 0 deg, sometimes not 0 deg.

I'm very sorry. It's my fault.

[ WRONG ] xtion_checkerboard.launch
```
<launch>
  <arg name="launch_openni2" default="true" />

  <include file="$(find openni2_launch)/launch/openni2.launch"
           if="$(arg launch_openni2)">
    <arg name="depth_registration" value="true"/>
<!--    <arg name="rgb_camera_info_url" value="file://${ROS_HOME}/camera_info/rgb_PS1080_PrimeSense.yaml" /> -->
<!--    <arg name="rgb_camera_info_url" value="file://$(find xtion_calibration)/info/ost.yaml" /> -->
  </include>
  <include file="$(find freenect_launch)/launch/freenect.launch"
           unless="$(arg launch_openni2)">
  </include>

  <!-- checkerboard -->
  <group ns="/camera/rgb/" >
    <!-- <param name="single" value="1" /> -->
    <param name="display" type="int" value="1" />
    <param name="rect0_size_x" value="0.020" />
    <param name="rect0_size_y" value="0.020" />
    <param name="grid0_size_x" value="7" />
    <param name="grid0_size_y" value="4" />
    <param name="type0" value="test_object" />

    <node name="checkerboard_detector" 
          pkg="checkerboard_detector"
          type="checkerboard_detector">
      <remap from="image" to="/camera/rgb/image_rect_color" />
      <remap from="camera_info" to="/camera/rgb/camera_info" />
    </node>

    <node name="objectdetection_tf_publisher"
          pkg="xtion_calibration"
          type="objectdetection_tf_publisher.py" >
      <param name="latest_publish" value="true"/>
    </node>
  </group>

  <!--  right or left(0.1) / up or down (0.0)/ fornt or back (0.15) -->

<!--  <node pkg="tf" 
        type="static_transform_publisher" 
        name="waist_to_checkerboard"
        args="0.060 -0.1105 0.335 1.5708 0.0 3.1415 /object /WAIST 100" /> -->

  <!-- Position NOT Sure!! Checkerboard on WAIST face to top -->
  <node pkg="tf" 
        type="static_transform_publisher" 
        name="waist_to_checkerboard"
        args="0.060 -0.070 0.022 1.5708 0.0 3.1415 /object /WAIST 100" />

  <!-- Position NOT Sure!! Checkerboard on computer face to top -->
<!--  <node pkg="tf" 
        type="static_transform_publisher" 
        name="waist_to_checkerboard"
        args="0.060 -0.095 -0.155 1.5708 0.0 3.1415 /object /WAIST 100" /> -->

  <!-- Checkerboard face to front -->
<!--  <node pkg="tf" 
        type="static_transform_publisher" 
        name="waist_to_checkerboard"
        args="0.060 0.312 0.0785 0.0 1.5708 1.5708 /object /WAIST 100" /> -->

  <node pkg="dynamic_tf_publisher" name="tf_publish" type="tf_publish.py"/>

  <!-- end checkerboard  -->

</launch>
```

y-yosuke, 2017-02-17:
I have a problem of multiple parent frames, when the checkerboard frame `object` has reference to `CHEST_JOINT0_Link` by the code shown below. 
It means `CHEST_JOINT0_Link` has two parent frames, `WAIST` and `object`, and these conflict.

I'm thinking about what kind of method is good. What is the good way to solve it?

1. Set `object` as a child frame of `WAIST` and keep the chest joint angle 0 deg by servo control
2. Set `object` as a child frame of `CHEST_JOINT0_Link`, set "Fixed Frame" to `CHEST_JOINT0_Link` in RViz and ignore `WAIST` frame
3. Or any other ways?

```
<launch>
  <arg name="launch_openni2" default="true" />

  <include file="$(find openni2_launch)/launch/openni2.launch"
           if="$(arg launch_openni2)">
    <arg name="depth_registration" value="true"/>
    <arg name="rgb_camera_info_url" value="file://${ROS_HOME}/camera_info/rgb_PS1080_PrimeSense.yaml" />
<!--    <arg name="rgb_camera_info_url" value="file://$(find xtion_calibration)/info/ost.yaml" /> -->
  </include>
  <include file="$(find freenect_launch)/launch/freenect.launch"
           unless="$(arg launch_openni2)">
  </include>

  <!-- checkerboard -->
  <group ns="/camera/rgb/" >
    <!-- <param name="single" value="1" /> -->
    <param name="display" type="int" value="1" />
    <param name="rect0_size_x" value="0.020" />
    <param name="rect0_size_y" value="0.020" />
    <param name="grid0_size_x" value="7" />
    <param name="grid0_size_y" value="4" />
    <param name="type0" value="test_object" />

    <node name="checkerboard_detector" 
          pkg="checkerboard_detector"
          type="checkerboard_detector">
      <remap from="image" to="/camera/rgb/image_rect_color" />
      <remap from="camera_info" to="/camera/rgb/camera_info" />
    </node>

    <node name="objectdetection_tf_publisher"
          pkg="xtion_calibration"
          type="objectdetection_tf_publisher.py" >
      <param name="latest_publish" value="true"/>
    </node>
  </group>

  <node pkg="tf" 
        type="static_transform_publisher" 
        name="chest_to_checkerboard"
        args="0.060 -0.1105 0.335 1.5708 0.0 3.1415 /object /CHEST_JOINT0_Link 100" />

  <node pkg="dynamic_tf_publisher" name="tf_publish" type="tf_publish.py"/>

  <!-- end checkerboard  -->

</launch>
```

k-okada, 2017-02-17:
Other method might be set object as a child of CHEST_JOINT0_Link and fix
objectdetection_tf_publisher to publish inverse of the pose.

But if 1 or 2 works ok, that's fine. I prefer 1.

y-yosuke, 2017-02-24:
I adopted 1.

> 1. Set object as a child frame of WAIST and keep the chest joint angle 0 deg by servo control

y-yosuke, 2017-03-04:
I am editing the draft of the section on camera calibration to add to the ROS Wiki.
After confirmation with real robot I will apply for Pull Request and add this section to ROS Wiki.

---

# 5.1 Camera TF Calibration Using Checkerboard

You can calibrate TF between robot link and camera link using checkerboard

## 1. Install checkerboard on Nextage robot

Install a checkerboard on chest (? or waist ?) of Nextage robot as shown below.

[image]
[image]

## 2. Run rtm_ros_bridge

Run rtm_ros_bridge on your PC.

```
$ roslaunch nextage_ros_bridge nextage_ros_bridge_real.launch nameserver:=%HOSTNAME%
```

http://wiki.ros.org/rtmros_nextage/Tutorials/Operating%20Hiro%2C%20NEXTAGE%20OPEN#Run_rtm_ros_bridge

## 3. Run python script and keep the angle of "CHEST_JOINT0" as 0

```
$ ipython -i `rospack find nextage_ros_bridge`/script/nextage.py -- --host %HOSTNAME%
```

You should keep the angle of "CHEST_JOINT0" as 0 [deg/rad].

```
In [1]: robot.goInitial()
In [2]: robot.servoOff()
In [3]: robot.servoOn('CHESET_JOINT0')
In [4]: robot.getJointAngles() 
```

## 4. Turn the camera so that the checkerboard appears

Turn the camera to the chekcerboard by your hands or using joint control.

Example of commands for using joint control
```
In [1]: robot.goInitial()
In [2]: robot.setJointAnglesOfGroup( 'head', [ 0.0, 45.0 ], 5.0 )
In [3]: robot.getJointAngles() 
```

## 5. Run checkerboard detector

Xtion
```
$ roslaunch nextage_calibration camera_checkerboard_chest.launch 
```

Kinect
```
$ roslaunch nextage_calibration camera_checkerboard_chest.launch launch_openni2:=false
```

[image]

## 6. Run RViz

For checking the operation status of each function, run RViz and add PointCloud2 and TF to the display.

```
$ rviz
```

## 7. Aquisition of TF between camera link and parent link

Run `tf_echo` to aquire TF between camera link and parent link.

```
$ rosrun tf tf_echo camera_link HEAD_JOINT1_Link
```

Example of `tf_echo` result

```
At time 59.089
- Translation: [0.198, 0.014, -0.103]
- Rotation: in Quaternion [0.000, -0.705, -0.002, 0.709]
            in RPY (radian) [0.532, -1.565, -0.535]
            in RPY (degree) [30.482, -89.646, -30.632]

```

## 8. Quit all nodes

`Ctrl-C` to quit all nodes.

## 9. Additional Calibration for Xtion camera

When your camera has no calibration infomation, run monocular camera calibration in the link below.

http://wiki.ros.org/camera_calibration/Tutorials/MonocularCalibration

-----

# 5.2 Gazebo Simulation - Camera TF Calibration Using Checkerboard

You can confirm how it works in Gazebo simulation.

## 1. Launch Gazebo with Nextage robot model

```
$ roslaunch nextage_gazebo nextage_world.launch
```

## 2. Spawn models of kinect and chekcerboard

After the robot finished the initial motion in Gazebo, launch `gazebo_kinect_checkerboard_chest.launch` to spawn kinect and the checkerborad model in Gazebo.

```
$ roslaunch nextage_calibration gazebo_kinect_checkerboard_chest.launch
```

[image]

## 3. Run checkerboard detector

```
$ roslaunch nextage_calibration camera_checkerboard_chest.launch launch_openni2:=false
```

[image]

## 4. Run RViz

Run RViz and add PointCloud2 and TF to the display.

```
$ rviz
```

[image]

## 5. Aquire TF between camera link and parent link

Run `tf_echo` to aquire TF between camera link and parent link.

```
$ rosrun tf tf_echo camera_link HEAD_JOINT1_Link
```

Example of `tf_echo` result

```
At time 59.089
- Translation: [0.198, 0.014, -0.103]
- Rotation: in Quaternion [0.000, -0.705, -0.002, 0.709]
            in RPY (radian) [0.532, -1.565, -0.535]
            in RPY (degree) [30.482, -89.646, -30.632]

```

## 6. Quit all nodes

`Ctrl-C` to quit all nodes.

k-okada, 2017-03-07:
lgtm, 
Please add how to update transformation betwee  'camera_link HEAD_JOINT1_Link' of your launch file based on inforamtion provided by 'tf_echo'
if you can provide `calibration.rviz` and start rviz as
```
rviz -d `rospack find nextage_calibration`/conf/calibration.rviz
```
, it's perfect

y-yosuke, 2017-03-07:
Now I'm testing the checkerboard on chest using real robot with servo control.

The Differences between PointCloud and RViz robot model seems about as below.
- dx, dy < 5mm
- dz : about 10mm (or RViz model hands are smaller than actual robot a little ???)

Overview on RViz
[image]

PointCloud2 of the hands - Top View
[image]

PointCloud2 of the hands - Front View
[image]

y-yosuke, 2017-03-07:
Then I'll try a checkerboard on the waist.

y-yosuke, 2017-03-07:
I'm testing the checkerboard on waist of real robot.
[image]

y-yosuke, 2017-03-07:
The checkerboard on waist is not recognized when it is installed at a predetermined position, but it is recognized that some angle is attached. 

It seems that it does not seem to be recognized by the light hit condition.

At a predetermined position
[image]

With some angles
[image]
[image]
[image]

y-yosuke, 2017-03-07:
When I turn off the room light above the robot, the checkerboard on waist is detected well.

[image]

The deviations between PointCloud2 and the robot model in RViz are like below.
- dx, dy, dz < 5mm

Overview of PointCloud2 and the robot model on RViz
[image]

Top View of Hands
[image]

Front View of Hands
[image]

y-yosuke, 2017-03-07:
Photo of the real robot with checkerboard on waist
[image]

TF result of checkerboard on waist
```
$ rosrun tf tf_echo camera_link HEAD_JOINT1_Link
At time 1488869848.083
- Translation: [-0.022, -0.081, -0.122]
- Rotation: in Quaternion [-0.001, -0.683, 0.025, 0.730]
            in RPY (radian) [-0.485, -1.495, 0.519]
            in RPY (degree) [-27.790, -85.654, 29.736]
At time 1488869848.533
- Translation: [-0.022, -0.080, -0.122]
- Rotation: in Quaternion [-0.001, -0.683, 0.024, 0.730]
            in RPY (radian) [-0.490, -1.495, 0.522]
            in RPY (degree) [-28.060, -85.685, 29.923]
At time 1488869849.458
- Translation: [-0.022, -0.080, -0.122]
- Rotation: in Quaternion [-0.001, -0.683, 0.025, 0.730]
            in RPY (radian) [-0.485, -1.495, 0.518]
            in RPY (degree) [-27.769, -85.645, 29.686]
At time 1488869850.466
- Translation: [-0.022, -0.080, -0.122]
- Rotation: in Quaternion [-0.001, -0.683, 0.025, 0.730]
            in RPY (radian) [-0.488, -1.495, 0.521]
            in RPY (degree) [-27.950, -85.666, 29.850]
At time 1488869851.463
- Translation: [-0.022, -0.080, -0.122]
- Rotation: in Quaternion [-0.001, -0.683, 0.025, 0.730]
            in RPY (radian) [-0.489, -1.496, 0.523]
            in RPY (degree) [-28.041, -85.694, 29.975]
At time 1488869852.469
- Translation: [-0.022, -0.080, -0.122]
- Rotation: in Quaternion [-0.001, -0.683, 0.025, 0.730]
            in RPY (radian) [-0.487, -1.495, 0.520]
            in RPY (degree) [-27.885, -85.655, 29.804]
```

y-yosuke, 2017-03-07:
Retake TF and photos of checkerboard on chest

TF
```
$ rosrun tf tf_echo camera_link HEAD_JOINTLink
At time 1488871926.392
- Translation: [-0.029, -0.074, -0.114]
- Rotation: in Quaternion [-0.003, -0.657, 0.009, 0.754]
            in RPY (radian) [-0.125, -1.434, 0.133]
            in RPY (degree) [-7.139, -82.136, 7.629]
At time 1488871926.895
- Translation: [-0.029, -0.074, -0.114]
- Rotation: in Quaternion [-0.003, -0.657, 0.009, 0.754]
            in RPY (radian) [-0.125, -1.433, 0.133]
            in RPY (degree) [-7.141, -82.126, 7.621]
At time 1488871927.906
- Translation: [-0.029, -0.074, -0.114]
- Rotation: in Quaternion [-0.003, -0.657, 0.009, 0.753]
            in RPY (radian) [-0.125, -1.434, 0.133]
            in RPY (degree) [-7.161, -82.148, 7.644]
At time 1488871928.889
- Translation: [-0.029, -0.074, -0.114]
- Rotation: in Quaternion [-0.003, -0.658, 0.009, 0.753]
            in RPY (radian) [-0.126, -1.434, 0.134]
            in RPY (degree) [-7.199, -82.159, 7.679]
At time 1488871929.889
- Translation: [-0.029, -0.074, -0.114]
- Rotation: in Quaternion [-0.003, -0.658, 0.009, 0.753]
            in RPY (radian) [-0.126, -1.434, 0.134]
            in RPY (degree) [-7.219, -82.166, 7.703]
At time 1488871930.885
- Translation: [-0.029, -0.074, -0.114]
- Rotation: in Quaternion [-0.003, -0.658, 0.009, 0.753]
            in RPY (radian) [-0.126, -1.434, 0.135]
            in RPY (degree) [-7.243, -82.176, 7.721]
```

Photos
[image]
[image]
[image]
[image]
[image]

k-okada, 2017-03-07:
looks nice!

y-yosuke, 2017-03-07:
As the result on the real robot looks good, 
so if there is nothing else, I will finish work on real machine today.

k-okada, 2017-03-07:
thank you, I think it's ok to close.

y-yosuke, 2017-03-07:
Thank you for the a lot of support. @k-okada @130s

y-yosuke, 2017-03-07:
Summary of Checkerboards on Chest / Waist - Strength / Weakness

- Checkerboard on Chest
  - Strength
    - Not have to remove the bolts of the structural member
       - Just replace the bolts of the chest cover
    - Robust for room light conditions
    - Can be installed not only horizontally but also vertically in front
  - Weakness
    - Too near for the depth sensor
       - Cannot see the checkerboard point cloud in RViz
    - Have to be controlled by servo
       - Waist-Chest angle must be Zero

- Checkerboard on Waist
  - Strength
    - In the range of the depth sensor 
      - Can see the checkerboard point cloud in RViz
    - Not have to be controlled by servo
  - Weakness
    - Have to remove the bolts of the structural member
    - Influenced by room light conditions

130s, 2017-03-07:
This is a great addition to NEXTAGE Open. Thank you @y-yosuke 
Would you also add the change https://github.com/tork-a/rtmros_nextage/issues/291#issuecomment-284150285 to the wiki? If not I can take care of that.

CC @shzargar

y-yosuke, 2017-03-08:
I will add the change https://github.com/tork-a/rtmros_nextage/issues/291#issuecomment-284150285 to the wiki with some modification. 
Thank you @130s .

y-yosuke, 2017-03-10:
I added a new section "Xtion/Kinect Camera TF Calibration" to ROS Wiki.
http://wiki.ros.org/rtmros_nextage/Tutorials/CalibrateKinect#Xtion.2BAC8-Kinect_Camera_TF_Calibration

130s, 2017-03-10:
#298 is now merged, and the new info is added to wiki https://github.com/tork-a/rtmros_nextage/issues/291#issuecomment-285553145
Thank you very much for the complete work @y-yosuke

closed 2017-03-10 by 130s

k-okada, 2017-03-11:
@y-yosuke

Sorry for late response,

1. please move 1.overview to 4. more elaborated calibration to the end
of the page, How about add  "3. Calibrate Head-mount Camera and Hands
coordinates (Working in progess)" section and put these section as
subsection, then the title of the page will be "Xtion/Kinect Camera TF
Calibration" instead of "Calibrate Head...."

2. If I understand correctly, 1.9 is for gazebo, so it should be 2.7
and we need to write how to update TF settings for real robot, maybe
running https://github.com/start-jsk/rtmros_hironx/blob/indigo-devel/hironx_ros_bridge/launch/hironx_kinect.launch
with arguments may ok, and TF_COORD_XYZ_HIRO014 can be TF_COORD_XYZ
--
◉ Kei Okada

y-yosuke, 2017-03-13:
I add a new file "nextage_calibration/launch/nextage_kinect_xtion.launch" based on "hironx_kinect.launch".

- new file:
  - https://github.com/y-yosuke/rtmros_nextage/blob/add_nextage_calibration/nextage_calibration/launch/nextage_kinect_xtion.launch
  - based on:
    - https://github.com/start-jsk/rtmros_hironx/blob/indigo-devel/hironx_ros_bridge/launch/hironx_kinect.launch
- Changes from "hironx_kinect.launch"
  - add launch option: `launch_openni2` to choose `openni2` for Xtion or `freenect` for Kinect
  - add launch option: `TF_COORD_XYZ` to modify `camera_link` TF 
  - add launch option: `TF_COORD_RPY` to modify `camera_link` TF 
- Description on this matter in ROS Wiki
  - http://wiki.ros.org/rtmros_nextage/Tutorials/CalibrateKinect#Use_of_the_aquired_camera_link_TF

[1 more reply]