tork-a/rtmros_nextage #292: How to use Moveit class and NextageClient Class in a python program

shzargar, 2017-01-30:
I want to use goInitial method from NextageClient Class and then set_pose_target from moveit_commander class. 

I tried to create 2 instances:
	robot = moveit_commander.RobotCommander()
        robot1 = nextage_client.NextageClient()
but, I got error message when I use:
robot1.goInitial()
or 
robot1.goInitial(tm=3)
or
robot1.goInitial(tm=3, wait=False)

When I remove the goInitial line, the program works (except starting from initial pose).

Output:
[ INFO] [1485745548.098251511]: Loading robot model 'NextageOpen'...
configuration ORB with Nxtag:15005
[hrpsys.py] , JntAnglesOfGr=torso, INITPOSE[i]=[0], tm=3, wait=False
Traceback (most recent call last):
  File "./nextage_moveit_botharm_1.py", line 291, in <module>
    main()
  File "./nextage_moveit_botharm_1.py", line 33, in main
    robot1.goInitial(tm=3, wait=False)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/nextage_ros_bridge/nextage_client.py", line 353, in goInitial
    return HIRONX.goInitial(self, tm, wait, init_pose_type)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/hironx_ros_bridge/hironx_client.py", line 391, in goInitial
    tm, wait=False)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/hrpsys/hrpsys_config.py", line 1058, in setJointAnglesOfGroup
    ret = self.seq_svc.setJointAnglesOfGroup(gname, angles, tm)
AttributeError: 'NoneType' object has no attribute 'setJointAnglesOfGroup'
terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::lock_error> >'
  what():  boost: mutex lock failed in pthread_mutex_lock: Invalid argument
Aborted (core dumped)

130s, 2017-01-30:
I can confirm that the same error happens, if I try to instantiate `nextage_client.NextageClient` by itself like you did.

**Reason why you get the error**
In fact, `NextageClient` needs some special steps to start, which you can see in [this portion](https://github.com/tork-a/rtmros_nextage/blob/e0d6c63b76e946fd2e90a1c0b42e1e9f5dacd549/nextage_ros_bridge/script/nextage.py#L47_L71) in `main` method in `nextage.py`. There basically we're setting some parameters that are required for the communication between lower level components (called hrpsys, OpenRTM).

**How can you avoid this error, and still use your custom python class for NEXTAGE Open**

- One way is to simply duplicate [nextage.py](https://github.com/tork-a/rtmros_nextage/blob/indigo-devel/nextage_ros_bridge/script/nextage.py), and replace the line where [NextageClient](https://github.com/tork-a/rtmros_nextage/blob/e0d6c63b76e946fd2e90a1c0b42e1e9f5dacd549/nextage_ros_bridge/script/nextage.py#L71) class is instantiated (also remove the rest of the file where the instance of NextageClient is accessed).
- The above way not smart nor clean because you simply duplicate the majority of the code, which is a bad practice in software development. I think [hrpsys_tools/hrpsys_tools_config.py](https://github.com/start-jsk/rtmros_common/blob/master/hrpsys_tools/scripts/hrpsys_tools_config.py) is intended to serve for this purpose (see also https://github.com/start-jsk/rtmros_common/issues/340 for the discussion when it was created), but unfortunately I'm not sure how to use that with NEXTAGE Open.

Closing although more questions welcomed.

closed 2017-01-30 by 130s

130s, 2017-01-30:
Although no additional info available, I documented the discussion above [on the tutorial](http://wiki.ros.org/rtmros_nextage/Tutorials/Programming_Hiro_NEXTAGE_OPEN_RTM#To_use_your_custom_Python_code_that_uses_hrpsys_commands).

shzargar, 2017-01-31:
Thanks, it is working for my now.