tork-a/rtmros_nextage #294: Incorrect manual description - Run rtm_ros_bridge (Nextage Open)

y-yosuke, 2017-02-14:
Incorrect
```
$ roslaunch nextage_ros_bridge nextage_ros_bridge.launch nameserver:=%HOSTNAME%  (NEXTAGE OPEN)
```

Correct - "_real" is needed.
```
$ roslaunch nextage_ros_bridge nextage_ros_bridge_real.launch nameserver:=%HOSTNAME%  (NEXTAGE OPEN)
```

k-okada, 2017-02-14:
thanks, if you mean "manual" to
http://wiki.ros.org/rtmros_nextage/Tutorials/Operating%20Hiro,%20NEXTAGE%20OPEN,
feel free to login and fix this

y-yosuke, 2017-02-14:
Thanks a lot, I forgot to paste the link.
Yes, I mean "manual" is http://wiki.ros.org/rtmros_nextage/Tutorials/Operating%20Hiro,%20NEXTAGE%20OPEN#Run_rtm_ros_bridge .

I applied for permission to edit wiki, so I will edit it after acceptance.

y-yosuke, 2017-02-14:
I fixed wiki and close this issue.

closed 2017-02-14 by y-yosuke